nordicsemi/IOS-nRF-Mesh-Library #265: Re-installing the app on the same phone causes trouble when re-importing mesh network

deverickasih, 2020-04-23:
**Describe the bug**
I cannot interact with devices after i reimport the mesh json file into the same phone i used to export the mesh json file, after i uninstall the app and reinstalled it.

**To Reproduce**
(1) i download the nrf mesh for the first time.
(2) i create network and add devices
(3) after testing out everything works (light can on/off), i then export the mesh json file
(4) i uninstall the app and then reinstall the app
(5) i import the mesh json file into the re-installed app
(6) but it doesn't work -> i can't turn on/off the light [<- if i use the 0x0001 provisioner]
(7) but it works -> i can turn on/off the light [<- if i use the provisioner that was generated by the nrf mesh after i reimport the mesh json file -> 0x199B provisioner] {the second provisioner was automatically generated in the iOS nrf mesh}

**Expected behavior**
Someone told me that if i switch phones, i must change the provisioner address on that other phone. But i am using the same phone, why must i also have to change the provisioner address for it to work?

I integrated the mesh library into a smart home project. The user have been using the mesh network in his phone for a couple of weeks. He logs out from our mobile app and re-login multiple times with no problems. And then, one day he accidentally deleted the app. And when he reinstalled the app, he now can't access anything in his house. But if i change his provisioner address, he can re-interact with his mesh devices. He did not change his phone. He is using the same phone.

Is this a bug or is this according to the ways of the mesh network? If this is not a bug then tell me, how many provisioners must i save in the mesh network for every user, just in case the user uninstalls the app and reinstalls it again and again?

:)

philips77, 2020-04-23:
This is expected behavior. The sequence numbers are not shared in the json database, they change with every message sent/received. When you import a configuration on a phone, whether it's a new phone, or one you used with the app previously, the `UserDefaults` are empty, with sequence numbers together. So the library starts sending messages with sequence number = 0. If the IV Index hasn't changed since, this will produce too low SeqAuth, and such messages will be discarded by other nodes. However, if you send messages over and over, you eventually reach the sequence number high enough for the message to be accepted. For it to be accepted by a remote node,  together with the IV Index it needs to be higher than last one received by this node.

In version 2.2.2 I'm adding a setter and getter for the sequence numbers for local elements, but the preferred way is would be to create a new provisioner with unique unicast address, rather than restoring the old sequence number. The sample app will not use those methods, but you can in your app. We just discourage you to do so :)

philips77, 2020-04-23:
There is one issue with the recommended method. Having a new provisioner adds new unicast addresses to the network. However, nodes have something called Replay Protection Count (RPC), which is the maximum number of addresses the node can EVER receive messages from. Each node must remember the last sequence number of every node it has received message from to ensure no replay attack is possible. The RPC is the size of the memory for those sequence numbers. If the buffer is full, messages from new nodes will be discarded, as the node cannot ensure the messages are not repeated. When building a device, you need to make sure the RPC is high enough for the predicted network size. Remember, that all Elements have unicast addresses, not only nodes.

philips77, 2020-04-23:
Fortunately not all nodes send messages, or not all send them to all nodes. So you may have more elements in the network than RPC of the nodes, but need to be careful.

closed 2020-04-27 by philips77