nordicsemi/IOS-nRF-Mesh-Library #268: Local Elements cannot extended after first node was provisioned

phillippbertram, 2020-04-28:
**Describe the bug**
It is currently not possible to dynamically extend the `localElements` of the `MeshNetworkManager` at runtime **after** the first device has been provisioned and was added as `node` to the `MeshNetwork`

**To Reproduce**
Steps to reproduce the behavior:

1. Create a `MeshNetwork` with a `localProvider` as in the example app

```swift
let localProvisioner = Provisioner(name: UIDevice.current.name,
                                           allocatedUnicastRange: [AddressRange(0x0001...0x199A)],
                                           allocatedGroupRange: [AddressRange(0xC000...0xCC9A)],
                                           allocatedSceneRange: [SceneRange(0x0001...0x3333)])

let network = networkManager.createNewMeshNetwork(withName: "My Network", by: localProvisioner)
```
2. Create some `Elements` and add it to `localElements`

```swift
let primaryElement = Element(name: "Primary Element", location: .first, models: [
    Model(sigModelId: GenericOnOffClientId, delegate: GenericOnOffClientDelegate())
])
let secondaryElement = Element(name: "Secondary Element", location: .second, models: [
    Model(sigModelId: GenericOnOffClientId, delegate: GenericOnOffClientDelegate())
])

networkManager.localElements = [primaryElement, secondaryElement]
```

✅ Here everything works as expected. The `localElements` then has 2 elements.

3. Add another Element
```swift
let thirdElement = Element(name: "Third Element", location: .third, models: [
    Model(sigModelId: GenericOnOffClientId, delegate: GenericOnOffClientDelegate())
])

var extendedElements = networkManager.localElements
extendedElements.append(thirdElement)
networkManager.localElements = extendedElements
```
✅  Here everything works as expected. The `localElements` then has 3 elements.

4. Add a Node to the network through the provisioning process

5. Add another Element like in step 3

❌  `networkManager.localElements` does not contain the new created element

**Expected behavior**
The `localElements` does contain the new created element even if a node exists in the current network.

**Platform details:**
 - Device: iPhoneX and other
 - OS: 13.x and below
 - Library Version 2.2.1

philips77, 2020-04-28:
Hi, I think the issue you have is with lack of available Unicast Address. The primary Element gets the address assigned during provisioning. Each following Element gets +1 to its address. The code ensures that Unicast Addresses do not repeat. If you provisioned a Node with a following address, and add a new Element to the local Node, the address cannot be assigned, so it's "cut" from local Elements. In step 4, during provisioning, set the address to some other address. In version 2.2.2 I'm adding a method to get a next available address from some offset. 2.2.2 will be released shortly.

closed 2020-04-28 by philips77

phillippbertram, 2020-04-28:
Perfect, thanks for the quick answer @philips77 
Then I am looking forward to the update, so that I can implement my UseCase if necessary. 😊

philips77, 2020-04-28:
2.2.2 released on CocoaPods.

philips77, 2020-04-28:
Find the new parameter in the method here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/cc1c35300f3dc20e33497df2f1de8d59e04324f8/nRFMeshProvision/Classes/Mesh%20API/MeshNetwork%2BAddress.swift#L89