nordicsemi/IOS-nRF-Mesh-Library #273: Not a Subscribe Model

SonnySwolf, 2020-05-12:
**Describe the bug**
PROVISIONER Vendor model can't subscribe 0xC0000.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Network'
2. Click on 'This PROVISIONER'
3. Click on 'Primary Element'
4. Click on 'Vendor Model ID'
5. Click on 'Subscribe'
6. Click on 'New Group'
7. Click on 'Done'
8. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I want to use the vendor model to receive a group message.

**Platform details:**
 - Device: [iPhone8]
 - OS: [iOS13.3]
 - Library Version (or Commit)
Application Version 2.2.3

philips77, 2020-05-12:
In the sample app, the implemented Vendor Model is "Simple On/Off Client" by Nordic. The sample app does not support subscriptions for this model:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f9a09aafe091e9597b99c6d615278a681d79561c/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff/SimpleOnOffClientDelegate.swift#L49-L51
As you see in this file, the only supported messages in this model are *SimpleOnOffStatus*.

You may implement subscriptions the way they are done e.g. in this model:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f9a09aafe091e9597b99c6d615278a681d79561c/Example/nRFMeshProvision/Mesh%20Network/GenericOnOffServerDelegate.swift#L34-L36
This model implements *GenericOnOffSet* and *GenericOnOffSetUnacknowledged*.

closed 2020-05-12 by philips77

SonnySwolf, 2020-05-13:
Thank you very much!!!