nordicsemi/IOS-nRF-Mesh-Library #274: nRF Mesh App disconnect after exchange att-mtu size

Kupeo, 2020-05-12:
**Describe the bug**
when use nRF Mesh app, After the att-mtu is exchanged, the connection is disconnected, and the analyzer shows that the master is actively disconnected.

While I'm goto system setting, Open Bluetooth, manual shut and open again, 
this problem has been solved. I'm not sure where the problem occurs

**Platform details:**
 - Device: [e.g. iPhone11]
 - OS: [e.g. iOS13.3.1]
 - nRF Mesh Version 2.2.1 Build Version 1

**Logs / Screenshots**

[image]

philips77, 2020-05-12:
This might have been related to caching. Turning Bluetooth Off and On invalidates caches.

Kupeo, 2020-05-12:
You meant this is iOS system cache issues?

philips77, 2020-05-13:
Yes. The connection is maintained by the system. nRF Mesh app could only disconnect if it would connect to an unsupported device, that is after service discovery. If unsupported services were cached (by connecting to this device previously, even if some other app was flashed on the peripheral), the nRF Mesh app would receive them and disconnected. This would be logged in Console app on Mac.

Can you reproduce the issue now, or does it work after restarting Bluetooth?

Kupeo, 2020-05-13:
sure, after Bluetooth restarted, the disconnect issues resolved.

But, I’m not able to catch app log, only with windows os.

closed 2020-08-28 by philips77