nordicsemi/IOS-nRF-Mesh-Library #275: Crash during proxy connection

fiveagle, 2020-05-14:
**Describe the bug**
Hi,
I have experienced a crash during the app startup while connecting to the proxy service.
The app is a project in development and I was under bad wifi network for the purpose of testing other things not related to Bluetooth

Thanks!

**Platform details:**
 - Device:  iPhone6s
 - OS: 13.4.1
 - Library Version 1.2.3

**Logs / Screenshots**
```
2020-05-14 13:03:43.176718+0200 VIEW Wireless[3057:1741536] [Bearer] Connected to 1888_594
2020-05-14 13:03:43.176874+0200 VIEW Wireless[3057:1741536] [Bearer] Discovering services...
2020-05-14 13:03:43.227496+0200 VIEW Wireless[3057:1741536] [Bearer] Connected to 1888_982
2020-05-14 13:03:43.227641+0200 VIEW Wireless[3057:1741536] [Bearer] Discovering services...
2020-05-14 13:03:43.240538+0200 VIEW Wireless[3057:1741536] [Bearer] Data In characteristic found
2020-05-14 13:03:43.240709+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out characteristic found
2020-05-14 13:03:43.240884+0200 VIEW Wireless[3057:1741536] [Bearer] Enabling notifications...
2020-05-14 13:03:43.532834+0200 VIEW Wireless[3057:1741536] [Bearer] Service found
2020-05-14 13:03:43.532985+0200 VIEW Wireless[3057:1741536] [Bearer] Discovering characteristrics...
2020-05-14 13:03:43.585592+0200 VIEW Wireless[3057:1741536] [Bearer] Service found
2020-05-14 13:03:43.585726+0200 VIEW Wireless[3057:1741536] [Bearer] Discovering characteristrics...
2020-05-14 13:03:44.418844+0200 VIEW Wireless[3057:1741536] [Bearer] Data In characteristic found
2020-05-14 13:03:44.418994+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out characteristic found
2020-05-14 13:03:44.419104+0200 VIEW Wireless[3057:1741536] [Bearer] Enabling notifications...
2020-05-14 13:03:44.448518+0200 VIEW Wireless[3057:1741536] [Bearer] Data In characteristic found
2020-05-14 13:03:44.448649+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out characteristic found
2020-05-14 13:03:44.448754+0200 VIEW Wireless[3057:1741536] [Bearer] Enabling notifications...
2020-05-14 13:03:44.681780+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out notifications enabled
2020-05-14 13:03:44.681914+0200 VIEW Wireless[3057:1741536] [Bearer] GATT Bearer open and ready
2020-05-14 13:03:44.682632+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x010100BF3E6239D61A8BA800000000E01E6D53767FA9A8
2020-05-14 13:03:44.692886+0200 VIEW Wireless[3057:1741780] [Bearer] -> 0x0223FBF2E961C894BEAE70D54081BD153CF4EE35
2020-05-14 13:03:45.640893+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x0223599C1EE37161924886CA8F22B4D10013A1A3D104
2020-05-14 13:03:45.646564+0200 VIEW Wireless[3057:1741811] [Bearer] -> 0x022308EC007C489824D9EBA9922E4519F335AEA0B8F8BD
2020-05-14 13:03:45.857235+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out notifications enabled
2020-05-14 13:03:45.857362+0200 VIEW Wireless[3057:1741536] [Bearer] GATT Bearer open and ready
2020-05-14 13:03:45.858790+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x010100BF3E6239D61A8BA800000000E01E6D53767FA9A8
2020-05-14 13:03:45.863580+0200 VIEW Wireless[3057:1741811] [Bearer] -> 0x0223D18CBFA55DE150C6BEA7304347CF97632BE0
2020-05-14 13:03:45.881501+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x0223DA6F69350E687BE9D9F1D5CFFF8A8EBFD6596101
2020-05-14 13:03:45.884477+0200 VIEW Wireless[3057:1741780] [Bearer] -> 0x022314F120B3524AF5D552D70C5CAEDB299E0C74C894F1
2020-05-14 13:03:45.887415+0200 VIEW Wireless[3057:1741536] [Bearer] Data Out notifications enabled
2020-05-14 13:03:45.887494+0200 VIEW Wireless[3057:1741536] [Bearer] GATT Bearer open and ready
2020-05-14 13:03:45.887935+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x010100BF3E6239D61A8BA800000000E01E6D53767FA9A8
2020-05-14 13:03:45.891217+0200 VIEW Wireless[3057:1741780] [Bearer] -> 0x022354FFAB2EC80B32876FC49F786B92A49C6848
2020-05-14 13:03:46.121284+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x022332CE944444F6349BBB93DD998B0ABD9504F65143
2020-05-14 13:03:46.121998+0200 VIEW Wireless[3057:1741536] [Bearer] <- 0x022397DF487E7B6F6A1EB5D9CB355B13C798D46D7FC3
Fatal error: Can't remove first element from an empty collection: file /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/swiftlang/swiftlang-1103.2.25.8/swift/stdlib/public/core/RangeReplaceableCollection.swift, line 618
2020-05-14 13:03:46.125333+0200 VIEW Wireless[3057:1741780] Fatal error: Can't remove first element from an empty collection: file /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/swiftlang/swiftlang-1103.2.25.8/swift/stdlib/public/core/RangeReplaceableCollection.swift, line 618
```

philips77, 2020-05-15:
I need more information where it crashes. The logs before Fatal error seem not related, or at least don't say what was going on there. I don't even know if it was caused by the library.

philips77, 2020-05-15:
Also, latest library version is 2.2.3, but it looks as if you just didn't update the version on bug report.

fiveagle, 2020-05-15:
You're right. I' sorry I didn't read the all log.
Anyway it crashed on ProxyFilter.swift row 342.

```Swift
let bufferEmpty = mutex.sync { buffer.isEmpty }
guard bufferEmpty else {
       let message = mutex.sync { buffer.removeFirst() }
       try? manager.send(message)
       return
}
```

It happened just once. Looking at the code it looks like it is pretty impossible to happen.
If it occurs again I'll post here.

Thanks

philips77, 2020-05-15:
Checking if buffer is empty is in a different mutex than removing the first item, so it's still possible. It should indeed be fixed.

philips77, 2020-08-28:
Hi @fiveagle, what do you think about the fix?

fiveagle, 2020-08-28:
Well I can’t upgrade to this version right now but overall it has not happened again. Ever. Reading the commit I would say this could be marked as solved, no matter what.  Thanks!

closed 2020-08-28 by philips77