nordicsemi/IOS-nRF-Mesh-Library #277: TTL for a big message

fiveagle, 2020-06-03:
Hi,

following https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/276 
we are investigating how to best use the TTL for the following use case:

We need to send a large payload (around 200 bytes) over an `AcknowledgedStaticVendorMessage`
We need to insure that the mesh does not get congested with this big message. So we considered to lower the TTL just for this message.

My questions are:
- Is the messages sizes limited somehow?
- Does setting the TTL just for the message enough to not congest the mesh
- Lowering the TTL does impact the reliability of the `AcknowledgedStaticVendorMessage` like timeout?

Thanks a lot.

philips77, 2020-06-03:
The maximum length of access layer data for segmented message is 380 bytes, when 32-bit TransMIC is used, or 376 bytes for 64-bit TransMIC. See chapter 3.7.3 or Table 3.42 in Mesh Profile 1.0.1. For vendor messages 2 additional bytes go for the Op Code, so it's 377 and 373 respectively.

TTL specifies maximum number of hops each message can be relayed before vanishing. It depends on how big network you have. A message with TTL = 0 will only be received by the connected Proxy, TTL = 1 is forbidden. If you have a lot of nodes nearby, they can all be reached with small TTL, but still each of them will try to relay. On the other hand, if you have nodes arranged in a chain, separated by long distance, long TTL is required to reach the last one.

Lowering TTL does influence timeouts, but they can be adjusted. Have a look at
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f9a09aafe091e9597b99c6d615278a681d79561c/nRFMeshProvision/Classes/MeshNetworkManager.swift#L70-L78
and `transmissionTimerInteral` and `acknowledgmentMessageInterval` in the same class.

fiveagle, 2020-06-05:
Hi,

when I use:

```swift
func send(_ message: ConfigMessage, to node: Node,
          withTtl initialTtl: UInt8? = nil) throws -> MessageHandle
```

the TTL used in the case `initialTTL == nil` is:

```swift
/// The Default TTL will be used for sending messages, if the value has
/// not been set in the Provisioner's Node. By default it is set to 5,
/// which is a reasonable value. The TTL shall be in range 2...127.
public var defaultTtl: UInt8 = 5
```

If so `defaultTtl ` can be intended as a global TTL for vendor messages send (Proxy)?
Or it is a TTL used just for provisioning actions and not for Proxy actions?

Thanks a lot.

philips77, 2020-06-05:
Default TTL is used for all messages sent by the node if not specified per message.

fiveagle, 2020-06-08:
Great. There is any similar setting also on Android?

Thanks

philips77, 2020-06-08:
Android lib doesn't allow to set TTL per messages, as far as I know.
Cc @roshanrajaratnam ?

roshanrajaratnam, 2020-06-09:
Android lib does not allow this as of now. There is an already open issue on this and will take a look at this.

fiveagle, 2020-06-10:
And what about a defaultTTL like iOS? Is anything like that on Android?
Anyway which TTL value is used to send messages at the moment?

Thank You @roshanrajaratnam

philips77, 2020-06-10:
Have a look at: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/326
Currently Android app and lib is using default TTL. In the app in can be set in Settings -> Provisioners -> Edit current provisioner -> Time to Live

philips77, 2020-08-17:
Released in 2.4.0.

closed 2020-08-17 by philips77