nordicsemi/IOS-nRF-Mesh-Library #279: Delete and reset method on MeshNetworkManager

alexisbar, 2020-06-05:
Hello, first of all thanks for this great work.

We're using the iOS and Android libraries in one of our projects. On Android, we have these methods in MeshManagerApi : 

`resetMeshNetwork()`
`deleteMeshNetworkFromDb()`

I didn't find any equivalent on iOS. Did I miss something ?
If not, would you be interested in a PR to add these two on iOS side ?

philips77, 2020-06-05:
On iOS there's:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f9a09aafe091e9597b99c6d615278a681d79561c/nRFMeshProvision/Classes/MeshNetworkManager.swift#L188-L200
The second method does not exist, as the `MeshNetworkManager` is using `Storage` protocol to keep configurations, not database. The default implementation (`LocalStorage`) can be found here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/nRFMeshProvision/Classes/Storage.swift
and is using a file. You may remove the file on your own, if needed. If you're using a custom storage implementation, you need to implement deleting on your own.

alexisbar, 2020-06-05:
Yes, I saw that. I was trying to figure out why the implementation differs between iOS and Android :

In the Android SDK, we can call `MeshManagerApi.generateMeshNetwork()`. It will be in charge to create the meshNetwork, a first provisioner and returning it to the app.

But in the iOS SDK, the logic isn't the same, as we only have a `MeshNetworkManager. createNewMeshNetwork(name: provisioner:)`. The provisioner has to be created on app side and sent to the SDK.

Is there any technical reason why the implementation between Android and iOS differs on this point ? 

I am actually implementing the reset / delete logic on app side but I thought it would be great to unify the logic between the 2 platforms, isn't it ?

We could add a generic `generateMeshNetwork()` in iOS, that could call the `createNewMeshNetwork(name:provisioner:)` internally to avoid doing this in app. Then, the reset and delete logic could be added in the Storage protocol with default implementation, in the same way than the `load()` and `save()` methods.

philips77, 2020-06-05:
The method that I pointed to above takes provisioner's name, not a provisioner. Android version just sets some default provisioner's name in this case, but I think user given one looks better and better suits me with internationalization.

philips77, 2020-06-05:
The same with Network name. Android sets a default name "Mesh Network", or similar, but iOS wants user given name.

alexisbar, 2020-06-08:
Ok, so i'll implement this on app side. Thanks for your time !

closed 2020-06-08 by alexisbar