nordicsemi/IOS-nRF-Mesh-Library #282: Identifying Device

BubblyNetDev, 2020-07-16:
can we identify a device after it is provisioned?

philips77, 2020-08-28:
Hello,
A newly provisioned Node will usually advertise for some time using *Node Identity Beacon*. 

Check Bluetooth Mesh Profile Specification 1.0.1, chapter 7.2.2.2.3: Advertising with Node Identity:
> Advertising using the Node Identity is used to identify a node based on the unicast address of the primary element and the network key of a subnet to which the node belongs. This can be useful when large amounts of data need to be delivered to a node via GATT for cases when the node cannot be easily identified or is not advertising. This advertisement may be used to initiate a GATT connection to the selected node.
>
> If PB-GATT is supported and Mesh Proxy Service is exposed, immediately after provisioning is completed using PB-GATT (see Section 5.2.2), the Mesh Proxy Service shall start advertising with Node Identity using the provisioned network.

A node may also be configured to advertise its identity later on using Config Node Identity Set message (which is currently not supported by this library).

philips77, 2020-08-28:
Also, when you send e.g. a Proxy Filter message, the returned status will contain the Unicast Address of the connected Node. So you may check what Unicast Address the connected device has, but it's not that simple to connect to a device with Unicast Address of your choice.

closed 2020-08-28 by philips77