nordicsemi/IOS-nRF-Mesh-Library #284: Publish.Retrasmit JSON export

fiveagle, 2020-08-04:
Hi,

I was debugging an issue about the scene server and I noticed that on the iOS meshNetwork json the model publish retransmit is saved as `count = 5` `interval = 150`
On the Android version (not the same mesh) it is saved as `count: 5`  `interval: 2`

My objective was to align the 2 platform. Are these values equivalent or no?
Does Android use the interval value as the iOS steps? Or is the other way around?

Thanks.

philips77, 2020-08-05:
Hello @fiveagle,

Thank you for the feedback. Are you using the latest version of nRF Mesh library for Android? I see that @roshanrajaratnam has fixed related issue here: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/328
Mind, that the JSON format is not final and will slightly change in near future (around October-December). We will make sure both libraries and sample apps are compatible and backwards compatible, meaning they will accept current JSONs but will produce new JSON only. I'm just resuming working on this project on iOS side and will create a new branch for you to test before we release a version with it. 

But before that, I'll implement heartbeats (like Roshan did on Android: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/pull/325) and perhaps scenes, depending on how fast will it take and how much time will I have.

fiveagle, 2020-08-05:
Hi,
Great news! Thank You for the quick response. 
My issue was not exactly on the json interoperability but more on the fact that setting the Retransmit with the same values we are ending having 2 different values in the json. My objective was to align the iOS configuration on Android using the json as a comparison. 
So in fact we have 2 different meshes that didn't imported each other and the results are:
`Publish.Retransmit(publishRetransmitCount: 5, intervalSteps: 2)` generates: `count = 5 interval = 150`
Instead on Android I end up having: `count: 5 interval: 2`
Are they equivalent? 
If Android express interval in seconds how can I replicate it in iOS using steps?

Thanks.

philips77, 2020-08-05:
Both should produce the same output in JSON. If you are using the latest versions of both libraries, you've found a bug. From the draft spec of JSON I see that `interval` should contain an interval in milliseconds, in range 50-1600 ms with 50ms step, which would mean that the bug is on Android side. Both JSONs will not be compatible. If the JSON was exported from Android version, try importing it there and exporting again. It should contain value 150, like the one on iOS.

fiveagle, 2020-08-06:
Unfortunately we can't update to the newest Android lib right now.
We are looking forward to update them later this year.

I'll keep this post updated.

Thanks a lot.

closed 2020-08-17 by philips77