nordicsemi/IOS-nRF-Mesh-Library #286: "No models are subscribed to this group."

jcarioti, 2020-08-20:
**Describe the bug**
I cannot seem to get a vendor model to subscribe to a group successfully. After setting up the app key, provisioning, binding, and subscribing to the group, the group says "No models are subscribed to this group." and I can't send messages to the group.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a fresh network
2. Create an app key
3. Provision device
4. Bind app key to device
5. Bind app key to model
6. Create group "New Group"
7. Subscribe model to "New Group"
8. Go to "New Group"

**Expected behavior**
I expect to be able to send messages to the group, and the model that's subscribed would receive these messages.

**Platform details:**
 - Device: iPhone 7 Plus
 - OS: iOS 13.4
 - Library Version (or Commit): 2.2.3 (from the App Store)

I might be misunderstanding the steps to create and publish messages to a group, but I don't see much in the documentation related to creating groups. Is there documentation or sample code somewhere that demonstrates this?

philips77, 2020-08-20:
Hello,
That's because in the Group tab the iOS nRF Mesh all lists only Generic On Off Server and Generic Level Server models. Currently the iOS app only those models. Additionally, on the Local Node tab it also is able to send Simple On off (Nordic proprietary vendor model). The only way to send vendor messages is be going into your model in Network tab and sending messages directly to the Node, not to the group. I think the Android app supports vendor models in the Groups tab. For iOS one we have it the backlog.

jcarioti, 2020-08-20:
Thanks for the quick response!

Does the SDK support sending messages to groups, though? Is it just the sample application that has this limitation?

philips77, 2020-08-28:
Yes, you're right.
Sorry, I thought I replied you 8 days ago, but perhaps I didn't click the Comment button.

philips77, 2020-08-28:
I added a message that clarifies that.
[image]

closed 2020-08-28 by philips77