nordicsemi/IOS-nRF-Mesh-Library #289: Randomly failing to discover services and leading to "Device not supported"

sourabhardwaj, 2020-09-09:
**Describe the bug**
Running the sample app on development kit `nRF52840-DK` (and also on our firmware board) is randomly failing to discover services leading to connection disconnect with the error message of "Device not supported"
I am basically having two but RANDOM issues:
1. Not able to provision as described above.
2. If provisioning is successful, the app is stuck in Requesting Composition Data.

**To Reproduce**
Just cloned app from the repo, made it compile and run on iPad. Tapping the + button to Provision Device. From the scanned list (2 boards for me), tapping one any board. 

**WorthNoting**
After going to iPads Bluetooth Settings', _turning the bluetooth Off and On_ makes both the above cases/issues work. But the same problem comes back after resetting / removing the node. Not sure if it anyone else have faced this problem but surely happening a lot for me.

**Expected behavior**
The sample app should be able to Provision and load Composition Data successfully **every time.**

**Platform details:**
 - Device: iPad 3
 - OS: iOS 13.3
 - App Version: 2.2.3 <master repo cloned>

**Logs / Screenshots**

**_Nordic board_**
2020-09-09 12:45:10.801432+0530 nRF Mesh[408:92785] [Bearer] Central Manager state changed to .poweredOn
2020-09-09 12:45:11.206446+0530 nRF Mesh[408:92785] [Bearer] Connecting to Dev Board...
2020-09-09 12:45:11.818963+0530 nRF Mesh[408:92785] [Bearer] Connected to Dev Board
2020-09-09 12:45:11.819245+0530 nRF Mesh[408:92785] [Bearer] Discovering services...
2020-09-09 12:45:11.962127+0530 nRF Mesh[408:92785] [Bearer] Device not supported
2020-09-09 12:45:11.962701+0530 nRF Mesh[408:92785] [Bearer] Cancelling connection...
2020-09-09 12:45:11.965448+0530 nRF Mesh[408:92785] [Bearer] Disconnected from Dev Board with error: Device not supported
2020-09-09 12:45:12.411416+0530 nRF Mesh[408:92946] [CoreBluetooth] XPC connection invalid

**_My board_**
2020-09-09 12:45:22.206242+0530 nRF Mesh[408:92785] [Bearer] Central Manager state changed to .poweredOn
2020-09-09 12:45:22.611192+0530 nRF Mesh[408:92785] [Bearer] Connecting to Ios dev...
2020-09-09 12:45:23.406476+0530 nRF Mesh[408:92785] [Bearer] Connected to Ios dev
2020-09-09 12:45:23.406650+0530 nRF Mesh[408:92785] [Bearer] Discovering services...
2020-09-09 12:45:23.611281+0530 nRF Mesh[408:92785] [Bearer] Device not supported
2020-09-09 12:45:23.611836+0530 nRF Mesh[408:92785] [Bearer] Cancelling connection...
2020-09-09 12:45:23.614616+0530 nRF Mesh[408:92785] [Bearer] Disconnected from Ios dev with error: Device not supported
2020-09-09 12:45:24.059975+0530 nRF Mesh[408:92946] [CoreBluetooth] XPC connection invalid

Please suggest what might be going wrong and how to fix this?

philips77, 2020-09-10:
This is caches issue. Your device seems not to have Service Changed characteristic enabled, so iOS caches app services. When you reset the device it changes services to support Mesh Provisioning Service, but the phone still assumes it contains Mesh Proxy Service without don't service discovery. Restarting Bluetooth clears the caches on iOS, so it begins to work again.

philips77, 2020-09-10:
Solution:
Enable Service Changed characteristic, or rotate Mac address and use a different one for unprovisioned and other when node is provisioned. Actually, you can even send each packet with a different Mac address, as Mesh does not rely on that.

sourabhardwaj, 2020-09-10:
Thank you for your response!

I had same thoughts but didn't know how to clear iOS Bluetooth Cache programatically. Rotating MAC accesses might not be ideal for our business model. Enabling Service Characteristics seems to the best. 

Do we enable Service Changed characteristic in our firmware Or is it just client setting? Or may be tracking the calls in `didUpdateValueFor characteristic`?

Kindly suggest.

philips77, 2020-09-10:
You may enable it in the firmware.
Regarding the method you mentioned, I don't really know exactly when it is called...

sourabhardwaj, 2020-09-10:
Thanks again! But I wonder how do I do it in Nordic Development Kit `nRF52840-DK` ?

philips77, 2020-09-10:
Go ahead and ask on DevZone. I have no idea. Specify which SDK are you using: nrf5 mesh SDK or nRF Connect SDK.

sourabhardwaj, 2020-09-14:
Thank you!
I have posted the [query on DevZone](https://devzone.nordicsemi.com/f/nordic-q-a/65781/randomly-failing-to-discover-services-and-leading-to-device-not-supported). I'll update this thread as soon as I get any resolution or update.
Please standby.

philips77, 2021-02-05:
As I see in the latest comment on DevZone, the issue is now closed.

closed 2021-02-05 by philips77