nordicsemi/IOS-nRF-Mesh-Library #296: How can I receive multicast? Are there any missing steps?

RengeRenge, 2020-12-01:
How can I receive multicast? Are there any missing steps?

1. Add a vendor model (id: 0x0000, companyId: 0xF018) for localElement
2. Create a group (address: 0xF000)
3. Config Model Subscription Add, bind the model and group

After I provisioning network for my mesh device, I can only control the device or receive the unicast from the device.
But I can't receive the multicast from the device (Post to 0xF000). 
By the way, sometimes I can receive multicast, but it still doesn't work after app restart.

philips77, 2020-12-02:
Ensure that group address is added to white list filter on Proxy tab. It's being added automatically on connection when any local model is subscribed to that group.

RengeRenge, 2020-12-14:
> Ensure that group address is added to white list filter on Proxy tab. It's being added automatically on connection when any local model is subscribed to that group.

thanks for your reply !
Under your reminder, I found that after the app notify ProxyService.dataOutUuid (2ADE), the mesh device immediately sent the proxyNetworkKey, and the app could not receive the characteristic update, but the device delay  send the proxyNetworkKey, it works well.
Is there something i can do?

philips77, 2021-01-19:
Sorry, I did no understand your last paragraph. Could you elaborate?

closed 2021-02-05 by philips77