nordicsemi/IOS-nRF-Mesh-Library #297: Question about Friendship support

fiveagle, 2020-12-09:
Hi,
I have seen that the 3.0 is in beta but unfortunately we can't integrate it yet.
Anyway this time I having some issues with a new device which needs to save battery. This means it is paired in friendship mode with another device that will work as a proxy while the battery one is going to ask him for messages every n seconds. 
The downside is that I cannot establish a proxy connection directly with the battery one.
We managed to make the provisioning work by enabling the functionality only after it, but we need to be able to write 1 publish message after the friendship mechanism is established. What I am experiencing is that after I send the publish message I receive a lot of messages that the SDK discards as reply attack.
I'm attaching the log. Can you please give us some hint? 

Thanks.

**Platform details:**
 - Device: iPhone8
 - OS: iOS 14
 - Library Version: 2.2.3

2020-12-09 18:55:02.906324+0100 VIEW Wireless[85526:12396572] [P406Debug] Publish...
2020-12-09 18:55:09.492060+0100 VIEW Wireless[85526:12396572] [P406Debug] Setting Model Publication...
2020-12-09 18:55:09.493444+0100 VIEW Wireless[85526:12397817] [FoundationModel] Sending ConfigModelPublicationSet(elementAddress: 189, modelIdentifier: 4096, companyIdentifier: nil, publish: 0x00B7 using App Key Index: 0, ttl: 255, flag: true, period: 0.0 sec, retransmit: Disabled) to: 00BC
2020-12-09 18:55:09.493950+0100 VIEW Wireless[85526:12397817] [Access] Sending Access PDU (opcode: 0x 3, parameters: 0xBD00B7000010FF00080010)
2020-12-09 18:55:09.495341+0100 VIEW Wireless[85526:12397817] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x65C86926F092F52A6ACAD181, transMic: 0x174ABA13) encrypted using key: 1888_980's Device Key
2020-12-09 18:55:09.496024+0100 VIEW Wireless[85526:12397817] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 1131, segO: 0, segN: 1, data: 0x65C86926F092F52A6ACAD181)
2020-12-09 18:55:09.500163+0100 VIEW Wireless[85526:12397817] [Network] Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 127, seq: 1131, src: 00AF, dst: 00BC, transportPdu: 0xC4C9CA16EA7C563521CCC7329F73A6BE, netMic: 0x40A490BF) encrypted using Primary Network Key (index: 0)
2020-12-09 18:55:09.501588+0100 VIEW Wireless[85526:12397817] [Bearer] -> 0x0062CBD12E34B0B62B6AC4C9CA16EA7C563521CCC7329F73A6BE40A490BF
2020-12-09 18:55:09.502875+0100 VIEW Wireless[85526:12397817] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 1131, segO: 1, segN: 1, data: 0x174ABA13)
2020-12-09 18:55:09.510608+0100 VIEW Wireless[85526:12397817] [Network] Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 127, seq: 1132, src: 00AF, dst: 00BC, transportPdu: 0x4A0989679A5B30C1, netMic: 0xEE974EC8) encrypted using Primary Network Key (index: 0)
2020-12-09 18:55:09.518550+0100 VIEW Wireless[85526:12397817] [Bearer] -> 0x00629E6915C500856BDB4A0989679A5B30C1EE974EC8
2020-12-09 18:55:09.932222+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x006241372752919C3A6EF45E54E013AADB3A90C836265609CD
2020-12-09 18:55:09.933250+0100 VIEW Wireless[85526:12397838] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 1, ttl: 7, seq: 11524, src: 00BC, dst: 00AF, transportPdu: 0xF45E54E013AADB, netMic: 0x3A90C836265609CD) received
2020-12-09 18:55:09.937030+0100 VIEW Wireless[85526:12397838] [LowerTransport] ACK (seqZero: 1131, blockAck: 0x00000003) receieved (decrypted using key: Primary Network Key (index: 0))
2020-12-09 18:55:11.858254+0100 VIEW Wireless[85526:12397816] [Access] Resending Access PDU (opcode: 0x 3, parameters: 0xBD00B7000010FF00080010)
2020-12-09 18:55:11.859485+0100 VIEW Wireless[85526:12397816] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0xB154DA822171026E38B49733, transMic: 0xD073BF68) encrypted using key: 1888_980's Device Key
2020-12-09 18:55:11.860203+0100 VIEW Wireless[85526:12397816] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 1133, segO: 0, segN: 1, data: 0xB154DA822171026E38B49733)
2020-12-09 18:55:11.863817+0100 VIEW Wireless[85526:12397816] [Network] Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 127, seq: 1133, src: 00AF, dst: 00BC, transportPdu: 0x5B7EFCF8F001E45E0C0B4B385E10C61E, netMic: 0x5AA1A65E) encrypted using Primary Network Key (index: 0)
2020-12-09 18:55:11.864671+0100 VIEW Wireless[85526:12397816] [Bearer] -> 0x0062C4F5B2EE019DB7B85B7EFCF8F001E45E0C0B4B385E10C61E5AA1A65E
2020-12-09 18:55:11.866053+0100 VIEW Wireless[85526:12397816] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 1133, segO: 1, segN: 1, data: 0xD073BF68)
2020-12-09 18:55:11.870982+0100 VIEW Wireless[85526:12397816] [Network] Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 127, seq: 1134, src: 00AF, dst: 00BC, transportPdu: 0xDDE24394EC49C2FB, netMic: 0x611C7C64) encrypted using Primary Network Key (index: 0)
2020-12-09 18:55:11.871602+0100 VIEW Wireless[85526:12397816] [Bearer] -> 0x00621440DFAC0A3687C9DDE24394EC49C2FB611C7C64
2020-12-09 18:55:12.812204+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x0062DB24907E45EA890ADE2B5E67FB7A8243EE1D9EF51A3C04
2020-12-09 18:55:12.813236+0100 VIEW Wireless[85526:12397816] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 1, ttl: 7, seq: 11525, src: 00BC, dst: 00AF, transportPdu: 0xDE2B5E67FB7A82, netMic: 0x43EE1D9EF51A3C04) received
2020-12-09 18:55:12.816760+0100 VIEW Wireless[85526:12397816] [LowerTransport] ACK (seqZero: 1133, blockAck: 0x00000001) receieved (decrypted using key: Primary Network Key (index: 0))
2020-12-09 18:55:12.817400+0100 VIEW Wireless[85526:12397816] [LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 1133, segO: 1, segN: 1, data: 0xD073BF68)
2020-12-09 18:55:12.821069+0100 VIEW Wireless[85526:12397816] [Network] Sending Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 127, seq: 1135, src: 00AF, dst: 00BC, transportPdu: 0xDDFA67E97C3CB1F9, netMic: 0x70DC724F) encrypted using Primary Network Key (index: 0)
2020-12-09 18:55:12.821724+0100 VIEW Wireless[85526:12397816] [Bearer] -> 0x0062B2D4AC9C224DBA3ADDFA67E97C3CB1F970DC724F
2020-12-09 18:55:13.291991+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x0062AC48A5D7613F4BF71B0F5F268E35848ACE5606526AFD25
2020-12-09 18:55:13.293153+0100 VIEW Wireless[85526:12397816] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 1, ttl: 7, seq: 11526, src: 00BC, dst: 00AF, transportPdu: 0x1B0F5F268E3584, netMic: 0x8ACE5606526AFD25) received
2020-12-09 18:55:13.296719+0100 VIEW Wireless[85526:12397816] [LowerTransport] ACK (seqZero: 1133, blockAck: 0x00000003) receieved (decrypted using key: Primary Network Key (index: 0))
2020-12-09 18:55:13.352033+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x0062CD39AB6F756E9371563EBB6776EE4D2C6B1311953DD34907204C04D4
2020-12-09 18:55:13.352732+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x006287049AE0811EE2C08A86830A335F1C762F336C0A7883
2020-12-09 18:55:13.353762+0100 VIEW Wireless[85526:12397816] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 4, seq: 67, src: 00BC, dst: 00AF, transportPdu: 0x563EBB6776EE4D2C6B1311953DD34907, netMic: 0x204C04D4) received
2020-12-09 18:55:13.354175+0100 VIEW Wireless[85526:12397817] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 4, seq: 68, src: 00BC, dst: 00AF, transportPdu: 0x8A86830A335F1C762F33, netMic: 0x6C0A7883) received
2020-12-09 18:55:13.354748+0100 VIEW Wireless[85526:12397816] [LowerTransport] Discarding packet (seqAuth: 67, expected > 11526)
2020-12-09 18:55:13.357704+0100 VIEW Wireless[85526:12397817] [LowerTransport] Discarding packet (seqAuth: 67, expected > 11526)
2020-12-09 18:55:13.531957+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x0062F705620C466DCE034C2CA91AF25E5CBBB1C28670D83D69A393AF4C37
2020-12-09 18:55:13.532966+0100 VIEW Wireless[85526:12397817] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 3, seq: 67, src: 00BC, dst: 00AF, transportPdu: 0x4C2CA91AF25E5CBBB1C28670D83D69A3, netMic: 0x93AF4C37) received
2020-12-09 18:55:13.533300+0100 VIEW Wireless[85526:12397817] [LowerTransport] Discarding packet (seqAuth: 67, expected > 11526)
2020-12-09 18:55:13.772283+0100 VIEW Wireless[85526:12396572] [Bearer] <- 0x0062F010BAB793AF59BBB2940E5237C21802B2E27F00C303
2020-12-09 18:55:13.773240+0100 VIEW Wireless[85526:12397817] [Network] Network PDU (ivi: 0, nid: 0x62, ctl: 0, ttl: 3, seq: 68, src: 00BC, dst: 00AF, transportPdu: 0xB2940E5237C21802B2E2, netMic: 0x7F00C303) received
[... cut ...]
2020-12-09 18:57:09.545696+0100 VIEW Wireless[85526:12398357] [FoundationModel] ConfigModelPublicationSet(elementAddress: 189, modelIdentifier: 4096, companyIdentifier: nil, publish: 0x00B7 using App Key Index: 0, ttl: 255, flag: true, period: 0.0 sec, retransmit: Disabled) sent from: 00AF, to: 00BC timed out
2020-12-09 18:57:09.546167+0100 VIEW Wireless[85526:12398357] [Access] Cancelling messages with op code: 3, sent from: 00AF to: 00BC
2020-12-09 18:57:09.551980+0100 VIEW Wireless[85526:12396572] [P406Debug] MeshNetworkManager failed to send Bind Message - error: Request timed out.

philips77, 2020-12-14:
Seems like you've reused the same unicast address before the IV Index was incremented by 2, so the app still remembers the last SEQ from the previous one (11k), but the new one started from 0 (on the log around 67).

fiveagle, 2020-12-15:
Thanks but how did I ended up there? I mean the app doesn't interfere directly with the SEQ management or IV index. 
Could it be something on the devices side? 
Should the app manage the IV index directly during the messages sending?
Thanks a lot.

fiveagle, 2020-12-18:
Hi,
the problem was on the device firmware side. Looks like we can close the ticket.
Thanks for your help.

closed 2020-12-18 by fiveagle