nordicsemi/IOS-nRF-Mesh-Library #299: Add a custom data interface to the node？

trusbe, 2020-12-18:
How to distinguish the node type? Lights, fans, sensors, etc., whether the node object can add a custom data type, so that users can customize node-related information according to the broadcast packet data of Unprovisioned devices, such as version number, device type, etc.

Because `Codable` is involved, it seems inconvenient to use `extension` to extend the SDK without changing the SDK file.

philips77, 2020-12-18:
Sorry, but i don't understand your question. For custom data you may include a vendor model on your node.

trusbe, 2020-12-21:
In this way, whether you can add a `deviceInfo` attribute to the` Node` class, the manufacturer can save some custom data in it, such as device type, software version, hardware version, etc.

philips77, 2020-12-21:
A Node class already had fields for that: `productIdentifier`, `versionIdentifier`, etc.
They are obtained by reading Composition Data Page 0 from the device, usually just after provisioning.

trusbe, 2020-12-22:
Oh, thank you very much, I found it.

closed 2021-01-19 by philips77