nordicsemi/IOS-nRF-Mesh-Library #300: Message discard issue

fiveagle, 2020-12-22:
Hi,
I am encountering a new problem regarding message discarding based on the seq.
A device with friendship capabilities is taking a lot of time for provisioning and message reply, it fails most of the time.
Apparently a lot of messages got discarded along the way. I'm attaching the complete log but we noticed that in one line it appears to have discarded a message with a valid seq:

```
2689 | 16:02:32.071772+0100    VIEW Wireless   Network PDU (ivi: 0, nid: 0x7D, ctl: 0, ttl: 2, seq: 183, src: 0121, dst: 011A, transportPdu: 0x5A19408E37029C616CBB02CE44AA6217, netMic: 0x4BBE2AB8) received
-- | --
16:02:32.071984+0100    VIEW Wireless   Discarding packet (seqAuth: 180, expected > 182)
```
As you can see the PDU has seq: 183 but we discard it.

Could it be something related to the friendship management?

Thanks a lot.

**Platform details:**
 - Device: iPhone8
 - OS: iOS14
 - Library Version 2.2.3

[log.txt](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/5731087/log.txt)

fiveagle, 2021-01-05:
Hello and Happy New Year!!

Did you have any opportunity to give a read to the Logs attached? Unfortunately we are still stuck with this behaviour. 
Any help is really appreciated.

Thanks.

philips77, 2021-01-05:
Hi, I'm back at the office in 10 days, but I'll try to take a look before.

fiveagle, 2021-01-22:
Hi, 
did you have any chance to give this a try?

Thanks

philips77, 2021-01-22:
Hello!
I've resume working on mesh and will look into this issue. On branch *develop* there's version 3.0-beta1. Could you give it a try? I'm fixing some bugs I would there, but the final version should be released next week or in 2 weeks.
The API has slightly changed, check out #295.

philips77, 2021-01-29:
Hi,
Did you try the new version?

fiveagle, 2021-02-03:
Hi!
Unfortunately not. We are still in the middle of a release. Hope to have some time next weeks.

philips77, 2021-11-18:
Hello!
Can we close the issue? Do you have any other issues with the library?

fiveagle, 2021-11-18:
Hi, the problem is still pending on our side but it does not happen most of the time. I’ll open a new thread with more detailed informations if it is the case possibly after we manage to update the library.

bhautikVolansys, 2022-07-07:
@philips77 

I am facing the same issue Discarding packet because of sequence.

Steps
I have provisioned the node and then exported the database.
I have imported that database and the issue come on every command.

`Network PDU (ivi: 0, nid: 0x7E, ctl: 0, ttl: 7, seq: 0, src: 0022, dst: C000, transportPdu: 0x46F5B91CE83484F23CFD9D80, netMic: 0xBAF630CA) received
Discarding packet (seqAuth: 0, expected > 77)`

philips77, 2023-10-12:
I think I figured it out. Finally, after 1.5 year. Just AFTER releasing 4.0.0.

The issue seems to happen when the library receives all segments of a segmented message and sends the ACK.
However, the ACK fails to reach the target node, so it repeats the unacknowledged segments. 
The issue is, that the library incorrectly rejects them before is able to resend the ACK. The transmitter sends them over and over and never gets expected ACK.

PR #567 (2023-10-16): Critical bug fix: Discarding packets fixed?
This PR fixes #300. It may be related to #439 and #228 or #218.

### Issue

During reassembly packets of a single segmented message may be received in arbitrary order and should be accepted. It is possible to get higher SEQ before a lower one, due to a packet loss, for example.

For that reason long time ago in lower transport layer a check has been added:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0070766bddd19adfd0f8d8957bef59fda42f6815/nRFMeshProvision/Layers/Lower%20Transport%20Layer/LowerTransportLayer.swift#L386-L399

In c [... cut]
closed 2023-10-16 by philips77