nordicsemi/IOS-nRF-Mesh-Library #301: Obtain CompositionData failed

liushuangddu, 2020-12-23:
As a *type of user*, I want < some goal > so that < some reason >.
Hi, I'm a developer from china. now I meet some wrong things. 
hopes you can give some suggests.
when I provisioned a device then the provisioner disconnect and then meshmanager connected to the deivce ,and immeditely to get the compositiondata  always failed or  needs many times retry.  but the demo is so smooth to obtain composition data, I was code my self manager class to manage the meshmanager , and I has checked many times that my code has no fault , the code is same to the demo  . the problem was sick me so many days. 
hunger for your response.

philips77, 2020-12-28:
Check if you've set the properties correctly. Here's how is fine in nRF Mesh app:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0861ceba2cb13fa0e60960cd3a818efb5b852803/Example/nRFMeshProvision/AppDelegate.swift#L47-L56

liushuangddu, 2021-01-05:
Thanks for your response. 
Of cause I did that, but the problem still exsit, I have tries many methods like that Obtain Composition Data with provisioned or first provision device and then go to obtain composition data. All always failed . the problem I don't understand is  way the demo is OK , my code is not OK , and now I plan to recode my manage class .
if you have any idea about that , hunger for your suggests.

liushuangddu, 2021-01-07:
Hi ， I have found the problem what its 。its about the seq 。when provisioning a new device the APP will record the seq and seq for a same message is increase， so I delete the APP and reinstall on the phone it will be work correct， but now  has a new problem；when I reset the node and reprovisioning the same device  the seq was cannot clear ， so how do I can  clear the seq  for the reset node.

philips77, 2021-01-19:
Hi, a unicast address cannot be reused until IV Index is incremented by 2.
If you'd check the version 3 (available still on *develop* branch), there's a *exclusion list* containing excluded addresses. It gets updated whenever you remove a node and when you provision it again, the address proposed is different than before, so SEQ numbers are calculated from 0 again. However, keep in mind, that all nodes have Replay Protection Count, which means how many SEQ number they can track. If you have too many nodes, messages from new nodes may start to be dismissed, as nodes cannot verify that they are not duplicates.

philips77, 2021-01-19:
The seq number is cleared when you remove a node, but only on the phone. Other nodes still remember the last seq sent from that unicast address. That's why you should never reuse the same address when provisioning new node. If IV Index increments by 2 (this takes at least 196h, when messages are sent very fast), the old seq is lower than new generated, so addresses can be reused again.

trusbe, 2021-01-28:
@liushuangddu  need to set `localElements`.

see: [https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Documentation/GETTING_STARTED.md](url)

Important: Even if you don't have any Models (for example only want to allow Configuration Messages in the app) you have to set the localElements property. In that case the Configuration Server and Client Models will not be properly initialized and you will be getting UnknownMessage instead of proper Status message.`

closed 2021-02-24 by liushuangddu