nordicsemi/IOS-nRF-Mesh-Library #304: Expose setting of IVIndex for MeshNetwork

kiyoshi-brilliant, 2021-01-15:
**Is your feature request related to a problem? Please describe.**
We store our IVIndex out of band of the mesh network, and use the mobile SDK only as a provisioner (not as an active mesh node). The iOS SDK seems to assume that we will obtain the current IVIndex from a Secure Network beacon after joining the Mesh network ourselves, but we initiate provisioning before joining the network (and even then only join briefly to do post provisioning configuration). 

**Describe the solution you'd like**
Expose ivIndex as publicly settable on the `MeshNetwork` class,
Expose IvIndex struct as public (as well as a publicly accessible constructor).
This seems to be the case on Android
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/blob/05ccaf31796939e8e47d138e2a91bf37c24d54bf/mesh/src/main/java/no/nordicsemi/android/mesh/MeshNetwork.java#L34

**Other Notes**
It seems android's IvIndex struct also accounts for the last transition date and whether IvRecovery is active, should these be present and accounted for in the iOS IvIndex struct and Secure Beacon processing code as well?

philips77, 2021-01-20:
Very interesting use case. Thank you for creating the issue. We're releasing 3.0 soon, so I'll implement requested feature there.

Regarding your last note, I'll discuss it with @roshanrajaratnam.

philips77, 2021-01-26:
Hello @kiyoshi-brilliant,

I just merged a PR opening the API for setting IV Index. Could you take a look this week?

Regarding your last note, on iOS I decided to keep the IV Recovery flag and transition date internal to the library. They should not be modified outside. Even the IV Index is internal and the setter just takes `Uint32` and `Bool` values.
I also added a method to restore the previous IV Index, which can be called just after loading the mesh network.

closed 2021-01-26 by philips77

kiyoshi-brilliant, 2021-01-27:
@philips77 Thanks for the update! For our use, we would prefer the recovery flag and transition date to be exposed and settable since we are also storing these out of band. During normal operation of our mesh networks (when IVUpdates are likely to occur), it is unlikely that our mobile application will be a participant (since we only join the network to do provisioning/configuration), so strong ownership by the mobile sdk of these fields doesn't really make sense for us. Recovery mode is normally entered when receiving a Secure Network Beacon, so it seems odd to conflate initialization (our intent) of the MeshNetwork instance with the mesh specified IVIndex Recovery mechanism. Could we at least add an optional last transition date to the setter, and only assume recovery mode to be true/ last transition date to be now if a value is not given?

philips77, 2021-01-27:
But why do you need to set the time or transition date, if i may ask? The whole point of this method is to allow your app to provision a new node with any IV Index without first connecting to the network to obtain the SNB. The time and iv recovery don't matter for this purpose. When you connect the phone to Gatt proxy it will get SNB with the current iv index, and if the received one is the same, it will just work. Even if you won't set iv index using this method, that would also work, as the time since last IV update would more or less match requirement. I see no need for such setters that you ask for.

kiyoshi-brilliant, 2021-01-28:
Correct me if I'm wrong but the `overwrite` function [here](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f9a09aafe091e9597b99c6d615278a681d79561c/nRFMeshProvision/Classes/Layers/Network%20Layer/SecureNetworkBeacon.swift#L119) will maintain the assigned IvIndex for the next 96 hours after calling the `setIvIndex` function (since we are setting the last transition time to `Date()`). So if we connect to GattProxy and SNB is different, SDK will ignore that SNB. e.g.:
1) Start app, set IVIndex to 2, normal operation. (for our use, we would probably provision a node before proceeding)
2) Another node initiates an IVUpdate, and sends SNB with IVIndex 3, update mode
3) App connects to gatt proxy, receives SNB with IVIndex 3, update mode (ignored because of SDK thinks last transition time < 96 hours)
4) Provision another Node A with IVIndex 2, normal operation
5) Node A will receive the SNB with IVIndex 3, update mode from elsewhere in the network
Is Node A expected to preserve its initial state of IVIndex 2, normal operation for 96 hours? or will it accept the SNB? Is provisioning an out of date IvIndex ok or to be avoided? For our use, we can call `setIvIndex` again before step 4 to avoid provisioning with an old IvIndex, but on Android we will be specifying the transition date which makes more sense to me.

philips77, 2021-01-28:
As IV Recovery flag is `true`, the 96h rule does not apply, so the IV Index can increase by 1 immediately. In point 3. your phone would transition to IV Index = 3 (update mode). However, It would still reject higher IV Indexes (even IV Index = 3 with normal operation), but it would even if you could set the transition Date.

The `setIvIndex` should only be used to set the current IV Index, or one less. If a different one is set, it will cause problems.
It's even better not to use this method if your IV Index didn't change by more than 42 and rely on IV Recovery. So let's say your app was connected to the network when IV Index was 3, but then months after that it's 15. As the transition date to IV Index = 3 is correct, it would use IV Recovery to jump to 15. Other nodes, that were provisioned with IV Index = 3 would also use IV Recovery, unless they have real time clock and were provisioned with small IV Index around the time the real IV Index was ~15. But that also cannot be solved by passing the transition date to the lib, rather by ensuring the right IV Index during provisioning.

kiyoshi-brilliant, 2021-01-28:
Ohh I didn't notice stateDiff will be 0 in recovery mode when the states are adjacent, thereby removing the 96h rule. Sounds good to me, thanks for resolving!

PR #313 (2021-01-26): Exposing setting IV Index
This PR resolves #304. It exposes API to increase the IV Index and additional one to restore the last value (normally, the last IV Index value is restored upon receiving Secure Network beacon).
The setter does some checks and makes assumptions:
* IV Index can only increase. If you set the IV Index to 123123, you'll not be able to reset it to your current 2. Not without using a hack and overriding the values in `UserDefaults`.
* When setter is used, and the IV Index is changed, the IV Recovery flag will be set to true, as the actual date when the last IV Index update h [... cut]