nordicsemi/IOS-nRF-Mesh-Library #305: Binding an application key to a localElement's model

ball-hayden, 2021-01-21:
Hi 👋 

Hopefully a quick question.

I'd like to bind an application key to one of the models I've defined in `localElements` so I can receive messages that are being broadcast by other nodes.

As far as I can tell, I can't - there's no public method to bind an application key to a model.

Am I missing something?

philips77, 2021-01-21:
You may send a config message to local node. Local node behaves as any other, as the iOS version of nRF Mesh lib supports Configuration Server model. Mind, that sending such method is required only once, as this is then saved, and is an asynchronous operation.

ball-hayden, 2021-01-22:
Ahhh. Of course.

This is working nicely.

```swift
    private func bindLocalModelApplicationKey(_ applicationKey: ApplicationKey, model: Model) throws {
        if(model.isBoundTo(applicationKey)) {
            return
        }

        let message = ConfigModelAppBind(applicationKey: applicationKey, to: model)

        try meshNetworkManager.send(message, to: model)
    }
```

closed 2021-01-22 by ball-hayden

philips77, 2021-01-22:
In version 3.0 (available in beta on *develop* branch) you'll find:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/aba91d04df8dc147295fbebdb18f73a74600e1f8/nRFMeshProvision/Classes/MeshNetworkManager.swift#L611