nordicsemi/IOS-nRF-Mesh-Library #316: Send Unacknowledged message and return UnknownMessage？

trusbe, 2021-01-28:
I created a custom pass-through Model, and Add the node device to a group address, use `Unacknowledged` to send a message, and find that it will return `UnknownMessage`. Is there a problem with my settings?

The following is my custom model:
```swift
public struct MUThroughputSetUnack: StaticVendorMessage, TransactionMessage {
    // The Op Code cosists of:
    // 0xF0-0000 - Vendor Op Code bitmask
    // 0x02-0000 - The Op Code defined by...
    // 0x00-c809 - Nordic Semiconductor ASA company ID (in Little Endian) as defined here:
    //             https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/
    public static let opCode: UInt32 = 0xF2c809
    
    public var tid: UInt8!

    public var pack: Data = Data()
    public var parameters: Data? {
        return Data() + tid + pack
    }
    
    public init(pack: Data) {
        self.pack = pack
    }
    
    public init?(parameters: Data) {
        guard parameters.count >= 1 else {
            return nil
        }
        tid = parameters[0]
        if parameters.count > 1 {
            pack = parameters.subdata(in: 1..<parameters.count)
        }
    }
}

```

Below is my print message :
```swift
2021-01-28 07:24:00 +0000 [MUGroupControlVC], send, Data: 313233343536
2021-01-28 15:24:00.200569+0800 MeshUART[2371:780392] [Model] Sending MUThroughputSetUnack(tid: Optional(207), pack: 6 bytes) from: Primary Element (0x0001), to: C000
2021-01-28 15:24:00.200735+0800 MeshUART[2371:780392] [Access] Sending Access PDU (opcode: 0xF2C809, parameters: 0xCF313233343536)
2021-01-28 15:24:00.203090+0800 MeshUART[2371:780392] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x7B164DBABF2C9B161145, transMic: 0x898DE19F) encrypted using key: AppKey 0 (index: 0)
2021-01-28 15:24:00.203219+0800 MeshUART[2371:780392] [LowerTransport] Sending Access Message (akf: 1, aid: 0x0D, szmic: 0, data: 0x7B164DBABF2C9B161145898DE19F)
2021-01-28 15:24:00.203586+0800 MeshUART[2371:780392] [Network] Sending Network PDU (ivi: 0, nid: 0x34, ctl: 0, ttl: 5, seq: 5831, src: 0001, dst: C000, transportPdu: 0x45F095AE46DDE084FB8CD0585CC873, netMic: 0xF6963365) encrypted using Primary Network Key (index: 0)
2021-01-28 15:24:00.204223+0800 MeshUART[2371:780392] [Network] Network PDU (ivi: 0, nid: 0x34, ctl: 0, ttl: 5, seq: 5831, src: 0001, dst: C000, transportPdu: 0x45F095AE46DDE084FB8CD0585CC873, netMic: 0xF6963365) received
2021-01-28 15:24:00.206319+0800 MeshUART[2371:780392] [LowerTransport] Access Message (akf: 1, aid: 0x0D, szmic: 0, data: 0x7B164DBABF2C9B161145898DE19F) received (decrypted using key: Primary Network Key (index: 0))
2021-01-28 15:24:00.206564+0800 MeshUART[2371:780392] [UpperTransport] Upper Transport PDU (encrypted data: 0x7B164DBABF2C9B161145, transMic: 0x898DE19F) received
2021-01-28 15:24:00.206704+0800 MeshUART[2371:780392] [Access] Access PDU (opcode: 0xF2C809, parameters: 0xCF313233343536) received (decrypted using key: AppKey 0 (index: 0))
2021-01-28 07:24:00 +0000 [MUGroupControlVC], didReceiveMessage, Message: UnknownMessage(opCode: 0xF2C809, parameters: CF313233343536)
2021-01-28 15:24:00.207465+0800 MeshUART[2371:780392] [Bearer] -> 0x0034853530A895F62BC745F095AE46DDE084FB8CD0585CC873F6963365
2021-01-28 07:24:00 +0000 [MUGroupControlVC], didSendMessage, Message: MUThroughputSetUnack(tid: Optional(207), pack: 6 bytes)

```

**Platform details:**
 - Device: iPhone12 Pro Max
 - OS: iOS 14.4
 - Library Version (or Commit)：3.0.0

philips77, 2021-01-28:
The message you receive goes through loopback mechanism, it is sent to a Group Address, and your local Models may also be subscribed to it. The Network Layer does not check if any local Model is subscribed, or not. Have a look at:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bc05b9af80b4a678e91f873943c8f89894c08f47/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L160-L170
where the `shouldLoopback` method looks like:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bc05b9af80b4a678e91f873943c8f89894c08f47/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L430-L436

The message you receive is returned as `UnknownMessage` as you have no local Model that could parse it. If you'd like to have the response parsed (not for that Unacknowledged message, but some other perhaps), you need to add your Client model to `localElements` like such:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bc05b9af80b4a678e91f873943c8f89894c08f47/Example/nRFMeshProvision/AppDelegate.swift#L127-L130
Here's an example Vendor Client model:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/develop/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff/SimpleOnOffClientDelegate.swift#L49
Even if you don't have any Ack messages, I'd recommend adding such, so other nodes could know that you may send such messages.

trusbe, 2021-01-28:
@philips77 

Thanks for the reply. 
My question is mainly about sending the Unacknowledged message. According to the document, there should be no reply, right? The same Unacknowledged message, I use `MeshNetworkManager.instance.send(<#T##message: MeshMessage##MeshMessage#>, to: <#T##Model#>)` to send, it is normal, there is no reply;

But using `MeshNetworkManager.instance.send(<#T##message: MeshMessage##MeshMessage#>, to: <#T##Group#>, using: <#T##ApplicationKey#>)` sent, but got a reply information.

The UnknownMessage mentioned in the question, I know the problem.

Thank you very much for reviewing. 

[image]

philips77, 2021-01-28:
Whether a message is acknowledged or not is just an agreement between client and server. If your server doesn't send ack for that op code, it's unack. In the lib api you just have it as you do. For Ack-ed you'd provide the op code of expected response.

philips77, 2021-01-29:
> But using MeshNetworkManager.instance.send(<#T##message: MeshMessage##MeshMessage#>, to: <#T##Group#>, using: <#T##ApplicationKey#>) sent, but got a reply information.

You don't get the ACK, at least that's not what's on the logs you provided. The message is sent to the Bearer:
```swift
2021-01-28 15:24:00.207465+0800 MeshUART[2371:780392] [Bearer] -> 0x0034853530A895F62BC745F095AE46DDE084FB8CD0585CC873F6963365
```
and there's no anything incoming that follows that message.
The `UnknownMessage` you receive is the same message that you send. As it is sent to a Group Address, it may also be processed by a model on the sending device, so via loopback it goes back, first going to all the layers down, then up.

Should you receive a message, it would be shown as `[Bearer] <- 0x...`

trusbe, 2021-02-01:
Got it, thanks for reply，Very useful to me。 @philips77

closed 2021-02-05 by philips77