nordicsemi/IOS-nRF-Mesh-Library #320: Time Set

fiveagle, 2021-02-05:
Hi,
I'm doing some preliminary research on a new feature I need to implement. We need to send a Time Set Message (opCode 0x5C). 
I see the library only includes `GenericDefaultTransitionTimeSet` but it doesn't look quite the same since the fields are different.

From the SIG Mesh Model specification:
[image]

How can we overcome this? Is this message implemented on version 3?

Thanks a lot.

philips77, 2021-02-05:
Hello,
There is currently no TimeSet message implemented. I'll be releasing 3.0 today, so there's no time to add it now, but we plan having 3.1 very soon afterwards.
If you implement it, consider creating a PR so we will gladly accept it!

fiveagle, 2021-02-05:
> If you implement it, consider creating a PR so we will gladly accept it!
Undoubtedly it's something we can consider after updating to 3.0 but we also need to keep iOS and Android in sync. 
Eventually I'll ask questions on this thread.

philips77, 2022-05-20:
Let me just reference #423 here, as Time Set was mentioned there.

philips77, 2023-06-16:
Closed with #443, long time ago.

closed 2023-06-16 by philips77