nordicsemi/IOS-nRF-Mesh-Library #321: An error thrown when import mesh network

KcLeung514, 2021-02-09:
**Describe the bug**
An error thrown when import mesh network. 
I tried import it at nRF Mesh (Android), it restored the mesh network, and able to control the devices.

**To Reproduce**
Steps to reproduce the behavior:
1. Export the mesh network file from Android-nRF-Mesh-Library
2. Launch nRF Mesh in iOS
3. Go to "Settings"
4. Click "Forget Network"
5. Click the folder icon at the right top
6. Select "Import"
7. Select "mesh network file" from google drive
8. See error

**Expected behavior**
Expect import without error at nRF Mesh (iOS).

**Platform details:**
 - Device: iPhone 6s
 - OS: 14.4
 - nRF Mesh ( iOS, version 3.0 )

**Logs / Screenshots**

[image]

meshNetwork.json
```
{
  "$schema": "http://json-schema.org/draft-04/schema#",
  "id": "TBD",
  "version": "1.0",
  "meshUUID": "4E691A13C53246AF9A37121CC330ACD9",
  "meshName": "nRF Mesh Network",
  "timestamp": "1970-01-01T08:00:00+0800",
  "netKeys": [
    {
      "name": "Network Key 1",
      "index": 0,
      "key": "301690B5CEE2A1CB4CC02B54E8D7AD98",
      "phase": 0,
      "minSecurity": "high",
      "timestamp": "1970-01-01T08:00:00+0800"
    }
  ],
  "appKeys": [
    {
      "name": "Application Key 1",
      "index": 0,
      "boundNetKey": 0,
      "key": "7019C176818FFA7F1122D988A442F7DA"
    },
    {
      "name": "Application Key 2",
      "index": 1,
      "boundNetKey": 0,
      "key": "F044A26D525AB4DBDE3AD8E751379C6F"
    },
    {
      "name": "Application Key 3",
      "index": 2,
      "boundNetKey": 0,
      "key": "78C935D8BEF8C3D56846E3DB08F19E8B"
    }
  ],
  "provisioners": [
    {
      "provisionerName": "FBB738A0-6A0D-11EB-A877-F787D8B17222",
      "UUID": "FBB738A06A0D11EBA877F787D8B17222",
[... 480 lines cut ...]
              "modelId": "1305",
              "bind": [],
              "subscribe": []
            },
            {
              "modelId": "1309",
              "bind": [],
              "subscribe": []
            }
          ]
        }
      ],
      "blacklisted": false
    }
  ],
  "groups": [
    {
      "name": "Test",
      "address": "C000",
      "parentAddress": "0000"
    }
  ],
  "scenes": []
}
```

philips77, 2021-02-09:
Oh, thanks for repotting the issue. 
Please, change the TBD in your JSON to
```
http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#
```
I'll push 3.0.1 later today with other bug fixes.

KcLeung514, 2021-02-09:
After change the meshNetwork id from `TBD` to `http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#` , it can be imported correctly at nRF Mesh (iOS).

Thank you. @philips77 :)

KcLeung514, 2021-02-11:
**Describe the bug**
An error thrown when import mesh network.
I tried import it at nRF Mesh (Android, version 3.0.0 ), it restored the mesh network, and able to control the devices.
The issue is about Publication Settings, if disable **publish period** is not allowed in iOS version, what is the recommended publication settings?

**To Reproduce**
Steps to reproduce the behavior:

Export the mesh network file from Android-nRF-Mesh-Library ( version 3.0.1 )
Launch nRF Mesh in iOS
Go to "Settings"
Click "Forget Network"
Click the folder icon at the right top
Select "Import"
Select "mesh network file" from google drive
See error

**Expected behavior**
Expect import without error at nRF Mesh (iOS).

**Platform details:**

Device: iPhone 6s
OS: 14.4
nRF Mesh ( iOS, version 3.0.1 )

**Logs / Screenshots**

[image]

meshNetwork.json
```
{
  "$schema": "http://json-schema.org/draft-04/schema#",
  "id": "http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#",
  "version": "1.0",
  "meshUUID": "A22D391F-1702-4A1F-A354-1927DE40BDE5",
  "meshName": "nrf Mesh Network",
  "timestamp": "1970-01-01T08:00:00+0800",
  "partial": false,
  "netKeys": [
    {
      "name": "Network Key 1",
      "index": 0,
      "key": "647D75020E1097A534093588A2196419",
      "phase": 0,
      "minSecurity": "insecure",
      "timestamp": "1970-01-01T08:00:00+0800"
    }
  ],
  "appKeys": [
    {
      "name": "Application Key 1",
      "index": 0,
      "boundNetKey": 0,
      "key": "76156CBB328597A234894837520FB28D"
    },
    {
      "name": "Application Key 2",
      "index": 1,
      "boundNetKey": 0,
      "key": "365DEA77F3C4AE6D45EA0DF63B48D69E"
    },
    {
      "name": "Application Key 3",
      "index": 2,
      "boundNetKey": 0,
      "key": "08851694D284D21389CA6FB8D2085D52"
    }
  ],
[... 362 lines cut ...]
          "index": 2,
          "location": "0000",
          "models": [
            {
              "modelId": "1000",
              "bind": [],
              "subscribe": []
            }
          ]
        }
      ],
      "excluded": false
    }
  ],
  "groups": [
    {
      "name": "Observable",
      "address": "C000",
      "parentAddress": "0000"
    }
  ],
  "scenes": [],
  "networkExclusions": []
}
```

philips77, 2021-02-11:
First of all, I love that you do testing! Seems like you found another issue with either of the apps. I'll have a look just now.

philips77, 2021-02-11:
So, in here:
```json
"publish": {
      "address": "C000",
      "index": 0,
      "ttl": 255,
      "period": {
         "numberOfSteps": 0,
         "resolution": 0
       },
       "retransmit": {
         "count": 1,
         "interval": 100
       },
       "credentials": 0
}
```
the `resolution` should be 100. nRF Mesh for Android should export 100, but I think iOS app will also need to be updated to accept 0 and migrate it quietly to 100. Expect 3.0.2 today.
If you could found any other issues before the release, it would be great!

philips77, 2021-02-11:
Thank you again for reporting the issues.
We're just releasing 3.0.2 for both Android and iOS, expect them to be on stores tomorrow.
Library releases for 3rd party apps have just been published.

Hopefully, all issues were resolved now. Should you find a new one, please create a new issue and we will handle it!

closed 2021-02-11 by philips77