nordicsemi/IOS-nRF-Mesh-Library #326: Sometimes cannot receive responses via proxy filter

alexlam112, 2021-02-17:
Hi. I am using the latest version of the library(3.0.2) and working on syncing devices’ status for all provisioners using proxy filter. The following is the flow:
  1. Setup publication for device
  2. Enable inclusion list proxy filter every time connected to a device
  3. Send unacknowledged message
  4. Device receive the unacknowledged message and publish the updated status
  5. Receive updated status via proxy filter

The problem I am facing is that sometimes the provisioner cannot receive the updated status via the proxy filter. Do you have any suggestions? Thanks!

Mesh Network json file: 
```
{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "appKeys": [
        {
            "boundNetKey": 0,
            "index": 0,
            "key": "3D0394C1AB542F5AC2B9A9DEB4A92672",
            "name": "control"
        }
    ],
    "groups": [
        {
            "address": "C000",
            "name": "Observable",
            "parentAddress": "0000"
        }
    ],
    "id": "http://www.bluetooth.com/specifications/assigned-numbers/mesh-profile/cdb-schema.json#",
    "meshName": "Mesh Network",
    "meshUUID": "D9F692C3-75A6-47C1-BBC1-28FFC441181F",
    "netKeys": [
        {
            "index": 0,
            "key": "DA455BAC3B96BEE096A11CAD26BB9756",
            "minSecurity": "secure",
            "name": "Primary Network Key",
            "phase": 0,
            "timestamp": "2021-02-17T09:31:28Z"
        },
        {
            "index": 1,
            "key": "C49BBC956B7179B94F97DC5EF9C14AE8",
            "minSecurity": "secure",
            "name": "main",
            "phase": 0,
            "timestamp": "2021-02-17T09:31:28Z"
        }
    ],
    "networkExclusions": [
        {
            "addresses": [
                "0001"
            ],
            "ivIndex": 0
        }
    ],
    "nodes": [
        {
            "appKeys": [
                {
                    "index": 0,
                    "updated": false
                }
            ],
            "configComplete": true,
            "crpl": "7FFF",
            "deviceKey": "F209E9761052403A9514133E678ADA44",
            "elements": [
                {
[... 310 lines cut ...]
                    "lowAddress": "CC9B"
                }
            ],
            "allocatedSceneRange": [
                {
                    "firstScene": "3334",
                    "lastScene": "3B33"
                }
            ],
            "allocatedUnicastRange": [
                {
                    "highAddress": "1D9A",
                    "lowAddress": "199B"
                }
            ],
            "provisionerName": "9B807820-70ED-11EB-AE14-43E1168A7765",
            "UUID": "9B807820-70ED-11EB-AE14-43E1168A7765"
        }
    ],
    "scenes": [],
    "timestamp": "2021-02-17T09:32:14Z",
    "version": "1.0.0"
}

```

philips77, 2021-02-17:
It may be that the status didn't reach the Proxy Node. You may adjust *network transmit* and *relay retransmit* parameters no the node and relays in between to increase the chance that the status is properly relayed.

alexlam112, 2021-02-18:
Thank you very much for your reply! I tried your suggestion but it is not working right now. I will have a try on more combinations to see if it can be the solution.
I would like to provide more details as I think the provided information is not quite clear to explain my situation and I am sorry about that. Let me elaborate more on my whole setup:
1. Got **more than one** mobile devices as provisioners, lets say two i.e. **iPhone SE2** & **Phone B(iOS or Android)**
2. Got **one ble device**
3. The network is created on **iPhone SE2**, and imported it on **Phone B**

The case is that if I send an unacknowledged message (e.g. GenericOnOffSetUnacknowledged) to the **ble device**, the device received the message and **did update its own status (e.g. turned on/off)**. 
For the publication part, if I use an **Android** device the status message can be received successfully every time. 
But for **iOS** devices, it is strange that sometimes one can receive the status message but another one cannot or both cannot receive the status message. And If I reconnect to the network, it may works sometimes.

That is my whole setup and the issue. Any other suggestions?
Thanks!

philips77, 2021-02-18:
If the status message is being sent to a group address (I thing `C000` in your case), ensure that that address is added to Proxy Filter. Also, ensure that the Proxy accepted it. Here, I present the logs when I sent 5 addresses, but proxy reports only 3:
[image]

alexlam112, 2021-02-22:
Hi, @philips77 . Thank you for you reply. I did set the proxy filter and it did accepted as well. I printed the proxy filter every time I send message in order to ensure the proxy filter is already being set. That's why I have no idea why I cannot receive the status message sometimes.

philips77, 2021-05-21:
Hi, did you manage to fix your issue?

alexlam112, 2021-06-08:
Hi, @philips77. I found that the proxyNetworkKey is not set yet when I try to send the ProxyConfigurationMessages. I did add a 2 second timeout before send the messages. Do you have any ideas or suggestions? Thanks

philips77, 2021-06-08:
The proxy filter is updated automatically here
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/7f1ed6104739a1ff8a8c7dfa605c55e506690621/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L358-L359
after a Secure Network Beacon is received. This method sets the `proxyNetworkKey`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/7f1ed6104739a1ff8a8c7dfa605c55e506690621/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L362-L384
The proxy filter initially will contain Unicast Addresses of all Elements, a All Nodes group address and all group addresses for which at least one model on any Element is subscribed for.
If you want to add more, I think you may do this in the `ProxyFilterDelegate`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/7f1ed6104739a1ff8a8c7dfa605c55e506690621/nRFMeshProvision/Classes/ProxyFilter.swift#L46-L52

alexlam112, 2021-06-08:
Thanks for your reply. Yes, I did the similar things and was working fine. But now it is strange that the proxy filter won't be updated automatically after I connect to the BLE mesh device so the `ProxyFilterDelegate` never being reached. And I tried sending the `ProxyConfigurationMessages` with a few seconds delay after connected to the BLE mesh device, the `Addresses` set from the `proxyFilterUpdated` in `ProxyFilterDelegate` remains empty. And I tried that it is working fine on Android with the same BLE mesh device. I have no idea what is going on...

alexlam112, 2021-06-08:
Seems that I cannot get the Secure Network Beacon after connecting to the proxy node. 

```
2021-06-08 18:22:19.513946+0800 Runner[2573:1773420] [bleMesh] Loading mesh network
2021-06-08 18:22:19.564211+0800 Runner[2573:1773420] [bleMesh] Initialised mesh network
2021-06-08 18:22:19.570619+0800 Runner[2573:1773420] [bleMesh] Scanning for peripherals
2021-06-08 18:22:19.652381+0800 Runner[2573:1773420] [bleMesh] Did discover peripheral - WiZ98302a
2021-06-08 18:22:19.653085+0800 Runner[2573:1773420] [bleMesh] Opening connection to peripheral
2021-06-08 18:22:19.659013+0800 Runner[2573:1773420] [Bearer] Central Manager state changed to .poweredOn
2021-06-08 18:22:19.659289+0800 Runner[2573:1773420] [Bearer] Connecting to WiZ98302a...
2021-06-08 18:22:21.383445+0800 Runner[2573:1773420] [Bearer] Connected to WiZ98302a
2021-06-08 18:22:21.384049+0800 Runner[2573:1773420] [Bearer] Discovering services...
2021-06-08 18:22:21.526303+0800 Runner[2573:1773420] BleManager: startScan succeed, scanning...
2021-06-08 18:22:21.591844+0800 Runner[2573:1773420] [bleMesh] Searching for unprovisioned devices
2021-06-08 18:22:21.598019+0800 Runner[2573:1773420] [bleMesh] Scanning for peripherals
2021-06-08 18:22:21.982475+0800 Runner[2573:1773420] [Bearer] Service found
2021-06-08 18:22:21.982544+0800 Runner[2573:1773420] [Bearer] Discovering characteristics...
2021-06-08 18:22:22.131202+0800 Runner[2573:1773420] [Bearer] Data In characteristic found
2021-06-08 18:22:22.131297+0800 Runner[2573:1773420] [Bearer] Data Out characteristic found
2021-06-08 18:22:22.131405+0800 Runner[2573:1773420] [Bearer] Enabling notifications...
2021-06-08 18:22:22.310555+0800 Runner[2573:1773420] [Bearer] Data Out notifications enabled
2021-06-08 18:22:22.310638+0800 Runner[2573:1773420] [Bearer] GATT Bearer open and ready
2021-06-08 18:22:28.807923+0800 Runner[2573:1773420] [bleMesh] Update proxy filter type...
2021-06-08 18:22:28.811140+0800 Runner[2573:1773420] [bleMesh] Proxy filter updated - type: (
    "Inclusion List"
), addresses: (
    "{(\n)}"
)
2021-06-08 18:22:29.874606+0800 Runner[2573:1773420] [bleMesh] Add some addresses to proxy filter...
2021-06-08 18:22:29.875063+0800 Runner[2573:1773420] [bleMesh] Completed schedule task once
2021-06-08 18:22:29.875177+0800 Runner[2573:1773420] [bleMesh] Schedule task status: 1/1
2021-06-08 18:22:29.876309+0800 Runner[2573:1773420] [bleMesh] Proxy filter updated - type: (
    "Inclusion List"
), addresses: (
    "{(\n)}"
)
2021-06-08 18:22:30.941212+0800 Runner[2573:1773420] [bleMesh] Add some addresses to proxy filter...
2021-06-08 18:22:30.941397+0800 Runner[2573:1773420] [bleMesh] Completed schedule task once
2021-06-08 18:22:30.941493+0800 Runner[2573:1773420] [bleMesh] Schedule task status: 2/1
2021-06-08 18:22:30.942005+0800 Runner[2573:1773420] [bleMesh] Proxy filter updated - type: (
    "Inclusion List"
), addresses: (
    "{(\n)}"
)
2021-06-08 18:22:30.960093+0800 Runner[2573:1773420] [bleMesh] Sending message to mesh address (
    "ble_mesh.VendorModelMessageAcked"
)
2021-06-08 18:22:30.962038+0800 Runner[2573:1773680] [Model] Sending ble_mesh.VendorModelMessageAcked from: PrimaryElement (0x0001), to: F001
2021-06-08 18:22:30.962625+0800 Runner[2573:1773680] [Access] Sending Access PDU (opcode: 0xD01706, parameters: 0x3B00010501)
2021-06-08 18:22:30.970511+0800 Runner[2573:1773680] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x668892B748B33A74, transMic: 0xB1A73F16) encrypted using key: control (index: 0)
2021-06-08 18:22:30.971715+0800 Runner[2573:1773680] [LowerTransport] Sending Access Message (akf: 1, aid: 0x3B, szmic: 0, data: 0x668892B748B33A74B1A73F16)
2021-06-08 18:22:30.977732+0800 Runner[2573:1773680] [Network] Sending Network PDU (ivi: 0, nid: 0x78, ctl: 0, ttl: 5, seq: 16, src: 0001, dst: F001, transportPdu: 0xEBA8A5A56859C469D484A096D0, netMic: 0x193435CD) encrypted using Primary Network Key (index: 0)
2021-06-08 18:22:30.982723+0800 Runner[2573:1773680] [Network] Network PDU (ivi: 0, nid: 0x78, ctl: 0, ttl: 5, seq: 16, src: 0001, dst: F001, transportPdu: 0xEBA8A5A56859C469D484A096D0, netMic: 0x193435CD) received
2021-06-08 18:22:30.984570+0800 Runner[2573:1773680] [LowerTransport] Access Message (akf: 1, aid: 0x3B, szmic: 0, data: 0x668892B748B33A74B1A73F16) received (decrypted using key: Primary Network Key (index: 0))
2021-06-08 18:22:30.987124+0800 Runner[2573:1773680] [UpperTransport] Upper Transport PDU (encrypted data: 0x668892B748B33A74, transMic: 0xB1A73F16) received
2021-06-08 18:22:30.987200+0800 Runner[2573:1773680] [Access] Access PDU (opcode: 0xD01706, parameters: 0x3B00010501) received (decrypted using key: control (index: 0))
2021-06-08 18:22:30.987797+0800 Runner[2573:1773420] [bleMesh] Recieved message - (
    "UnknownMessage(opCode: 0xD01706, parameters: 3B00010501)"
)
2021-06-08 18:22:30.987866+0800 Runner[2573:1773680] [Bearer] -> 0x407894D95EAEF1693F0AEBA8A5A56859C469D484
2021-06-08 18:22:30.988212+0800 Runner[2573:1773420] [bleMesh] Did Sent message (
    "ble_mesh.VendorModelMessageAcked"
)
2021-06-08 18:22:30.988559+0800 Runner[2573:1773420] [Bearer] -> 0xC0A096D0193435CD
```

philips77, 2021-06-08:
Check your firmware. A proxy node should send one Secure Network Beacon per each Network Key it knows just after enabling notifications.
You may also try using e.g. nRF Connect and check if you get something (it will not be decoded) if you don't trust your app. 

On Android the nRF Mesh is using the Primary Network Key by default, so it would also work with a device not sending SNB at al. Is that correct, @roshanrajaratnam?

roshanrajaratnam, 2021-06-08:
@philips77 correct, we stick to the primary network key by default.

alexlam112, 2021-06-10:
Hi, @philips77. Thank you for your reply. I tried using the nRF connect app to scan for the Secure Network Beacon and found that the device did sent the beacon but my app just cannot receive/recognise it. Is it possible that I use the `ConfigBeaconGet` message to get the SNB and update the `proxyNetworkKey`?

philips77, 2021-06-10:
That's weird. An incoming data should be logged just after:

      2021-06-08 18:22:22.310638+0800 Runner[2573:1773420] [Bearer] GATT Bearer open and ready

with <- marking. 
Did you try nRF Mesh app? Does it work there? Perhaps it's something with wrong with your setup.

alexlam112, 2021-06-10:
Yes. I tried using the nRF Mesh app. The BLE mesh device can be provisioned but the proxy filter can neither change proxy filter type nor add address to the list (remain unchanged after loading).

philips77, 2021-06-10:
Could you paste the logs from nRF Mesh?

alexlam112, 2021-06-10:
Sure. I tried adding address to the inclusion list twice but failed and no any extra logs printed.
```
2021-06-10 16:58:19.419229+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Cancelling connection...
2021-06-10 16:58:19.420677+0800 nRFMeshProvision_Example[3149:2109477] [CoreBluetooth] XPC connection invalid
2021-06-10 16:58:19.420776+0800 nRFMeshProvision_Example[3149:2109477] [CoreBluetooth] XPC connection invalid
2021-06-10 16:58:19.423683+0800 nRFMeshProvision_Example[3149:2104132] Unbalanced calls to begin/end appearance transitions for <nRFMeshProvision_Example.ConfigurationViewController: 0x1024430b0>.
2021-06-10 16:58:37.514345+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Central Manager state changed to .poweredOn
2021-06-10 16:58:37.919599+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Connecting to WiZ983198...
2021-06-10 16:58:38.241039+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Connected to WiZ983198
2021-06-10 16:58:38.241188+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Discovering services...
2021-06-10 16:58:38.718155+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Service found
2021-06-10 16:58:38.718236+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Discovering characteristics...
2021-06-10 16:58:38.838346+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Data In characteristic found
2021-06-10 16:58:38.838418+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Data Out characteristic found
2021-06-10 16:58:38.838530+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Enabling notifications...
2021-06-10 16:58:39.017749+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] Data Out notifications enabled
2021-06-10 16:58:39.017840+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] GATT Bearer open and ready
2021-06-10 16:58:39.292553+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Requesting Provisioning Capabilities
2021-06-10 16:58:39.292596+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Sending Provisioning Invite (attention timer: 5 sec)
2021-06-10 16:58:39.292659+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x030005
2021-06-10 16:58:39.347764+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0x03010100010001000000000000
2021-06-10 16:58:39.347873+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Device Capabilities: Number of elements: 1
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: Static OOB Information Available
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
2021-06-10 16:58:39.347953+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Provisioning Capabilities received
2021-06-10 16:58:54.210624+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Provisioning started
2021-06-10 16:58:54.210855+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: Static OOB)
2021-06-10 16:58:54.211238+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x03020000010000
2021-06-10 16:58:54.212156+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Sending Provisioner Public Key (0x92041E47ABE3A5790BAB15BE47207DABA1A7B139A82C740179750BE80901E52CA8AD7938A272A7B1BB8C9B5BCD3EB7330D752DAAAAFF5A0CA714428E86701A38)
2021-06-10 16:58:54.212469+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x430392041E47ABE3A5790BAB15BE47207DABA1A7
2021-06-10 16:58:54.214126+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x83B139A82C740179750BE80901E52CA8AD7938A2
2021-06-10 16:58:54.214623+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x8372A7B1BB8C9B5BCD3EB7330D752DAAAAFF5A0C
2021-06-10 16:58:54.215285+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0xC3A714428E86701A38
2021-06-10 16:58:54.468238+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0x430341CC8B27CD3A85D7BC1D46F805E18257EE34
2021-06-10 16:58:54.498797+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0x834DC6FB6C9979D042A71B145035B84D0AEB0C3F
2021-06-10 16:58:54.499108+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0x835A4DFE1324DF1CDE0F4093B1F07E66D65E7681
2021-06-10 16:58:54.499296+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0xC30868D3382C4F56F4
2021-06-10 16:58:54.499809+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Device Public Key (0x41CC8B27CD3A85D7BC1D46F805E18257EE344DC6FB6C9979D042A71B145035B84D0AEB0C3F5A4DFE1324DF1CDE0F4093B1F07E66D65E76810868D3382C4F56F4) received
2021-06-10 16:59:12.016764+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Sending Provisioner Confirmation (0xAE8011CDB8602267D2842ABB7A546A10)
2021-06-10 16:59:12.016906+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x0305AE8011CDB8602267D2842ABB7A546A10
2021-06-10 16:59:12.077433+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] <- 0x0305CF03BA1A8A4C8382E17A28222413E549
2021-06-10 16:59:12.077553+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Device Confirmation (0xCF03BA1A8A4C8382E17A28222413E549) received
2021-06-10 16:59:12.077598+0800 nRFMeshProvision_Example[3149:2104132] [Provisioning] Sending Provisioner Random (0x868593A241D2026F04E2ABE46BA32809)
2021-06-10 16:59:12.077685+0800 nRFMeshProvision_Example[3149:2104132] [Bearer] -> 0x0306868593A241D2026F04E2ABE46BA32809
[... cut ...]
2021-06-10 16:59:19.423632+0800 nRFMeshProvision_Example[3149:2109833] [UpperTransport] Upper Transport PDU (encrypted data: 0x872B5B8DC0E3, transMic: 0x7E1BC023) received
2021-06-10 16:59:19.424074+0800 nRFMeshProvision_Example[3149:2109833] [Access] Response Access PDU (opcode: 0x8003, parameters: 0x00000000) received (decrypted using key: WiZ983198's Device Key)
2021-06-10 16:59:19.424477+0800 nRFMeshProvision_Example[3149:2109833] [FoundationModel] ConfigAppKeyStatus(networkKeyIndex: 0, applicationKeyIndex: 0, status: Success) received from: 0003
```

philips77, 2021-06-11:
Which SDK are you using, nRF SDK for Mesh, or NCS?

philips77, 2021-06-11:
Spec says: 

> Upon connection, the Proxy Server shall send a Secure Network Beacon for each known subnet to the Proxy Client.

philips77, 2021-12-16:
Hello,
Did you manage to solve your problem? Can we close the issue now?

philips77, 2023-10-17:
Closing the issue due to long time of inactivity.

closed 2023-10-17 by philips77