nordicsemi/IOS-nRF-Mesh-Library #330: Friendship node reset

fiveagle, 2021-02-24:
Hi,
We have different behaviours when send by app a software reset to a BLE mesh device (using Mesh SDK 4.0.0) from Android platform as compared to IOS platform.
The side effect is that with a IOS platfom a dirty reset happens (a mesh provisioned device doesn’t clear its configuration).
Debugging the device firmware when a software reset is sent from IOS app, we noticed that the event NRF_MESH_EVT_FRIENDSHIP_TERMINATED in the app_mesh_core_event_cb() happens before the real reset callback function config_server_evt_cb().
Instead, debugging the device firmware when a software reset is sent from ANDROID app only the real reset callback function config_server_evt_cb() is processed.
We need that the reset callback function config_server_evt_cb() is processed before app_mesh_core_event_cb() caused by NRF_MESH_EVT_FRIENDSHIP_TERMINATED because here we activate the proxy and it seems a problem for the reset procedure.
Unfortunately we have just produced and sold some devices and at the moment we don’t want sort the problem out in the fw device, just because there is a button on the device that reset itself.
Could you explain the reason of such different IOS behaviour?
Is there a way to sort the problem out in the IOS app?

Thanks a lot.

**Platform details:**
 - Library Version 2.2.3

philips77, 2021-02-25:
As far as I understood, you're sending `ConfigNodeReset` message, which is very short, so should be sent in a single packet even with MTU = 23, so MTU or DLE should not make any difference. Did you compare connection intervals?
Is it 100% of time that Android does the job correctly, and iOS messes up? 
Are you sending the reset message through a GATT proxy in both cases, or are you resetting the node you're connected to? Or perhaps, it works when reset is sent through proxy, and fails when connected to the reseted node?
Did you collect any logs from both platforms and compared them?

philips77, 2021-02-25:
One suggestion from our SDK expert: 
From the iOS app try sending the Node reset message multiple times (at least 2 or 3 times), once in each connection interval or several times in the same connection interval. This should force the stack to execute node reset anyway.

And a question: 
Is this happening only when the node is in active Friendship and other wise works fine?

fiveagle, 2021-02-26:
Hi,

> Is this happening only when the node is in active Friendship and other wise works fine?

Only in Friendship

> Did you compare connection intervals?

No

> Is it 100% of time that Android does the job correctly, and iOS messes up?

Yes

> Are you sending the reset message through a GATT proxy in both cases, or are you resetting the node you're connected to?

Always proxy

> Did you collected any logs from both platforms and compared them?

Yes we collect Logs from the device during the same operation from both apps

> From the iOS app try sending the Node reset message multiple times (at least 2 or 3 times), once in each connection interval or several times in the same connection interval. This should force the stack to execute node reset anyway.

What do you mean with "connection interval"? I use basically the same code as in the sample app which worked great with every device till now.

Thanks a lot!

closed 2021-04-15 by philips77