nordicsemi/IOS-nRF-Mesh-Library #332: How to receive the message published by the node on the mobile phone?

trusbe, 2021-03-01:
Hello, may the mobile phone and the node be in the same network. The node will actively publish messages to the network. The mobile phone needs to receive these messages. What should I do?

philips77, 2021-03-01:
Hello,
Yes, our nRF Mesh libraries and apps support that feature. You need to configure the node to publish to a group address (or phone's unicast address, but group addresses are a better fit in this case). For that:
1. Provision your node, or import mesh network configuration
2. Create a group, if you don't have one already
3. Create an Application Key, or use one you already have
4. Add that Application Key to the node if it doesn't know it
5. Bind the node's model that you want to publish to this Application Key if it's not bound already
6. Configure publishing on the model: select the Application Key and the Group as destination
7. Bind your local model that should receive the messages to the same Application Key
8. Subscribe your local model to the Group address
9. Add the group address to Inclusion List filter, so that your proxy relayed messages targeting that address to your phone over GATT bearer

trusbe, 2021-03-05:
Thank you very much, according to your programming, I have already realized it.

closed 2021-03-05 by philips77