nordicsemi/IOS-nRF-Mesh-Library #337: How to load a mesh configuration on a second device with working messaging?

jcarioti, 2021-03-09:
**Describe the bug**
I'm creating a mesh network on one device, provisioning a few nodes, saving the mesh configuration to a common location, and then loading the mesh config on a second device. The second device gets a successful connection event, but mesh messages are ignored.

I am unsure of how to add the second device to the mesh so it can participate in mesh messaging. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a mesh network
2. Provision a node
3. Save JSON config outside of device
4. Load JSON config on new device
5. Connect to nodes
6. Try to send a message

**Expected behavior**
The second device can participate in the mesh messaging.

**Platform details:**
 - Device: iPhone 12
 - OS: iOS 14.4
 - Library Version (or Commit): 2.2.2

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

philips77, 2021-03-10:
Hello,
Before exporting the network configuration to another phone you need to go to Settings->Provisioners and create a new provisioner object for your other phone. Here define the ranges that the new Provisioner should use, and assign a Unicast Address to it from that range. By default a new Provisioner has no address set, so you need to check that. Then, with that ready, export the network configuration. You may either export full config and then select the new Provisioner on the other phone after importing, or export partial config with only the other Provisioner. You'll find those options on the export screen.
The reason why the other phone must be a new Provisioner is that it needs a unique address.

jcarioti, 2021-03-10:
> Hello,
> Before exporting the network configuration to another phone you need to go to Settings->Provisioners and create a new provisioner object for your other phone. Here define the ranges that the [...]

Thank you, I will try this as soon as I get a chance. One question, is the JSON that's produced when you export the same as the JSON that's saved internally by the library? Or does exported JSON get modified before export?

philips77, 2021-03-11:
The internally saved Json wraps the external in a tag which could allow to save some additional user data, but i never found a need for that.

closed 2021-04-15 by philips77