nordicsemi/IOS-nRF-Mesh-Library #339: SegmentedAccessMessage Overflow crash

fiveagle, 2021-03-25:
Hi,
congratulations for releasing 3.x, I have been taking a look at the example project and it has really nice improvements.

Anyway I am writing because I have encountered a nasty overflow crash while implementing a new very long message for our vendor model.

I crashed on  `SegmentedAccessMessage.swift` r:128 SDK 2.2.3. Reason: `"Swift runtime failure: arithmetic overflow"`

```
self.segmentOffset = offset
        
let lowerBound = Int(offset * 12)
```

Apparently my offset is 22 and so 22 * 12 = 264 which is greater than Uint8 upperBound 256

I have solved doing:
```   
let lowerBound = Int(offset) * 12
```

Do you think is the correct approach? I am wondering if that could potentially have bad consequences on messages in general.

Thanks,

philips77, 2021-03-26:
This looks like a correct approach. I will prepare a fix and release 3.1.1 soon.

philips77, 2021-03-26:
Released.

PR #340 (2021-03-26): Fix: Crash when segmenting a long access message
This PR fixes #339.
closed 2021-03-26 by philips77