nordicsemi/IOS-nRF-Mesh-Library #342: A few questions about unicast addresses

trusbe, 2021-04-08:
Hello, here are a few questions about unicast addresses, please ask:

1. How to set the specified unicast address to local Provisioner? What interface is used?
2. After the node's unicast address is reset, can this address be reassigned?
3. Can the unicast address of a device that has been connected to the network be modified? What interface is used?

philips77, 2021-04-08:
Hello,

> 1. How to set the specified unicast address to local Provisioner? What interface is used?

You should use the following method:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1bdfe99e42d9a9896e6df71a73b1444aba078fa/nRFMeshProvision/Classes/Mesh%20API/MeshNetwork%2BProvisioner.swift#L340-L350
To get the local provisioner, use:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1bdfe99e42d9a9896e6df71a73b1444aba078fa/nRFMeshProvision/Classes/Mesh%20API/MeshNetwork%2BProvisioner.swift#L35-L39

> 2. After the node's unicast address is reset, can this address be reassigned?

It can be reassigned to a new device during provisioning, but only when the IV Index increases by 2 or more since the node with the address was removed from the network. Until then it is added to:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1bdfe99e42d9a9896e6df71a73b1444aba078fa/nRFMeshProvision/Classes/Mesh%20Model/MeshNetwork.swift#L64-L65

> 3. Can the unicast address of a device that has been connected to the network be modified? What interface is used?

No. You'd have to reset it and provision it using a different unicast address. A working device cannot change it's address, and that's not really required for anything. What address a device has is not important, unless you have some logic assigned to it (e.g. light switches has addresses < 0x0010), but such logic introduces a lot of problems. I highly discourage. Addresses could be pretty much random and what address a device has is not important.

philips77, 2021-04-08:
One other note, the unicast address that a provisioner can have assigned to must be in the provisioner's unicast address range. Use:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1bdfe99e42d9a9896e6df71a73b1444aba078fa/nRFMeshProvision/Classes/Mesh%20API/Provisioner%2BRanges.swift#L45-L52
to assign new range to a provisioner.

closed 2021-04-08 by philips77

trusbe, 2021-04-12:
@philips77 

Thank you very much for your reply. The above information is very helpful to me. 

There is another question. After the node's unicast address is reset, the unicast address of the device needs to be removed from the `networkExclusions`. This interface is not allowed to be called, please ask if there are other The solution?

Is it possible to directly increase the current IV Index by 2? Does the interface use the interface in this file`MeshNetwork+IvIndex.swift`?

Thank you.

philips77, 2021-04-12:
> After the node's unicast address is reset, the unicast address of the device needs to be removed from the networkExclusions. This interface is not allowed to be called, please ask if there are other The solution?

The unicast address will be removed from exclusions automatically when IV Index is incremented by 2+. The reason why you can't do this manually, is that other nodes remember the SEQ number of the last message sent from (all of) its address. Had you provisioned next device with the same address before IV Index is greater than 2+ comparing to when the device was reset, those nodes would have discarded its messages until they are greater than the old SEQ.

> Is it possible to directly increase the current IV Index by 2? Does the interface use the interface in this fileMeshNetwork+IvIndex.swift?

IV Index change can be requested by any node (which has this feature enabled), and usually is done when it's SEQ counter reaches 50% of the maximum value. It takes at least 96h to transition to the next IV Index Phase and 2 phases are needed to increment IV Index by 1. The nRF Mesh lib does not support initiating IV Index update at the moment and I don't think it's necessary.

You should give up the idea of reassigning the same address to a different node. Perhaps you will be able to do this, but it may take a long time. With normal use, the IV Index transition is very rare.

trusbe, 2021-04-13:
understood, Thank you very much for your patience reply.