nordicsemi/IOS-nRF-Mesh-Library #348: Importing Android 2.1 on iOS 3.x

fiveagle, 2021-04-29:
Hi,
we are doing some test about 3.x backwards compatibility and we encounter a failed MeshNetwork deconding when opening with iOS 3.x an Android 2.1.4 
Apparently the `groups` array key is missing.
On Android we didn't add any node besides the provisioner. 
I have fixed by adding a fallback on an empty group array. 
`groups = (try? container.decode([Group].self, forKey: .groups)) ?? []`
Is it a safe fix to assume it empty?

This is also present on iOS 2.x

Thank You.

Codable Error:

```
▿ DecodingError
  ▿ keyNotFound : 2 elements
    - .0 : CodingKeys(stringValue: "groups", intValue: nil)
    ▿ .1 : Context
      - codingPath : 0 elements
      - debugDescription : "No value associated with key CodingKeys(stringValue: \"groups\", intValue: nil) (\"groups\")."
      - underlyingError : nil
```

JSON:
```
{
  "meshUUID" : "023FD9E1063E438789EE661B2BC958CA",
  "provisioners" : [
    {
      "allocatedGroupRange" : [
        {
          "lowAddress" : "C000",
          "highAddress" : "CC9A"
        }
      ],
      "allocatedUnicastRange" : [
        {
          "lowAddress" : "0001",
          "highAddress" : "199A"
        }
      ],
      "UUID" : "3B330BD65A4D4D9A89533FE44EB092EF",
      "provisionerName" : "nRFMeshProvisioner",
      "allocatedSceneRange" : [
        {
          "lastScene" : "3333",
          "firstScene" : "0001"
        }
      ]
    }
  ],
  "nodes" : [
    {
      "features" : {
        "friend" : 2,
        "proxy" : 2,
        "relay" : 2,
        "lowPower" : 2
      },
      "elements" : [
        {
          "models" : [
            {
              "modelId" : "0001",
              "bind" : [

              ],
              "subscribe" : [

              ]
            }
          ],
          "name" : "Element:0x0001",
          "location" : "0000",
          "index" : 0
        }
      ],
      "unicastAddress" : "0001",
      "configComplete" : true,
      "netKeys" : [
        {
          "index" : 0,
          "updated" : false
        }
      ],
      "defaultTTL" : 5,
      "appKeys" : [
        {
          "index" : 0,
          "updated" : false
        },
        {
          "index" : 1,
          "updated" : false
        },
        {
          "index" : 2,
          "updated" : false
        }
      ],
      "blacklisted" : false,
      "UUID" : "3B330BD65A4D4D9A89533FE44EB092EF",
      "security" : "low",
      "name" : "nRFMeshProvisioner",
      "deviceKey" : "61849102C0A0D4C275171F3E7E93B3D1"
    }
  ],
  "id" : "TBD",
  "netKeys" : [
    {
      "phase" : 0,
      "minSecurity" : "high",
      "key" : "C5F39C0F7438891EB0B7B266A94FDEFD",
      "timestamp" : "2021-04-29T12:58:53+0200",
      "name" : "NetworkKey1",
      "index" : 0
    }
  ],
  "$schema" : "http:\/\/json-schema.org\/draft-04\/schema#",
  "version" : "1.0",
  "meshName" : "nRFMeshNetwork",
  "timestamp" : "2021-04-29T12:58:53+0200",
  "appKeys" : [
    {
      "key" : "D3A2A00EA92EA8101FBC47870F17E976",
      "name" : "ApplicationKey1",
      "boundNetKey" : 0,
      "index" : 0
    },
    {
      "key" : "A189D199D081EBC7043A7A64AF516FE7",
      "name" : "ApplicationKey2",
      "boundNetKey" : 0,
      "index" : 1
    },
    {
      "key" : "9D46ED6CCA0B3D3ED1FAD55374AD585E",
      "name" : "ApplicationKey3",
      "boundNetKey" : 0,
      "index" : 2
    }
  ]
}
```

philips77, 2021-04-29:
Yes. But instead of 
```swift
groups = (try? container.decode([Group].self, forKey: .groups)) ?? []
```
you may do:
```swift
groups = try container.decodeIfPresent([Group].self, forKey: .groups) ?? []
```

philips77, 2021-04-29:
I should probably fix this in the library. Groups are required, so the issue was with nRF Mesh 2.1.x on Android, but iOS version could assume there are no groups.

fiveagle, 2021-04-29:
Ok great!

Thankyou

philips77, 2021-06-21:
Ups, I forgot to fix this in 3.1.3... This will need to wait until I'm back from vacations.

PR #357 (2021-06-25): Allowing groups to be nil when importing
This PR fixes #348.
closed 2021-06-25 by philips77