nordicsemi/IOS-nRF-Mesh-Library #350: EXC_BAD_ACCESS in LowerTransportLayer#assemble

ball-hayden, 2021-06-10:
**Describe the bug**

`EXC_BAD_ACCESS` in LowerTransportLayer#assemble.

```
* thread #35, queue = 'com.apple.root.background-qos', stop reason = EXC_BAD_ACCESS (code=1, address=0x10)
    frame #0: 0x0000000199f73474 libobjc.A.dylib`objc_msgSend + 20
    frame #1: 0x00000001896783f8 libswiftCore.dylib`Swift.Dictionary._Variant.removeValue(forKey: τ_0_0) -> Swift.Optional<τ_0_1> + 556
  * frame #2: 0x0000000101d7c830 mobile`closure #1 in LowerTransportLayer.assemble(_0=0x00000002839572d0, self=0x0000000281c14bd0, key=751896992) at LowerTransportLayer.swift:474:43
    frame #3: 0x0000000101c4a854 mobile`closure #1 in BackgroundTimer.init(block=0x0000000101d7fd14 mobile`partial apply forwarder for closure #1 (nRFMeshProvision.BackgroundTimer) -> () in nRFMeshProvision.LowerTransportLayer.assemble(segment: nRFMeshProvision.SegmentedMessage, createdFrom: nRFMeshProvision.NetworkPdu) -> Swift.Optional<nRFMeshProvision.LowerTransportPdu> at <compiler-generated>, self=0x00000002839572d0, repeats=false) at BackgroundTimer.swift:53:13
    frame #4: 0x00000001014dd8cc mobile`thunk for @escaping @callee_guaranteed () -> () at <compiler-generated>:0
    frame #5: 0x0000000107431528 libdispatch.dylib`_dispatch_client_callout + 16
    frame #6: 0x00000001074341cc libdispatch.dylib`_dispatch_continuation_pop + 524
    frame #7: 0x0000000107447a8c libdispatch.dylib`_dispatch_source_invoke + 1436
    frame #8: 0x0000000107443048 libdispatch.dylib`_dispatch_root_queue_drain + 376
    frame #9: 0x0000000107443970 libdispatch.dylib`_dispatch_worker_thread2 + 152
    frame #10: 0x00000001ce415568 libsystem_pthread.dylib`_pthread_wqthread + 212
```

**To Reproduce**

Steps to reproduce the behavior:
1. Create a network with 9 active nodes
2. Wait

**Expected behavior**

The app shouldn't crash.

**Platform details:**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Library Version (or Commit)

**Logs / Screenshots**

```
2021-06-10 16:24:15.842707+0100 mobile[517:128179] Sending ACK (seqZero: 1967, blockAck: 0x00000001)
2021-06-10 16:24:15.913458+0100 mobile[517:128179] Sending Network PDU (ivi: 0, nid: 0x0B, ctl: 1, ttl: 5, seq: 1304, src: 7B4A, dst: 0006, transportPdu: 0x56E35A34320840, netMic: 0xABAB75F92A3F8513) encrypted using Edge Network Key (index: 0)
2021-06-10 16:24:16.530379+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1461, src: 399E, dst: 7B4A, transportPdu: 0x6316800863691ABF39203E5F154D71, netMic: 0xD0D48061) received
2021-06-10 16:24:16.531378+0100 mobile[517:128088] Discarding packet (seqAuth: 1455, expected > 1457)
2021-06-10 16:24:16.570040+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 2346, src: 4008, dst: 7B4A, transportPdu: 0x0C0994AC49A22F0CFFDA44AF683E21, netMic: 0xB3CEF95A) received
2021-06-10 16:24:16.570429+0100 mobile[517:128088] Discarding packet (seqAuth: 2340, expected > 2342)
2021-06-10 16:24:16.615258+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1974, src: 0006, dst: 7B4A, transportPdu: 0x1EC92C55A104AAD5404126A62AC1488C, netMic: 0xA307AA32) received
2021-06-10 16:24:16.615642+0100 mobile[517:128088] Discarding packet (seqAuth: 1967, expected > 1969)
2021-06-10 16:24:16.696935+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1079, src: 2CCF, dst: 7B4A, transportPdu: 0x892A420904EBA112568DB3FAFBA9FD08, netMic: 0x2A0CA600) received
2021-06-10 16:24:16.697317+0100 mobile[517:128088] Discarding packet (seqAuth: 1072, expected > 1074)
2021-06-10 16:24:16.802206+0100 mobile[517:128179] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1584, src: 3339, dst: 7B4A, transportPdu: 0x89E2269A74BA7D37BD4A6930579565, netMic: 0x6D596DEE) received
2021-06-10 16:24:16.803416+0100 mobile[517:128179] Discarding packet (seqAuth: 1578, expected > 1580)
2021-06-10 16:24:16.808215+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 2470, src: 2CD1, dst: 7B4A, transportPdu: 0x324DDFA3F120D315F45D4F1455A4B3, netMic: 0x47824DB9) received
2021-06-10 16:24:16.808458+0100 mobile[517:128088] Discarding packet (seqAuth: 2464, expected > 2466)
2021-06-10 16:24:16.818088+0100 mobile[517:128179] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1975, src: 0006, dst: 7B4A, transportPdu: 0xEEE717DB0E11DD09F5CDB2756DE0F4, netMic: 0x227D1CF2) received
2021-06-10 16:24:16.818792+0100 mobile[517:128179] Discarding packet (seqAuth: 1967, expected > 1969)
2021-06-10 16:24:16.821085+0100 mobile[517:128088] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 63, src: 266B, dst: 7B4A, transportPdu: 0xBB6894B064E38B1CAB93948E52F1BEBD, netMic: 0x4AA615C3) received
2021-06-10 16:24:16.821673+0100 mobile[517:128088] Discarding packet (seqAuth: 55, expected > 62)
2021-06-10 16:24:16.830723+0100 mobile[517:128179] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 1080, src: 2CCF, dst: 7B4A, transportPdu: 0xDB545D7E9CD2C258E29329F9064AD5, netMic: 0x48512AF4) received
2021-06-10 16:24:16.830964+0100 mobile[517:128179] Discarding packet (seqAuth: 1072, expected > 1074)
2021-06-10 16:24:16.894126+0100 mobile[517:128179] Network PDU (ivi: 0, nid: 0x0B, ctl: 0, ttl: 6, seq: 64, src: 266B, dst: 7B4A, transportPdu: 0x1A4B3DFB68F029674977A68FE80738, netMic: 0x7201C654) received
2021-06-10 16:24:16.895928+0100 mobile[517:128179] Discarding packet (seqAuth: 55, expected > 62)
2021-06-10 16:24:21.604405+0100 mobile[517:127542] Incomplete message timeout: cancelling message (src: 3339, seqZero: 1578, received segments: 0x00000001)
2021-06-10 16:24:21.604422+0100 mobile[517:128179] Incomplete message timeout: cancelling message (src: 2CD1, seqZero: 2464, received segments: 0x00000001)
```

ball-hayden, 2021-06-10:
I'm afraid I haven't yet managed to isolate this to give a more complete reproduction.

ball-hayden, 2021-06-10:
I'm also getting it with this backtrace though:

```
* thread #29, queue = 'com.apple.root.background-qos', stop reason = EXC_BAD_ACCESS (code=1, address=0xbdb6cd4ac90)
    frame #0: 0x00000001898e40b0 libswiftCore.dylib`_swift_release_dealloc + 16
    frame #1: 0x00000001898da214 libswiftCore.dylib`swift_arrayDestroy + 68
    frame #2: 0x0000000189693d90 libswiftCore.dylib`Swift._DictionaryStorage.deinit + 404
    frame #3: 0x0000000189693e08 libswiftCore.dylib`Swift._DictionaryStorage.__deallocating_deinit + 12
    frame #4: 0x00000001898e40bc libswiftCore.dylib`_swift_release_dealloc + 28
    frame #5: 0x0000000189678a38 libswiftCore.dylib`Swift.Dictionary._Variant.removeValue(forKey: τ_0_0) -> Swift.Optional<τ_0_1> + 2156
  * frame #6: 0x0000000103523e70 mobile`closure #1 in LowerTransportLayer.assemble(_0=0x0000000280bb9590, self=0x0000000282fcf100, key=1074268522) at LowerTransportLayer.swift:463:63
    frame #7: 0x00000001033f2854 mobile`closure #1 in BackgroundTimer.init(block=0x0000000103527d14 mobile`partial apply forwarder for closure #1 (nRFMeshProvision.BackgroundTimer) -> () in nRFMeshProvision.LowerTransportLayer.assemble(segment: nRFMeshProvision.SegmentedMessage, createdFrom: nRFMeshProvision.NetworkPdu) -> Swift.Optional<nRFMeshProvision.LowerTransportPdu> at <compiler-generated>, self=0x0000000280bb9590, repeats=false) at BackgroundTimer.swift:53:13
    frame #8: 0x0000000102c858cc mobile`thunk for @escaping @callee_guaranteed () -> () at <compiler-generated>:0
    frame #9: 0x0000000108b99528 libdispatch.dylib`_dispatch_client_callout + 16
    frame #10: 0x0000000108b9c1cc libdispatch.dylib`_dispatch_continuation_pop + 524
    frame #11: 0x0000000108bafa8c libdispatch.dylib`_dispatch_source_invoke + 1436
    frame #12: 0x0000000108bab048 libdispatch.dylib`_dispatch_root_queue_drain + 376
    frame #13: 0x0000000108bab970 libdispatch.dylib`_dispatch_worker_thread2 + 152
    frame #14: 0x00000001ce415568 libsystem_pthread.dylib`_pthread_wqthread + 212
```

ball-hayden, 2021-06-17:
Also seeing it from https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/7f1ed6104739a1ff8a8c7dfa605c55e506690621/nRFMeshProvision/Classes/Layers/Lower%20Transport%20Layer/LowerTransportLayer.swift#L489

ball-hayden, 2021-06-17:
I still can't reliably trigger any of these, but I strongly suspect these are racy.
Swift isn't my strong point, but I believe that dictionaries are _not_ threadsafe - could the issue be that the timer(s) are firing at the same time, or one of the timers fires at the same time the ack arrives?

ball-hayden, 2021-06-17:
I've had a go at fixing this in https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/pull/351, but I'm not entirely sure I've caught all of it.

philips77, 2021-06-21:
PR merged, I'll release 3.1.3 soon.

closed 2021-06-21 by philips77

ball-hayden, 2021-06-21:
Thank you @philips77