nordicsemi/IOS-nRF-Mesh-Library #352: EXC_BAD_ACCESS in AccessLayer

okappi0258, 2021-06-18:
**Describe the bug**

EXC_BAD_ACCESS in AccessLayer

If I try to send a series of signals publish GenericOnOffSet, It breaks.

```
func send(_ message: MeshMessage,
              from element: Element, to destination: MeshAddress,
              withTtl initialTtl: UInt8?, using applicationKey: ApplicationKey) {
        // Should the TID be updated?
        var m = message
        if var tranactionMessage = message as? TransactionMessage, tranactionMessage.tid == nil {
            // Ensure there is a transaction for our destination.
            let k = key(for: element, and: destination)
            transactions[k] = transactions[k] ?? Transaction()
            // Should the last transaction be continued?
            if tranactionMessage.continueTransaction, transactions[k]!.isActive {
                tranactionMessage.tid = transactions[k]!.currentTid()
            } else {
                // If not, start a new transaction by setting a new TID value.
                tranactionMessage.tid = transactions[k]!.nextTid()
            }
            m = tranactionMessage
        }
```

EXC_BAD_ACCESS occurs here.

`transactions[k] = transactions[k] ?? Transaction()`

Why is this error occurring?
Please help me.

**Platform details:**
- Device:iPad Pro (Ⅱ generation)
- OS:iOS 14.6
- macOS:Big Sur 11.4
- Xcode:12.5

**Logs / Screenshots**
[Presentation] Attempt to present <UIAlertController: 0x10185e800> on <XXXXXXX: 0x103012000> (from <XXXXXXX: 0x11301a200>) which is already presenting <UIAlertController: 0x112813800>.

igathedub, 2021-06-18:
Yeah, same bug here. I think this comes from multithreading on newer hardware (arm64e). I do not have a solution either. Someone?

okappi0258, 2021-06-18:
Thank you for your comment.

I just tried it on arm64 iPhone and the same thing happened.
Apparently, it is not the fault of the new hardware (arm64e).

okappi0258, 2021-06-19:
I am sending signals triggered by values that are changed by a timer.
Could this be the cause?

okappi0258, 2021-06-21:
Does anyone know the cause of the problem?

philips77, 2021-06-21:
I'm working on a solution. I wrapped it all in a `mutex`, so threading should not be a problem. I agree with @igathedub that it looks like a threading issue, as it only happens when you want to send messages quickly.

philips77, 2021-06-21:
Please, try 3.1.3. I'm closing the issue, as I believe it should work now. If I'm wrong, feel free to reopen it. I wasn't able to test it so I rely on your findings.

closed 2021-06-21 by philips77

okappi0258, 2021-06-21:
Thank you so much!
I'll give it a try and let you know.

okappi0258, 2021-06-22:
Thank you @philips77 
The error is gone.
However, I want to light multiple LEDs at the same time without using groups, but there is a little time lag.
Is the time lag caused by UpperTransportLayer or something else?
Sorry for my poor English.

philips77, 2021-06-23:
I think the time lag is due to the fact that you need to send separate messages to each node, when not using groups. Each message may take different amount of time to reach the target, may need to be repeated, etc. In general, it's better to use group addresses whenever possible. For example, you may need to replace a device with another, and you will not be able to reuse the same unicast address, so you'll have to reconfigure all other nodes with the new address. Should you used groups, it's enough to subscribe the new node to the same groups and it's done.

okappi0258, 2021-06-24:
Thanks for the advice. I would like to try it out. 
Maybe I will ask again. Thank you for your cooperation.

PR #353 (2021-06-21): Bugfix/352
This PR aims to fix #352. The code that was causing a problem has been wrapped in a mutex block, so should be thread safe now. The fix has not been tested, but looks good to me.