nordicsemi/IOS-nRF-Mesh-Library #360: How to change each of the four LEDs to a different color at the same time

okappi0258, 2021-06-26:
I want to change all four LEDs to different colors at the same time.
Is it possible to combine 4 LEDs into one group and change each of them to a different color with a single GenericOnOffSet message?

philips77, 2021-07-02:
For setting a color you should use more complex model, eg. https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/nRFMeshProvision/Classes/Mesh%20Messages/Lighting/LightHSLSet.swift

philips77, 2021-07-02:
The Light HSL Server model could control all the different LEDs to present desired color.

okappi0258, 2021-07-02:
Thank you for your comment.

Can the Light HSL Server model be used to change 4 LEDs to different colors with one message?
For example, LED1:red, LED2:blue, LED3:red, LED4:blue.
Does this mean that it should not be a Genericonoffset message?

I apologize for my lack of knowledge, but I would appreciate it if you could tell me.

philips77, 2021-07-03:
You never have a single LED that produces any color, there are usually 3. With Light HSL Set you set a color and the fw needs to control available LEDs accordingly. Is like a single light with differently colors. If you have very simple design, with 4 LEDs with can just be turned on or off, the principle is similar.

okappi0258, 2021-07-03:
I understand a little more now．
What is the difference between the Generic on off set message and the Light HSL Set message?

philips77, 2021-07-05:
The first one has just one boolean value, while the latter lightness, hue and saturation: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/nRFMeshProvision/Classes/Mesh%20Messages/Lighting/LightHSLSet.swift#L61
You may also create a custom vendor model with vendor messages tailored for your need. That will be mush simpler than implementing Light models.

philips77, 2021-07-05:
For an example how to implement a vendor model in the iOS mesh lib, check out how the Simple On Of messages are created in the sample app:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/tree/master/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff

okappi0258, 2021-07-06:
Hmm.
Are you saying that the Light HSL Set is simpler than the Generic on off set, so it might not cause delays?

I want to change the color of the LED to match the time.
For example, 
LED1:red, LED2:blue, LED3:red, LED4:blue.
5 seconds later...
LED1:blue, LED2:red, LED3:blue, LED4:red.
5 seconds later...
LED1:red, LED2:blue, LED3:red, LED4:blue.
…

In the case of the Light HSL Set, am I correct in assuming that the LED is originally set to that color?
I have a lot of questions I want to ask and I don't think I'm communicating them well. I apologize for that.

trusbe, 2021-07-06:
It is recommended to use vendor model to customize the communication protocol;

okappi0258, 2021-07-06:
Thanks for the advice.
I am currently using the following generic on off set.

'''
import Foundation

public struct GenericOnOffSet: AcknowledgedGenericMessage, TransactionMessage, TransitionMessage {
    public static let opCode: UInt32 = 0x8202
    public static let responseType: StaticMeshMessage.Type = GenericOnOffStatus.self

    public var tid: UInt8!
    public var parameters: Data? {
        let data = Data([color, tid])
        print("≈data", data.map { String(format: "%.2hhx", $0) }.joined())
        print("≈color", color)
        print("≈tid", tid)
        if let transitionTime = transitionTime, let delay = delay {
            return data + transitionTime.rawValue + delay
        } else {
            return data
        }
    }

    /// The new state of Generic OnOff Server.
    public let color: UInt8

    public let transitionTime: TransitionTime?
    public let delay: UInt8?

    /// Creates the Generic OnOff Set message.
    ///
    /// - parameter isOn: The target value of the Generic OnOff state.
    public init(_ color: UInt8) {
        self.color = color
        self.transitionTime = nil
        self.delay = nil
    }

    /// Creates the Generic OnOff Set message.
    ///
    /// - parameters:
    ///   - isOn: The target value of the Generic OnOff state.
    ///   - transitionTime: The time that an element will take to transition
    ///                     to the target state from the present state.
    ///   - delay: Message execution delay in 5 millisecond steps.
    public init(_ color: UInt8, transitionTime: TransitionTime, delay: UInt8) {
        self.color = color
        self.transitionTime = transitionTime
        self.delay = delay
    }

    public init?(parameters: Data) {
        guard parameters.count == 2 || parameters.count == 4 else {
            return nil
        }
        color = parameters[0]
        print("≈public init", color)
        tid = parameters[1]
        if parameters.count == 4 {
            transitionTime = TransitionTime(rawValue: parameters[2])
            delay = parameters[3]
        } else {
            transitionTime = nil
            delay = nil
        }
    }
}
'''

It's a little customized.
Is this generic on off set the cause of the delay?

trusbe, 2021-07-06:
To change the color of four LEDs at the same time, you need to customize the content of your protocol data；

For example, four Uint8 values to control the PWM of each LED separately, so it is still recommended that you do not modify it on the Generic Model, and it is recommended to create a Vendor Model.

okappi0258, 2021-07-06:
A Vendor Model means creating a completely new model from scratch, right?

trusbe, 2021-07-06:
Yes, you can refer to the SDK Demo code. You need to add related Model and protocol content with your embedded colleagues, so that you can get rid of the shackles of the standard Model, and you can control it any way you want.

okappi0258, 2021-07-06:
I see. It seems difficult for me.
As a first aid, I want to shorten the time it takes for the signal to be retransmitted, but which part is causing the problem?

philips77, 2021-07-06:
You cannot modify Generic On Off Set message by changing on off flag to a color. That's a completely new message type, for which you need to use vendor message, like the Simple on off set message i pointed above. The genetic ones are defined by the spec.

philips77, 2021-07-06:
Basically, main diff between what you have above and need, is different message I'd. Instead of 0x8202, you need to add company identifier. Please, follow the Bluetooth mesh spec for details or ask DevZone.

philips77, 2021-07-06:
> It seems difficult for me. Is it enough to improve the Simple OnOff model to create a new model?

Yes. But you need to match your firmware, so the same model (but server, not client) should be on your device.

> As a first aid, I want to shorten the time it takes for the signal to be retransmitted, but which part is causing the problem?

You need to send 1 message, not 4, to change the color. All 4 LEDs needs to be controlled by one model.

okappi0258, 2021-07-06:
Thank you so much for your help.
I guess I should improve the Simple OnOff model so that I can control 4 LEDs with one message.
What part of the Simple OnOff model do I need to improve exactly?

philips77, 2021-07-07:
> Is it like combining 4 LEDs into one group and sending an array of parameters for the color you want to change to each LED?

Yes, for example. That way you'll be controlling the state of each LED on the client side. Or you may send a color as e.g. uint8 and do the magic on the device side. It's up to you. The point is, that all LEDs should be controlled with one message. You may also keep the genetic on off models to turn on off all LEDs and allow more genetic apps to control it. Also up to you.

closed 2021-07-07 by philips77

okappi0258, 2021-07-07:
Thank you for everything.
I'll try my best.