nordicsemi/IOS-nRF-Mesh-Library #361: Nrf mesh cannot detect my light device

mesha227, 2021-06-30:
I am unable to detect the light bulb through nrf mesh even after using different phone models, iPhone and android. I am currently using nrf mesh version 3.1.4 and nrf sdk mesh version 5.0.0
I could still connect to it via nrf connect.

philips77, 2021-07-03:
Are you sure that your light bulb is unprovisioned and advertising as such?
Could you tell how do you see it in nRF Connect?

mesha227, 2021-07-05:
this is what i see on the nRF connect
[image]

philips77, 2021-07-05:
Those are all custom services and characteristics, with nothing in common with Bluetooth mesh. Seems like your device doesn't support mesh, so it will not work with nRF Mesh.

closed 2021-07-05 by philips77