nordicsemi/IOS-nRF-Mesh-Library #362: Vendor Messages - Unable to receive response

tu-lucaspalaian, 2021-07-01:
Good afternoon, I would like to get help on this as Im stuck right now.

I have an AcknowledgedStaticVendorMessage which its OpCode is 0xF0 and I send an Int32 and Int8 as its parameters. Its status code is 0xD2

```swift
import nRFMeshProvision

struct SetIdAcknowledged: AcknowledgedStaticVendorMessage, SetConfig {
    static var responseType: StaticMeshMessage.Type = SetIdStatus.self
    static var opCode: UInt32 = OpCodes.SET_ID_ACKNOWLEDGED.rawValue // 0xF0
    
    let parameters: Data?
    let nodeId: Int
    
    init?(parameters: Data) {
        self.parameters = parameters
        self.nodeId = .zero
    }
    
    init(nodeId: Int) {
        var byteArray: [UInt8] = .init()
        nodeId.appendAsInt32(to: &byteArray)
        byteArray.append(TransactionIdProvider.shared.get())
        self.nodeId = nodeId
        self.parameters = Data(byteArray)
    }
}

struct SetConfigStatus: StaticVendorMessage {
    static var opCode: UInt32 = OpCodes.SET_ID_STATUS.rawValue // 0xD2
    
    var parameters: Data?
    
    init?(parameters: Data) {
        self.parameters = parameters
    }
}
```

In node.elements there is a vendor model which has the 0x1000 id and 0x1000 company identifier. So with those values I have  added a model into my local elements

```swift
networkManager.localElements = [ Element(name: "Primary Element", location: .first, models: [
     .init(vendorModelId: VendorConstants.VENDOR_MODEL_ID, // 0x1000
             companyId: VendorConstants.COMPANY_ID, // 0x1000
             delegate: NodeVendorMessageDelegate())
     ])
]
````

The delegate looks like this
```swift
class NodeVendorMessageDelegate: ModelDelegate {
    var messageTypes: [UInt32 : MeshMessage.Type]
    
    var isSubscriptionSupported: Bool = false
    
    var publicationMessageComposer: MessageComposer? = nil
    
    init() {
        let messages: [StaticVendorMessage.Type] = [
            SetIdStatus.self
        ]
        messageTypes = messages.toMap()
    }
    
    func model(_ model: Model, didReceiveAcknowledgedMessage request: AcknowledgedMeshMessage, from source: Address, sentTo destination: MeshAddress) throws -> MeshMessage {
        fatalError()
    }
    
    func model(_ model: Model, didReceiveUnacknowledgedMessage message: MeshMessage, from source: Address, sentTo destination: MeshAddress) {
        
    }
    
    func model(_ model: Model, didReceiveResponse response: MeshMessage, toAcknowledgedMessage request: AcknowledgedMeshMessage, from source: Address) {
        
    }
}
```

So at this point at least I would expect to receive an UnknownMessage when sending my custom SetId vendor message. But it doesn't.

Here is the list of events/messages that im sending PREVIOUS of sending the custom vendor message

1. ConfigCompositionDataGet
2. ConfigDefaultTtlSet
3. ConfigNetworkTransmitSet
4. ConfigNetKeyAdd
5. ConfigAppKeyAdd
6. ConfigModelAppBind // using primary app key that I added in the step 4 and node's vendor model
7. ConfigModelPublicationSet // using node's vendor model and it's element address
8. CUSTOM VENDOR MESSAGE SetIdAck

What I have tried?
I have tried adding the companyId that I got from the node's vendor model to the opcode like this and nothing works (ref. https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/138)
- 0xF01000
- 0xF00001

Could anyone give me a hand on this please? Im open to share any other detail if needed

philips77, 2021-07-03:
Hello,
The opcode 0xF0 is outside of available range for vendor messages. Available range is 0x00-0x3F. The 2 msb are set to 1 to indicate the vendor opcode type, but the fw receives 0x30 in your case.
And yes, you need to concatenate the company id. Have a look in this sample:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff/Messages/SimpleOnOffSet.swift#L35-L40.

philips77, 2021-07-03:
In your case the opcode skills be: 0xF00010.

closed 2021-11-18 by philips77