nordicsemi/IOS-nRF-Mesh-Library #363: Error occurred in group transmission

okappi0258, 2021-07-14:
This is the same person who asked about how to change 4 LEDs to different colors at the same time.
I was able to successfully change them at the same time. However, sometimes I fail to do so.
The four LEDs are grouped into one group, but when they fail, only one of them changes color.
What is the cause?

igathedub, 2021-07-14:
Hi, logs help a ton. Sometimes there are sequence number problems for example.

okappi0258, 2021-07-14:
Thank you for your comment.

Here is the log when it worked.
```
2021-07-09 14:06:52.718992+0900 XXXX-ios-mobile-client[3660:1872615] [Model] Sending GenericOnOffSet(tid: Optional(85), color: 3, transitionTime: Optional(Immediate), delay: Optional(0)) from: Primary Element (0x0001), to: C007
≈data 0355
≈color 3
≈tid Optional(85)
2021-07-09 14:06:52.719316+0900 XXXX-ios-mobile-client[3660:1872615] [Access] Sending Access PDU (opcode: 0x8202, parameters: 0x03550000)
2021-07-09 14:06:52.720450+0900 XXXX-ios-mobile-client[3660:1872615] [UpperTransport] Sending Upper Transport PDU (encrypted data: 0x707A3EA262BF, transMic: 0xEA665166) encrypted using key: New Application Key (index: 0)
2021-07-09 14:06:52.720771+0900 XXXX-ios-mobile-client[3660:1872615] [LowerTransport] Sending Access Message (akf: 1, aid: 0x38, szmic: 0, data: 0x707A3EA262BFEA665166)
2021-07-09 14:06:52.722773+0900 XXXX-ios-mobile-client[3660:1872615] [Network] Sending Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 127, seq: 200, src: 0001, dst: C007, transportPdu: 0xBAE0D73BDCB6C51909F7CF, netMic: 0x563FC66C) encrypted using Primary Network Key (index: 0)
2021-07-09 14:06:52.723226+0900 XXXX-ios-mobile-client[3660:1872615] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 127, seq: 200, src: 0001, dst: C007, transportPdu: 0xBAE0D73BDCB6C51909F7CF, netMic: 0x563FC66C) received
2021-07-09 14:06:52.725590+0900 XXXX-ios-mobile-client[3660:1872615] [LowerTransport] Access Message (akf: 1, aid: 0x38, szmic: 0, data: 0x707A3EA262BFEA665166) receieved (decrypted using key: Primary Network Key (index: 0))
2021-07-09 14:06:52.725853+0900 XXXX-ios-mobile-client[3660:1872615] [UpperTransport] Upper Transport PDU (encrypted data: 0x707A3EA262BF, transMic: 0xEA665166) received
2021-07-09 14:06:52.726073+0900 XXXX-ios-mobile-client[3660:1872615] [Access] Access PDU (opcode: 0x8202, parameters: 0x03550000) receieved (decrypted using key: New Application Key (index: 0))
≈public init 3
2021-07-09 14:06:52.726943+0900 XXXX-ios-mobile-client[3660:1872615] [Model] GenericOnOffSet(tid: Optional(85), color: 3, transitionTime: Optional(Immediate), delay: Optional(0)) received from: 0001, to: C007
≈public init 3
≈public init 3
≈public init 3
≈public init 3
≈public init 3
≈public init 3
Ω didReceiveMessage GenericOnOffSet(tid: Optional(85), color: 3, transitionTime: Optional(Immediate), delay: Optional(0))
2021-07-09 14:06:52.729920+0900 XXXX-ios-mobile-client[3660:1872615] [Bearer] -> 0x001CD863F5FED5BD3E39BAE0D73BDCB6C51909F7CF563FC66C
2021-07-09 14:06:53.164612+0900 XXXX-ios-mobile-client[3660:1870715] [Bearer] <- 0x001C731177786B1CE11691CBE4E4894CB45976788EDD
2021-07-09 14:06:53.165484+0900 XXXX-ios-mobile-client[3660:1870715] [Bearer] <- 0x001CA0A2090C2AB647B4B4C58B90A8DCAB1B14EC462A
2021-07-09 14:06:53.166322+0900 XXXX-ios-mobile-client[3660:1870715] [Bearer] <- 0x001C23610E779BC2FB43F6E91ABC0670ECA2C9D996C6
2021-07-09 14:06:53.167216+0900 XXXX-ios-mobile-client[3660:1870715] [Bearer] <- 0x001C066916FCEDD1DC74AF4F79329E520AD8EA4370FE
2021-07-09 14:06:53.167973+0900 XXXX-ios-mobile-client[3660:1872611] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 4, seq: 98, src: 000F, dst: 0001, transportPdu: 0x91CBE4E4894CB459, netMic: 0x76788EDD) received
2021-07-09 14:06:53.168406+0900 XXXX-ios-mobile-client[3660:1872616] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 3, seq: 89, src: 000C, dst: 0001, transportPdu: 0xB4C58B90A8DCAB1B, netMic: 0x14EC462A) received
2021-07-09 14:06:53.168624+0900 XXXX-ios-mobile-client[3660:1872375] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 3, seq: 89, src: 0012, dst: 0001, transportPdu: 0xF6E91ABC0670ECA2, netMic: 0xC9D996C6) received
2021-07-09 14:06:53.170206+0900 XXXX-ios-mobile-client[3660:1872661] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 3, seq: 95, src: 0009, dst: 0001, transportPdu: 0xAF4F79329E520AD8, netMic: 0xEA4370FE) received
2021-07-09 14:06:53.172085+0900 XXXX-ios-mobile-client[3660:1872611] [LowerTransport] Access Message (akf: 1, aid: 0x38, szmic: 0, data: 0xFCD2F694639760) receieved (decrypted using key: Primary Network Key (index: 0))
2021-07-09 14:06:53.172733+0900 XXXX-ios-mobile-client[3660:1872611] [UpperTransport] Upper Transport PDU (encrypted data: 0xFCD2F6, transMic: 0x94639760) received
2021-07-09 14:06:53.174120+0900 XXXX-ios-mobile-client[3660:1872611] [Access] Response Access PDU (opcode: 0x8204, parameters: 0x03) receieved (decrypted using key: New Application Key (index: 0))
2021-07-09 14:06:53.175429+0900 XXXX-ios-mobile-client[3660:1872611] [Model] GenericOnOffStatus(color: 3, targetState: nil, remainingTime: nil) received from: 000F, to: 0001
Ω didReceiveMessage GenericOnOffStatus(color: 3, targetState: nil, remainingTime: nil)
Ω GenericOnOffStatus(color: 3, targetState: nil, remainingTime: nil)
2021-07-09 14:06:53.177782+0900 XXXX-ios-mobile-client[3660:1872616] [LowerTransport] Access Message (akf: 1, aid: 0x38, szmic: 0, data: 0x0A8338042E6882) receieved (decrypted using key: Primary Network Key (index: 0))
2021-07-09 14:06:53.178279+0900 XXXX-ios-mobile-client[3660:1872616] [UpperTransport] Upper Transport PDU (encrypted data: 0x0A8338, transMic: 0x042E6882) received
2021-07-09 14:06:53.178597+0900 XXXX-ios-mobile-client[3660:1872616] [Access] Access PDU (opcode: 0x8204, parameters: 0x03) receieved (decrypted using key: New Application Key (index: 0))
[... cut ...]
2021-07-09 14:05:27.008133+0900 XXXX-ios-mobile-client[3660:1872373] [Access] Response Access PDU (opcode: 0x8204, parameters: 0x02) receieved (decrypted using key: New Application Key (index: 0))
2021-07-09 14:05:27.009178+0900 XXXX-ios-mobile-client[3660:1872373] [Model] GenericOnOffStatus(color: 2, targetState: nil, remainingTime: nil) received from: 000F, to: 0001
Ω didReceiveMessage GenericOnOffStatus(color: 2, targetState: nil, remainingTime: nil)
Ω GenericOnOffStatus(color: 2, targetState: nil, remainingTime: nil)
```

Sorry this is so long.

philips77, 2021-07-29:
First of all, you should not modify GenericOnOffSet. This message format is defined by the specification and should only be used to turn on or off. For sending colors use vendor model and vendor messages, which can look exactly the same, or even simpler, as you don't need TID and transition time and delay, I believe, but with different Op Code. Vendor messages must contain Company ID (16 bit) and 6-bit Op Code. Have a look at how nRF Mesh declares Simple On Off Client model and messages:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/tree/master/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff
and how to initialize the model here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/Example/nRFMeshProvision/AppDelegate.swift#L129-L132
Such client model specifies, that the Simple On Off Client model accepts Simple On Off Status message, so that it could be parsed when received. You may publish Set messages from this model as such:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff/SimpleOnOffClientDelegate.swift#L53-L62

Answering your question, it looks like the Set message hasn't reach all the target nodes, or the Status messages from those nodes didn't reach the phone. As you're saying the LEDs didn't change, most probably the messages were lost on the way. Try increasing Network Transmit on the proxy. Also, when sending messages to a group address perhaps it's better to use unacknowledged messages, otherwise you may get network congestion. If needed, set Publications on target nodes so that they publish Status message when the state changes. You may set Publish Retransmission parameter there, so each Status is sent multiple times if you think messages are getting lost.

okappi0258, 2021-07-30:
Thank you for your answer.
I would like to try it first with an unacknowledged message. I see that I should not edit GenericOnOffSet. In the meantime, here is the code for GenericOnOffSet.

philips77, 2021-11-18:
Right. But you may name however you want. In nRF5 SDK for Mesh Nordic has a sample which is called Simple On Off, that's why the name in this app.

Did you manage to solve your issue? Sorry for the long delay, but I was busy with other projects.

okappi0258, 2021-11-19:
Yes. I can now send to the group successfully. However, I have a new problem.
I want to store an array (such as [2,2,4,4]) in data and send it as a parameter, but I don't know how to do that.

philips77, 2021-11-19:
Have a look at 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/Example/nRFMeshProvision/Mesh%20Network/Simple%20OnOff/Messages/SimpleOnOffSet.swift#L34-L61
This sample vendor message has just 1 byte, which is set to 0x00 or 0x01 based on the `isOn` parameter.
In your case, instead of 
```swift
/// The state.
let isOn: Bool
```
you may have 
```swift
let someData: Data
```
and just return those data from
```swift
var parameters: Data? {
    return someData
}
```

That way, the whole `Data` will be sent. But remember, that that would allow any size of the data, so perhaps adding some higher level API, like in this `SimpleOnOffSet` message, can be a good choice.

I'll close the issue now, as it has been resolved. Feel free to open a new one if you encounter another problem.