nordicsemi/IOS-nRF-Mesh-Library #364: Bearer closes when sending config message

joeynelson42, 2021-07-29:
**Describe the bug**
I have implemented the basic provisioning process in my app, and am attempting now to communicate via the Proxy service.

I try to send the first config message (ConfigCompositionDataGet) to a provisioned device but as soon as the GattBearer opens and I send the message, the bearer closes.

I know it's not a firmware issue as I am able to successfully complete all these steps via the Nordic Mesh app.

I've read through #301 and implemented the suggestions there, but with no success. Any help is appreciated!

**To Reproduce**
Steps to reproduce the behavior:
1. Provision device
2. Create and open GattBearer
3. Send ```ConfigCompositionDataGet(page: 0)``` message

**Expected behavior**
I expected the message to successfully send via the GattBearer.

**Platform details:**
 - Device: iPhone 12 mini
 - OS: iOS 14.6
 - Library Version: 3.1.4

**Logs / Screenshots**
```
Bearer did connect
Bearer did discover services
Bearer did open.
MESH LOG: Sending ConfigCompositionDataGet(page: 0) to: 0002
MESH LOG: Sending Access PDU (opcode: 0x8008, parameters: 0x00)
MESH LOG: Sending Upper Transport PDU (encrypted data: 0x795B72, transMic: 0xAAE391A3) encrypted using key: WarrenMesh  1246's Device Key
MESH LOG: Sending Access Message (akf: 0, szmic: 0, data: 0x795B72AAE391A3)
MESH LOG: The bearer is closed.
MESH LOG: Resending Access PDU (opcode: 0x8008, parameters: 0x00)
MESH LOG: Sending Upper Transport PDU (encrypted data: 0xC440BF, transMic: 0x00DAB681) encrypted using key: WarrenMesh  1246's Device Key
MESH LOG: Sending Access Message (akf: 0, szmic: 0, data: 0xC440BF00DAB681)
MESH LOG: The bearer is closed.
MESH LOG: Resending Access PDU (opcode: 0x8008, parameters: 0x00)
MESH LOG: Sending Upper Transport PDU (encrypted data: 0x61C07D, transMic: 0xD84D674E) encrypted using key: WarrenMesh  1246's Device Key
MESH LOG: Sending Access Message (akf: 0, szmic: 0, data: 0x61C07DD84D674E)
MESH LOG: The bearer is closed.
MESH LOG: Resending Access PDU (opcode: 0x8008, parameters: 0x00)
MESH LOG: Sending Upper Transport PDU (encrypted data: 0xEE2191, transMic: 0x6AEA48C6) encrypted using key: WarrenMesh  1246's Device Key
MESH LOG: Sending Access Message (akf: 0, szmic: 0, data: 0xEE21916AEA48C6)
MESH LOG: The bearer is closed.
MESH LOG: Response to Access PDU (opcode: 0x8008, parameters: 0x00) not received (timeout)
MESH LOG: ConfigCompositionDataGet(page: 0) sent from: 0001, to: 0002 timed out
MESH LOG: Cancelling messages with op code: 32776, sent from: 0001 to: 0002
```

philips77, 2021-07-29:
It may be that you didn't set the bearer as transporter in `MeshNetworkManager` object:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/Example/nRFMeshProvision/AppDelegate.swift#L151
In the nRF Mesh app we're using `NetworkConnection` object, which can handle multiple simultaneous connections (currently limit is set to 1), but you can just set your GATT bearer object there.

joeynelson42, 2021-07-29:
Worked like a charm! Thanks!

closed 2021-07-29 by joeynelson42