nordicsemi/IOS-nRF-Mesh-Library #365: Not Receiving Config Message Responses

joeynelson42, 2021-07-29:
**Describe the Issue**
I am sending configuration messages to a provisioned device and I am receiving no delegate callback responses in return.

When I send the ```ConfigCompositionDataGet``` and ```ConfigDefaultTtlGet``` messages I receive a callback to the ```didSendMessage``` callback, however the ```didReceiveMessage``` is never called. When I attempt to send ```ConfigAppKeyAdd``` it attempts to send but eventually times out with the log message:

```Response to Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44) not received (timeout)```

On the firmware side, all these messages are seen, and the ```ConfigAppKeyAdd``` is successful.

I've set the MeshNetworkManager's delegate, I've set the localElements to empty and confirmed the existence of the configuration models, I've tried combing over the example app and digging through the different access layers, but to no avail.

**Platform details:**
 - Device: iPhone 12 mini
 - OS: iOS 14.6
 - Library Version:  3.1.4

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

```
Bearer did connect
Bearer did discover services
Bearer did open.
LOG: Sending ConfigCompositionDataGet(page: 0) to: 0002
LOG: Sending Access PDU (opcode: 0x8008, parameters: 0x00)
LOG: Sending Upper Transport PDU (encrypted data: 0x83F6A1, transMic: 0xACCD7009) encrypted using key: nRF5x Mesh Scene Client's Device Key
LOG: Sending Access Message (akf: 0, szmic: 0, data: 0x83F6A1ACCD7009)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 232, src: 0001, dst: 0002, transportPdu: 0xA506B3FF499540BF, netMic: 0xB48A2F4D) encrypted using Primary Network Key (index: 0)
MeshNetwork sent message: ConfigCompositionDataGet(page: 0)
LOG: Cancelling messages with op code: 32776, sent from: 0001 to: 0002
LOG: Sending ConfigDefaultTtlGet() to: 0002
LOG: Sending Access PDU (opcode: 0x800C, parameters: 0x)
LOG: Sending Upper Transport PDU (encrypted data: 0x399E, transMic: 0x866707B6) encrypted using key: nRF5x Mesh Scene Client's Device Key
LOG: Sending Access Message (akf: 0, szmic: 0, data: 0x399E866707B6)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 233, src: 0001, dst: 0002, transportPdu: 0x0857FB84815B60, netMic: 0xD7ED5372) encrypted using Primary Network Key (index: 0)
MeshNetwork sent message: ConfigDefaultTtlGet()
LOG: Cancelling messages with op code: 32780, sent from: 0001 to: 0002
LOG: Sending ConfigAppKeyAdd(networkKeyIndex: 0, applicationKeyIndex: 0, key: 16 bytes) to: 0002
LOG: Sending Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44)
LOG: Sending Upper Transport PDU (encrypted data: 0x07EF29290875F881D46E4BFD112F89CBB20DCCA7, transMic: 0x7C8F85DD) encrypted using key: nRF5x Mesh Scene Client's Device Key
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 0, segN: 1, data: 0x07EF29290875F881D46E4BFD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 235, src: 0001, dst: 0002, transportPdu: 0x6BF1DF2207FAC198AB2E0CA6080BA015, netMic: 0xB2CB810C) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 1, segN: 1, data: 0x112F89CBB20DCCA77C8F85DD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 236, src: 0001, dst: 0002, transportPdu: 0x9EA41FE457198DD5D982485808AA5F63, netMic: 0xB12E3C63) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 0, segN: 1, data: 0x07EF29290875F881D46E4BFD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 237, src: 0001, dst: 0002, transportPdu: 0x11983E0AFF7A3D9F62F3262A74EAC116, netMic: 0xA42C2D56) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 1, segN: 1, data: 0x112F89CBB20DCCA77C8F85DD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 238, src: 0001, dst: 0002, transportPdu: 0x8D15B25E9085FF03260B7C5CBB43CB24, netMic: 0x9154F5ED) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 0, segN: 1, data: 0x07EF29290875F881D46E4BFD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 239, src: 0001, dst: 0002, transportPdu: 0xEF6AD1097AF33CA848A453A7E05B0801, netMic: 0x8FB922B2) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 234, segO: 1, segN: 1, data: 0x112F89CBB20DCCA77C8F85DD)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 240, src: 0001, dst: 0002, transportPdu: 0x4727CC21C151B5F2BA3F0EC1AA36BDA6, netMic: 0xF5118D8D) encrypted using Primary Network Key (index: 0)
LOG: Resending Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44)
LOG: Sending Upper Transport PDU (encrypted data: 0xA2A08BF87F07596E117708F3C3AFA4FC873E7351, transMic: 0x8736C67C) encrypted using key: nRF5x Mesh Scene Client's Device Key
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 0, segN: 1, data: 0xA2A08BF87F07596E117708F3)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 242, src: 0001, dst: 0002, transportPdu: 0xBAE54303773B8C16A66421E3D7216556, netMic: 0x3F76510F) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 1, segN: 1, data: 0xC3AFA4FC873E73518736C67C)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 243, src: 0001, dst: 0002, transportPdu: 0xC701B4DC83D2C240068631C1D3DF800D, netMic: 0x4571FE16) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 0, segN: 1, data: 0xA2A08BF87F07596E117708F3)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 244, src: 0001, dst: 0002, transportPdu: 0xD0573B3DF9B9B2B5E2BEC9FFC8A4EEA5, netMic: 0xBC464DCD) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 1, segN: 1, data: 0xC3AFA4FC873E73518736C67C)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 245, src: 0001, dst: 0002, transportPdu: 0x59E56A7EBBE91880CB372183D07AABAE, netMic: 0x96F59E19) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 0, segN: 1, data: 0xA2A08BF87F07596E117708F3)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 246, src: 0001, dst: 0002, transportPdu: 0xCB1B5A5EFDCA5B8271619492CFBFB981, netMic: 0x775F16DA) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 241, segO: 1, segN: 1, data: 0xC3AFA4FC873E73518736C67C)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 247, src: 0001, dst: 0002, transportPdu: 0x371F92D701F61D79A5C9D9B1A5306640, netMic: 0xDAF55F61) encrypted using Primary Network Key (index: 0)
LOG: Resending Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44)
LOG: Sending Upper Transport PDU (encrypted data: 0x40053F6919F948D26A066922150861ED9C5FF2E1, transMic: 0x833CD461) encrypted using key: nRF5x Mesh Scene Client's Device Key
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 248, segO: 0, segN: 1, data: 0x40053F6919F948D26A066922)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 249, src: 0001, dst: 0002, transportPdu: 0xDC7DFDCD2F4018192BDA16614D92E1DE, netMic: 0xA4041886) encrypted using Primary Network Key (index: 0)
[... 6 lines cut ...]
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 248, segO: 0, segN: 1, data: 0x40053F6919F948D26A066922)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 253, src: 0001, dst: 0002, transportPdu: 0x514D30826585806D0CF40D9557AAD916, netMic: 0xDE661495) encrypted using Primary Network Key (index: 0)
LOG: Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 248, segO: 1, segN: 1, data: 0x150861ED9C5FF2E1833CD461)
LOG: Sending Network PDU (ivi: 0, nid: 0x66, ctl: 0, ttl: 5, seq: 254, src: 0001, dst: 0002, transportPdu: 0x76991FAF4275A6B882C7022C86A1AB93, netMic: 0xDBB35701) encrypted using Primary Network Key (index: 0)
LOG: Resending Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44)
[... cut ...]
LOG: Response to Access PDU (opcode: 0x 0, parameters: 0x00000080A51EABC558689EA6F5D90944C51B44) not received (timeout)
LOG: ConfigAppKeyAdd(networkKeyIndex: 0, applicationKeyIndex: 0, key: 16 bytes) sent from: 0001, to: 0002 timed out
LOG: Cancelling messages with op code: 0, sent from: 0001 to: 0002
MeshNetwork failed to send message: ConfigAppKeyAdd(networkKeyIndex: 0, applicationKeyIndex: 0, key: 16 bytes), with error: Request timed out.
LOG: Cancelling messages with op code: 0, sent from: 0001 to: 0002
ManageNetworkCoordinator did fail to configure lamp with error: Request timed out..
```

philips77, 2021-07-30:
This time you forgot to set the `dataDelegate` in your bearer object to the mesh network manager instance:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/f4a80ab641534449b331dbed4583be613dcb7b0e/nRFMeshProvision/Classes/Bearer/Bearer.swift#L98-L100
Outgoing messages are delivered to the bearer, but incoming aren't delivered to the manager. As the manager it transport agnostic, and in theory you can have your own bearer implementation, you need to pass incoming and outgoing manually. 

It's in the first paragraph of the documentation: 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Documentation/GETTING_STARTED.md#mesh-network-manager

closed 2021-07-30 by philips77

philips77, 2021-07-30:
Or here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/Documentation/GETTING_STARTED.md#gatt-bearers