nordicsemi/IOS-nRF-Mesh-Library #366: Can you please provide guidelines how to create hsl model , because only this model not available in nrf mesh examples lots of people need but they confuse.  In light work thats imp feature . Its helps lots of people

pawarpratap555, 2021-08-15:
Can you please provide guidelines how to create hsl model , because only this model not available in nrf mesh examples lots of people need but they confuse.  In light work thats imp feature . Its helps lots of people

philips77, 2021-08-16:
nRF Mesh app contains samples for Generic On Off and Genetic Level models. HSL Client model should not be hard, as you only receive status messages, if I'm not mistaken. Check out Generic On Off Client model implementation.

The library already contains all required messages for HSL models.

pawarpratap555, 2021-08-16:
I already done example of CTL, Generic On off and dimming using nrf52840dk board. can you please tell everyone how could we develope HSL model for nrf52840dk and also how to shown on hsl model nRF mesh app. 

which document we prefer because lot of confusion to everyone how could we done hsl like CTL example

thanks and Regards

philips77, 2021-08-17:
Regarding HSL Server models on nrf side, please ask on DevZone. Our support team should know this kind of things.
Regarding HSL Client model in the nRF Mesh app, we have it in our backlog, but I can't promise any dates. For sure not this year, and I don't see it coming next year as well, sorry. We have a lot on our plate. But we accept and like PRs!

closed 2021-08-17 by philips77