nordicsemi/IOS-nRF-Mesh-Library #372: Device disconnecting issue

LukasHakulin, 2021-11-04:
**Use case description**

Hi, I'm trying to subscribe my group into models.
My issue is randomly disconnecting of device (my `GattBearer`).

I'm using `NetworkConnection` class from example project and sending `ConfigModelSubscriptionAdd` message.
Circa two of three cases it's OK(it's random behaviour), but then I get error from `BearerDelegate` -> `func bearer(_ bearer: Bearer, didClose error: Error?)`

`2021-11-04 07:40:57 +0000🟦 BearedDidClose Optional(Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
2021-11-04 08:40:57.229498+0100 DeviceBtMesh[6733:2802142] [CoreBluetooth] XPC connection invalid`

From another issue I understand `XPC connection invalid` can be connect with multiple `CBCentralManager` instances...
[https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/156](url)

`NetworkConnection` will discover the same device and open as GattBearer again. 

But meanwhile I received:
`func meshNetworkManager( _ manager: MeshNetworkManager, failedToSendMessage message: MeshMessage, from localElement: Element, to destination: Address, error: Error)`

with `Request timed out` error.

**Result**

Sending messages is failing randomly.

**Questions**

Is it possible to improve connection stability?  
Do you have similar experience?

Thanks in advance for any suggestions, I could not find answer in documentation, or the other issues.

philips77, 2021-11-04:
Is the device which you're connecting to close to the phone? Can it be that there is just poor network quality, eg long distance, a lot of other devices, including Wi-Fi?

LukasHakulin, 2021-11-05:
It's in the office. On my desk, 30 cm from my iPhone. We have WI-FI here but I'm sitting at least 10m from repeater.
That's it, nothing unusual.

philips77, 2021-11-05:
So that looks like an issue with the firmware on your device. Did you try Nordic mesh samples? Do you see the same behavior?

LukasHakulin, 2021-11-09:
Hi! I found issue on our device, I can see `BLE_MESH: Proxy SAR timeout` message in our device log but still I don't understand where is problem and what I can do to keep device connected.
Also behaviour is random, sometimes I got this message on start of provisioning procedure and sometimes I can communicate with device more than 10 mins in row without any problem.

Have you any experience with similar behaviour?

philips77, 2021-11-09:
> Have you any experience with similar behaviour?

No, i did not. Is this timeout causing the device to disconnect? Can't it just be ignored instead? Does that proxy SAR message indicate a failure in Proxy protocol reassembly over GATT to or is it coming from a higher later?

LukasHakulin, 2021-11-10:
I will try to answer as much as I know:

> Is this timeout causing the device to disconnect?

Yes, from logs I can say, with every SAR timeout log is device disconnected.

> Can't it just be ignored instead?

I have no control over device disconnecting, I just got `func bearer(_ bearer: Bearer, didClose error: Error?)` callback from `BearerDelegate`. So I don't see how can I "ignored" that... Maybe I don't understand your question correctly?

> Does that proxy SAR message indicate a failure in Proxy protocol reassembly over GATT to or is it coming from a higher later?

I don't understand Mesh communication so deeply, so I will try to debug that behaviour And I will post my answer later.
But I can post here part of code from our framework with SAR implementation, maybe you will discover something...

```switch (PDU_SAR(data)) {
    case SAR_COMPLETE:
        if (client->buf.len) {
            BT_WARN("Complete PDU while a pending incomplete one");
            return -EINVAL;
        }

        client->msg_type = PDU_TYPE(data);
        net_buf_simple_add_mem(&client->buf, data + 1, len - 1);
        proxy_complete_pdu(client);
        break;

    case SAR_FIRST:
        if (client->buf.len) {
            BT_WARN("First PDU while a pending incomplete one");
            return -EINVAL;
        }

        k_delayed_work_submit(&client->sar_timer, PROXY_SAR_TIMEOUT);
        client->msg_type = PDU_TYPE(data);
        net_buf_simple_add_mem(&client->buf, data + 1, len - 1);
        break;

    case SAR_CONT:
        if (!client->buf.len) {
            BT_WARN("Continuation with no prior data");
            return -EINVAL;
        }

        if (client->msg_type != PDU_TYPE(data)) {
            BT_WARN("Unexpected message type in continuation");
            return -EINVAL;
        }

        k_delayed_work_submit(&client->sar_timer, PROXY_SAR_TIMEOUT);
        net_buf_simple_add_mem(&client->buf, data + 1, len - 1);
        break;

    case SAR_LAST:
        if (!client->buf.len) {
            BT_WARN("Last SAR PDU with no prior data");
            return -EINVAL;
        }

        if (client->msg_type != PDU_TYPE(data)) {
            BT_WARN("Unexpected message type in last SAR PDU");
            return -EINVAL;
        }

        k_delayed_work_cancel(&client->sar_timer);
        net_buf_simple_add_mem(&client->buf, data + 1, len - 1);
        proxy_complete_pdu(client);
        break;
    }

philips77, 2021-11-14:
> I can see BLE_MESH: Proxy SAR timeout message in our device log but still I don't understand where is problem and what I can do to keep device connected.

As I understand, it is the device that disconnects when encounter this error. So by "ignore it", i meant on the device firmware side.
And for that you'll need to ask on devzone, as the support team is more qualified in this manner.

closed 2021-11-18 by philips77