nordicsemi/IOS-nRF-Mesh-Library #381: Node :  new Node with custom uuid and other info

Wtoto, 2021-11-24:
Hi, I need add a new node to meshnetwork, but there is only one pulic init method;
Please, make some internal init method to public;
This time, My requirement is to add custom uuid;
Thank you;

philips77, 2021-11-24:
Hello,
To add a node to a network you need to provision it using the mesh network manager's method:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/nRFMeshProvision/Classes/MeshNetworkManager.swift#L272-L279

Sample code:
```swift
let unprovisionedDevice = UnprovisionedDevice(advertisementData: advertisementData)

let bearer = PBGattBearer(target: discoveredPeripheral)
bearer.logger = ...
bearer.delegate = ...

let provisioningManager = meshNetworkManager.provision(unprovisionedDevice: unprovisionedDevice, over: bearer)
provisioningManager.delegate = ...
provisioningManager.logger = ...
provisioningManager.identify(andAttractFor: blinkingTime /* in seconds */)
// [...]
provisioningManager.provision(usingAlgorithm:       .fipsP256EllipticCurve,
                              publicKey:            publicKey,
                              authenticationMethod: authenticationMethod)
```
Here you may find those in the sample app:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/Example/nRFMeshProvision/View%20Controllers/Network/Provisioning/ScannerTableViewController.swift#L157

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/Example/nRFMeshProvision/View%20Controllers/Network/Provisioning/ScannerTableViewController.swift#L111-L113

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/Example/nRFMeshProvision/View%20Controllers/Network/Provisioning/ProvisioningViewController.swift#L87-L91

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/Example/nRFMeshProvision/View%20Controllers/Network/Provisioning/ProvisioningViewController.swift#L235-L287

Wtoto, 2021-11-24:
I know the regular sigmesh provision process；  But now， i implement the Bearer protocl by myself； so，i need add the new node to mesh with more init method

songhongkang, 2021-11-30:
@Wtoto  嗨，方便请教一下吗？，如何把mesh网络分享出去，其他手机也能控制

Wtoto, 2021-11-30:
> @Wtoto 嗨，方便请教一下吗？，如何把mesh网络分享出去，其他手机也能控制

使用相同的 networkkey，appkey 和 devicekey就可以了，需要云端保存这些信息；

songhongkang, 2021-11-30:
@Wtoto 似乎不行，Documents/MeshNetwork.json 。这个文件我都保存到云端，还是控制不了。
放点给个微信，请教下大神吗？

songhongkang, 2021-11-30:
@Wtoto   Documents/MeshNetwork.json、Library/Preferences/4F772E49-E777-4552-8A09-E85E9AAF0088.plist
这两个文件我都上传到云端，还是控制不了。

trusbe, 2021-11-30:
@songhongkang 

两个手机不能使用同一个本地地址，也就是说你下载的网络里面的本地节点你需要改掉那个地址；

```
// 修改本地节点地址
let randomInt = UInt16.random(in: 0x1900...0x199A)
try! manager.meshNetwork?.assign(unicastAddress: Address(randomInt), for: provisioner)
_ = manager.save()
```

例如上面我使用的就是给出一个活动范围，每个手机的本地节点地址不一样，这样就可以尽量避免两个手机同一个地址，然后存在 Index 循环计数滞后的问题；

songhongkang, 2021-11-30:
@trusbe  供应器地址必须是这个范围吗  0x1900...0x199A
如果是这个范围。供应器的地址用完了，还能复用吗？复用的地址能控制吗？

trusbe, 2021-11-30:
@songhongkang 

这个地址只是举个例子，你只要保证同时控制网络的两个手机本地地址不是一样的就可以了。

songhongkang, 2021-11-30:
@trusbe  刚才调试了一下，在A手机用的适配器地址是0x1900，在B手机用其他地址都可以，就是不能用0x1900
如果想在B手机用0x1900地址，有办法吗？
如果在B手机上不能在用0x1900，得出一个结论，地址和手机有绑定的关系？

trusbe, 2021-11-30:
@songhongkang 

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/master/nRFMeshProvision/Classes/MeshNetworkManager.swift

你可以试试修改 `SequenceNumber` 这个参数，是因为这个循环计数的原因，不能完全解决上面的问题，建议最保险的方法是随机本地节点地址。

songhongkang, 2021-12-01:
@trusbe  谢谢，我把账号和 `SequenceNumber ` 绑定好了，现在用同一个适配器的地址可以控制了。
```
    public func setSequenceNumber(sequence : Int){
        guard let defaults = UserDefaults(suiteName: meshNetWorkManger.meshNetwork?.uuid.uuidString) else {
            return
        }
        let node  = meshNetWorkManger.meshNetwork?.nodes.first
        guard let address = node?.unicastAddress else {
            return
        }
        defaults.set(sequence & 0x00FFFFFF, forKey: "S\(address.hex)")
    }

非常感谢帮忙!
    
```

closed 2021-12-01 by philips77

songhongkang, 2022-02-14:
@trusbe 你好，方便找你请教一个问题吗？发送 添加APPKey的指令，设备端搜不到指令。有一部手机很容易复现这个问题

songhongkang, 2022-02-14:
@trusbe  大佬，我的微信  
[image]