nordicsemi/IOS-nRF-Mesh-Library #384: Why mesh messages are automatically segmented when payload is bigger than 11 bytes?

tu-lucaspalaian, 2021-11-30:
I have seen into a library documented field that messages will are being segmented after 11 bytes. As far as I know, the Android library does not have this restriction and they only rely on MTU size to segment messages.

_MeshMessage.swift | Line 77_
```
/// To force segmentation for shorter messages return `true` despite
/// payload length. If payload size is longer than 11 bytes this
/// field is not checked as the message must be segmented.
var isSegmented: Bool { get }
```

_AccessPdu.swift | Line 53_
```
/// Whether the outgoing message will be sent as segmented, or not.
var isSegmented: Bool {
     guard let message = message else {
          return false
     }
     return accessPdu.count > 11 || message.isSegmented
}
```

I might be misunderstanding something, Is there any reason to have this behavior?

philips77, 2021-11-30:
Hello,
Have a look at Mesh Profile 1.0.1, chapter 3.5.2.1. Table 3.10 defines max payload size of an unsegmented access message to 40-120 bits, that is 5-15 bytes. Upper Transport Layer requires 4 or 8 bytes in a payload for checksum (chapter 3.6.2), leaving max 11 bytes for the actual payload.
You have it summarized in chapter 3.6.3 in table 3.16.

philips77, 2021-11-30:
Cc: @roshanrajaratnam

tu-lucaspalaian, 2021-11-30:
Great! Thanks for the information. 

My question then would be why I would have to add a delay (around 1.4 seconds) between segmented messages? When building an AcknowledgedStaticVendorMessage with a payload of 30 bytes, it takes up to 3-4 seconds to return an answer. BUT the first message is (almost) "immediate". With the delay added, all the segmented messages are sent fast (as expected) but without it it seems the library has to retry and the second time for that message it actually works.

The sending flow would be something like this. 

With Delay:
**First Message (Segmented)** -> ACK **OK** -> delay (**1.4**) -> **Second Message (Segmented)** -> ACK **OK** -> Delay..

Without Delay
**First Message (Segmented)** -> ACK **OK** -> **Second Message (Segmented)** -> Library timeout -> Retry sending (automatically by the library) -> ACK **OK** -> ...

We do this process up to four messages for one of our features.
Our Android app using the nordic mesh library takes almost less than a second to complete all the 4 messages and we on iOS are taking up to five times more time.

I have played with TTL and a couple of others properties of NetworkManager with no luck.

We are sending our message trough a single proxy as with a broadcast address 0xFFFF. We are controlling which node takes the messages using a property of our own into the payload.

I don't know if the answer fits this thread or its too attached for our custom solution, but with the given information do you have any suggestion for what's happening around those messages needing retry?

tu-lucaspalaian, 2021-11-30:
As an extra note, packages into the same flow with less than 11 bytes are taking the same (short) time that our android counterpart takes.

philips77, 2021-12-01:
Could you paste here the logs you get from the library, with those delays added and without?
What is this Ack OK? Is it the confirmation of the last segment, or your Access layer status message? Or just callback, that a message has been sent?

tu-lucaspalaian, 2021-12-01:
By **ACK OK** I meant that I received a response for the sent message.

The message we are trying to send is AcknowledgedStaticVendorMessage **TrainSetupAcknowledged, op code: 0xF60010** with a response type of StaticVendorMessage *PeripheralStatus, op code: 0xD30010* 

Attached both logging files I got from the library by implementing LoggerDelegate
[logs_with_custom_delay.log](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/7634116/logs_with_custom_delay.log)
[logs_without_custom_delay.log](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/7634117/logs_without_custom_delay.log)

philips77, 2021-12-03:
Hello,

Sorry, it took some time to dig into your logs and the code of the library.

Let me guide you through my findings. 

First, I've cleaned and reattached the logs below with my comments.
[logs_without_custom_delay_cleaned.log](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/7649425/logs_without_custom_delay_cleaned.log)
[logs_with_custom_delay_cleaned.log](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/files/7649426/logs_with_custom_delay_cleaned.log)
I have removed all logs regarding received messages other then the status messages, that you actually expect.

Now, have a look at:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/nRFMeshProvision/Classes/Layers/Lower%20Transport%20Layer/LowerTransportLayer.swift#L632-L660

As your segmented messages are sent to a Group Address (0xFFFF), the segments are not acknowledged by the Lower Transport Layer of receiving nodes. Instead, to increase chances of delivery, the library sends all segments twice, with a random delay between batches. The `seqZero` for both batches is the same, as that is the same Upper Transport PDU.

Notice, that `lowerTransportLayerDidSend` method is called after the Lower Transport Layer will finish sending the second batch (line 657 above).
That's why, in your logs without custom delay you have a fragment like that:
```
[...]
========= Sending TrainSetupAcknowledged to All, targetting 0005 ======
# Network Manager 10:04:53.735 # Sending TrainSetupAcknowledged(parameters: Optional(30 bytes)) from: Primary Element (0x0001), to: FFFF
# Network Manager 10:04:53.735 # Sending Access PDU (opcode: 0xF60010, parameters: 0x080D05312702014F52014F59014F56014F52014F52014F52014F52044F56)
# Network Manager 10:04:53.739 # Sending Upper Transport PDU (encrypted data: 0x72822B6588C8D43369A11B0C853BB631CFC97D757066182A2F2BCC195D47E31BA6, transMic: 0x8F86F2C2) encrypted using key: Neural Trainer Primary (index: 0)

// Here the library still waits until the second batch is sent.

======= Resending segmented message =========
# Network Manager 10:04:54.240 # Sending Segmented Access Message (akf: 1, aid: 0x1A, szmic: 0, seqZero: 5086, segO: 0, segN: 2, data: 0x7CFD73CACD87721A873FE508)
# Network Manager 10:04:54.273 # Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 5091, src: 0001, dst: FFFF, transportPdu: 0xAAD0044915293F20FBAC871B5132EBE7, netMic: 0xB3903698) encrypted using Primary Network Key (index: 0)
# Network Manager 10:04:54.307 # Sending Segmented Access Message (akf: 1, aid: 0x1A, szmic: 0, seqZero: 5086, segO: 1, segN: 2, data: 0xEDB9F7997F6BF58DAE038A05)
# Network Manager 10:04:54.340 # Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 5092, src: 0001, dst: FFFF, transportPdu: 0x3D618C4DEB7966D3E67AE62D3AEE9D10, netMic: 0xE9F5C5FB) encrypted using Primary Network Key (index: 0)
# Network Manager 10:04:54.374 # Sending Segmented Access Message (akf: 1, aid: 0x1A, szmic: 0, seqZero: 5086, segO: 2, segN: 2, data: 0x81)
# Network Manager 10:04:54.396 # Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 5093, src: 0001, dst: FFFF, transportPdu: 0xE6A7F1AB00, netMic: 0x906E604A) encrypted using Primary Network Key (index: 0)

// Now it can actually send new message.

========= Sending TrainSetupAcknowledged to All, targetting 0005, continuation ======
# Network Manager 10:04:54.428 # Sending Segmented Access Message (akf: 1, aid: 0x1A, szmic: 0, seqZero: 5090, segO: 0, segN: 3, data: 0x72822B6588C8D43369A11B0C)
# Network Manager 10:04:54.452 # Sending Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 5, seq: 5094, src: 0001, dst: FFFF, transportPdu: 0xA6CF587A98EAD25C34D39C836238F204, netMic: 0xC1F242F6) encrypted using Primary Network Key (index: 0)
[...]
```

Other than that, I see nothing weird and worth investigating in the library.

The issue you observe must be caused by something else. Have a look at line 68+ of the logs without custom delay. After the first batch is actually sent, the library starts sending second **TrainSetupAcknowledged ** message. It sends, and no response is received. Then it repeats it again, still with no success. That ends at line 106. Here, the status should be received (at least with high probability), but it's not.
What can be the cause of that? There was a part of the logs in your files, that I removed, proving that the Proxy node has received all segments of the second message, targeting 0005, when it was first sent:
```
# Network Manager 10:04:54.566 # Network PDU (ivi: 0, nid: 0x0F, ctl: 0, ttl: 4, seq: 5094, src: 0001, dst: FFFF, transportPdu: 0xE25963217AACC789BF47A321F8222318, netMic: 0x533432BA) received
[...]
# Network Manager 10:04:54.567 # Segmented Access Message (akf: 1, aid: 0x1A, szmic: 0, seqZero: 5090, segO: 0, segN: 3, data: 0x72822B6588C8D43369A11B0C) received (decrypted using key: Primary Network Key (index: 0))
```
Notice the TTL is 4. It was initially sent with TTL = 5, processed by Proxy node, which decreased TTL to 4 and sent it back to the phone, as it is subscribed to FFFF.
That means, that you didn't get the status not because of the nRF Mesh library, or even the Proxy node, but something must have failed further on the path. The message was sent immediately after second batch of the 1st message, so perhaps some packets were dropped, or your node was still busy processing what it has just received. The custom delay seems helping it to finish what it was doing. Try checking the logs on your final node 0005.

philips77, 2021-12-03:
Also, may I ask why fo you sent messages to FFFF with the real target inside the Access PDU layload? Also, why FFFF, not some custom group? This looks like building a custom routing on top of BLE spec.

philips77, 2021-12-03:
Regarding retransmission of a segmented PDU when sent to Group address, check Mesh Profile 1.0.1, chapter 3.5.3.3 (Segmentation behavior):
> If the Lower Transport PDUs of the segmented Upper Transport PDU are being sent to a group address or a virtual address, then the lower transport layer shall send all Lower Transport PDUs of the segmented Upper Transport PDU. It is recommended to send all Lower Transport PDUs multiple times, introducing small random delays between repetitions.

philips77, 2021-12-16:
I'm closing the issue hoping I answered your question.

closed 2021-12-16 by philips77