nordicsemi/IOS-nRF-Mesh-Library #386: Set Default Proxy Filter

adriansergheev, 2021-12-07:
Hi,
This does seem more appropriate for GitHub Discussions, however:
It does not seem possible (or perhaps I can't figure out how) to set the exclusion list as default for the proxy filter. I do have to re-set it manually, which is not a good solution.

Am I missing something?

philips77, 2021-12-07:
You're right, initially the library adds all Node's element's addresses and group address with at least one subscribed model, plus all nodes group address.
Setting a different initial state would require changes in the source. Let's say it's a feature request. I'll come to you when I resume work on this project.

adriansergheev, 2021-12-08:
What would be the desired interface for this? 
i.e, do you think it should be passed in thorough MeshNetworkManager init or something else?
I could try opening a PR

philips77, 2021-12-09:
I think it could be in the init, yes. The `ProxyFilter` instance is created in the init:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/nRFMeshProvision/Classes/MeshNetworkManager.swift#L221-L231
You could create an enum, something like:
```swift
public enum ProxyFilterSetup {
  case automatic
  case inclusionList(addresses: Set<Address>)
  case exclusionList(addresses: Set<Address>)
}
```
which you could pass as an additional param there, initially set to `.automatic`. You'd pass it to the `ProxyFilter`'s init and use in 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/nRFMeshProvision/Classes/ProxyFilter.swift#L221-L238
The current impl would of course be the automatic one.

How about that?

adriansergheev, 2021-12-09:
Having the enum associated type holding the set of the addresses also requires quite a lot of changes, and I might break something I don't fully understand.
I also do not understand the concept of .automatic in this context, so I went with a simpler solution which required minimal changes:
This seems to work, but it retains the peculiarity that you have to add your addresses after each reset
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/pull/387

closed 2022-04-26 by philips77