nordicsemi/IOS-nRF-Mesh-Library #391: Export database to another iPhone device but unable to operate BLE device with another iPhone.

mirantvolansys, 2021-12-24:
**Describe the bug**
I have one lightning controller which is operated on/off by a "Generic on-off server". And it is working as per expectation.

Issue:- When I export my database from 1 iPhone to another and try to operate the same on/off operation from 2nd iPhone then it does not work. Mostly I get timeout error or error of getting bearer close.
Note:- I have tried updating the local provisioner's unicast address but failed.
-> I am using the latest build.

-> Anything I have missed? Can you please give me a solution, how can I operate the BLE device from another device 2 in which I have imported the database?

mirantvolansys, 2021-12-30:
Any updates on this?

philips77, 2021-12-30:
Hello, sorry for the late response. It's holiday season and you may expect longer than usual waiting times.

In general, before exporting, you should create a new provisioner with unique address and address ranges and only then export. But what you described, with charging the address, should also work.

mirantvolansys, 2021-12-30:
Hi @philips77,

We have tried both ways, especially for iPhone.

Approch1:- We have created another provisioner before exporting the database
Approch2:- We have export provisioner from iPhone 1 and import it to iPhone 2. After that, I have changed provision's address to unique one.

And tried to operate our BLE devices from iPhone2 which is provision/configure from iPhone 1.

We can't operate the BLE devices.

Can you check iOS SDK again and do the same steps that I have done?

Kindly provide the way or solution if you found one for iOS SDK.

philips77, 2021-12-30:
On the second iPhone, do you have automatic connection set in Proxy tab, or did you connect manually to any device with GATT proxy feature?
Usually, a device with proxy feature supports only one GATT connection, so you may need 2+ devices in the network that you can use.

mirantvolansys, 2022-01-03:
Hi @philips77 ,

We have used an automatic in connection set.

Right now, with 3 devices it's working. After more testing, I will close this issue.

Thank you so much for the help.

closed 2022-01-03 by mirantvolansys