nordicsemi/IOS-nRF-Mesh-Library #392: Found Error while we use multiple network keys in Nordic SDK

mirantvolansys, 2021-12-30:
**Describe the bug**
-> we have created a different application key and bound it with a different network key for the section. But due to using multiple network keys, we found an issue in configuring devices. We have attached a video for it.

-> As per the video, When we create a network key section 1 and bind it with app key section 1 and provision and configure device 1 that time it is working successfully. When we create network key section 2, bind it with app key section 2, provision it till the time it is working properly. But when we try to configure it that time always we got a request time out error in the nordic SDK itself.

Note:- In android, it is working fine.

**To Reproduce**
Do steps as per the video

**Expected behavior**
Users can use multiple network keys and application keys and can configure BLE devices.

**Platform details:**
 - Device: iPhone 12
 - OS: iOS 15.1.1
 - Library Version: 3.1.4

**Logs / Screenshots/ Video**

https://user-images.githubusercontent.com/53000658/147736804-182b7637-436f-499f-9141-3d362f1ddf5e.mp4

mirantvolansys, 2022-01-03:
hey,

Any updates on it?

philips77, 2022-01-04:
Hello, that will not work, as the second Composition Data Get message can't be relayed by the proxy. The new network let is only known to the new node, and the app isn't connected to it directly, but it's using previously connected proxy node. 

For a message to be relayed, the relay it proxy node has to be able to decode it to decrement ttl value before retransmission.

Send the new network key to the connected proxy node and try getting composition data on the new node again.

mirantvolansys, 2022-01-05:
Hi @philips77,

We have tested on the nordic App. Right now, it's working.

I will close this ticket once we are tested more on it.

mirantvolansys, 2022-01-11:
Hi @philips77,

I understand that we need to update the proxy manually for using a particular BLE device associated with the particular network key.

But now the issue is, When we search for the proxy node available then that time how can I identify that I want to use this proxy node. Because when we search peripheral as the proxy node that time we can't get any specific data to compare.

Like:- Searched proxy node gave me the name, identifier, etc, but did not give me a specific network key associated with that node.

So, how can I identify the network key associated with the proxy node which we found in searching?

mirantvolansys, 2022-01-12:
Hi @philips77 ,

Any update on this??

How can I get a network key or network Id which is equivalent to our network keys which we have bound to the device at the time of provision?

closed 2022-01-12 by mirantvolansys