nordicsemi/IOS-nRF-Mesh-Library #397: When I use an iPhone 8 or below,it always fails to unbind Model's Appkey when adding devices

yyzanll, 2022-03-04:
1. We have obtained TTL and Element and Element has been bound appKey successfully
2. We entered Element's Model and failed to bind the AppKey of the model

philips77, 2022-03-04:
Could you share logs? They should be available in Console app on Mac after connecting the phone with a cable.

songhongkang, 2022-03-09:
2022-03-09T17:24:19+0800:[Bearer] Central Manager state changed to .poweredOn
2022-03-09T17:24:19+0800:[Bearer] Connecting to Life_is_good...
2022-03-09T17:24:20+0800:[Bearer] Connected to Life_is_good
2022-03-09T17:24:20+0800:[Bearer] Discovering services...
2022-03-09T17:24:20+0800:[Bearer] Service found
2022-03-09T17:24:20+0800:[Bearer] Discovering characteristics...
2022-03-09T17:24:20+0800:[Bearer] Data In characteristic found
2022-03-09T17:24:20+0800:[Bearer] Data Out characteristic found
2022-03-09T17:24:20+0800:[Bearer] Enabling notifications...
2022-03-09T17:24:20+0800:[Bearer] Data Out notifications enabled
2022-03-09T17:24:20+0800:[Bearer] GATT Bearer open and ready
2022-03-09T17:24:20+0800:[Provisioning] Requesting Provisioning Capabilities
2022-03-09T17:24:20+0800:[Provisioning] Sending Provisioning Invite (attention timer: 5 sec)
2022-03-09T17:24:20+0800:[Bearer] -> 0x030005
2022-03-09T17:24:20+0800:[Bearer] <- 0x03010100010000000000000000
2022-03-09T17:24:20+0800:[Provisioning] Device Capabilities: Number of elements: 1
Algorithms: FIPS P-256 Elliptic Curve
Public Key Type: None
Static OOB Type: None
Output OOB Size: 0
Output OOB Actions: None
Input OOB Size: 0
Input OOB Actions: None received
2022-03-09T17:24:20+0800:[Provisioning] Provisioning Capabilities received
2022-03-09T17:24:21+0800:[Provisioning] Provisioning started
2022-03-09T17:24:21+0800:[Provisioning] Sending Provisioning Start (algorithm: FIPS P-256 Elliptic Curve, public Key: No OOB Public Key, authentication Method: No OOB)
2022-03-09T17:24:21+0800:[Bearer] -> 0x03020000000000
2022-03-09T17:24:21+0800:[Provisioning] Sending Provisioner Public Key (0xBCF66149C0AF234E45D9D4027BAB02DB1DB09231357EA31ED4EC0CBAF3880E4EB6D2816739F142D53037FA4F7B72EDC31AF6EFB837B60A522768646A8E8D18E7)
2022-03-09T17:24:21+0800:[Bearer] -> 0x4303BCF66149C0AF234E45D9D4027BAB02DB1DB0
2022-03-09T17:24:21+0800:[Bearer] -> 0x839231357EA31ED4EC0CBAF3880E4EB6D2816739
2022-03-09T17:24:21+0800:[Bearer] -> 0x83F142D53037FA4F7B72EDC31AF6EFB837B60A52
2022-03-09T17:24:21+0800:[Bearer] -> 0xC32768646A8E8D18E7
2022-03-09T17:24:21+0800:[Bearer] <- 0x43035763733773E50DE0B6DBD515DE8A71A06F0B
2022-03-09T17:24:21+0800:[Bearer] <- 0x83179C831600090E7346FC2EB487AA543A1A8166
2022-03-09T17:24:21+0800:[Bearer] <- 0x8344638841ABA8443162C0854341E12C09CE524C
2022-03-09T17:24:21+0800:[Bearer] <- 0xC3A22C1D54B1E9E096
2022-03-09T17:24:21+0800:[Provisioning] Device Public Key (0x5763733773E50DE0B6DBD515DE8A71A06F0B179C831600090E7346FC2EB487AA543A1A816644638841ABA8443162C0854341E12C09CE524CA22C1D54B1E9E096) received
2022-03-09T17:24:21+0800:[Provisioning] Sending Provisioner Confirmation (0x2AC39D0A33F5F21202899C046C518B1B)
2022-03-09T17:24:21+0800:[Bearer] -> 0x03052AC39D0A33F5F21202899C046C518B1B
2022-03-09T17:24:21+0800:[Bearer] <- 0x0305F5EF0B36B158CC01727BC38A3713A1D3
2022-03-09T17:24:21+0800:[Provisioning] Device Confirmation (0xF5EF0B36B158CC01727BC38A3713A1D3) received
2022-03-09T17:24:21+0800:[Provisioning] Sending Provisioner Random (0x1711767825A792BA2FF9B83E0B7F6093)
2022-03-09T17:24:21+0800:[Bearer] -> 0x03061711767825A792BA2FF9B83E0B7F6093
2022-03-09T17:24:21+0800:[Bearer] <- 0x03068165C28BFBC900BD9A67463A1563DE2F
2022-03-09T17:24:21+0800:[Provisioning] Device Random (0x8165C28BFBC900BD9A67463A1563DE2F) received
2022-03-09T17:24:21+0800:[Provisioning] Sending Encrypted Provisioning Data (0xB2CAAB02BE7E5C1FC65596D38CCB73F9DE4E67711B3697AA84348944A60C422E10)
2022-03-09T17:24:21+0800:[Bearer] -> 0x4307B2CAAB02BE7E5C1FC65596D38CCB73F9DE4E
2022-03-09T17:24:21+0800:[Bearer] -> 0xC367711B3697AA84348944A60C422E10
2022-03-09T17:24:21+0800:[Bearer] <- 0x0308
2022-03-09T17:24:21+0800:[Provisioning] Provisioning Complete received
2022-03-09T17:24:21+0800:[Provisioning] Provisioning complete
2022-03-09T17:24:21+0800:[FoundationModel] Sending ConfigAppKeyAdd(networkKeyIndex: 0, applicationKeyIndex: 0, key: 16 bytes) to: 0003
2022-03-09T17:24:21+0800:[Access] Sending Access PDU (opcode: 0x 0, parameters: 0x000000B2450D5551B94F420527B2A0BEDF5011)
2022-03-09T17:24:21+0800:[UpperTransport] Sending Upper Transport PDU (encrypted data: 0xB4FB786C7EE5B32BEB9E09CC64D22661BDF34A4E, transMic: 0xC1F0C0B3) encrypted using key: Life_is_good's Device Key
2022-03-09T17:24:21+0800:[LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 7174, segO: 0, segN: 1, data: 0xB4FB786C7EE5B32BEB9E09CC)
2022-03-09T17:24:21+0800:[Network] Sending Network PDU (ivi: 0, nid: 0x34, ctl: 0, ttl: 5, seq: 7175, src: 0001, dst: 0003, transportPdu: 0x1B4794F3B257DD7B74E3F7B9BE0DD411, netMic: 0xB662658E) encrypted using Primary Network Key (index: 0)
2022-03-09T17:24:21+0800:[Bearer] Central Manager state changed to .poweredOn
2022-03-09T17:24:21+0800:[Bearer] Connecting to Life_is_good...
2022-03-09T17:24:21+0800:[Bearer] Connected to Life_is_good
2022-03-09T17:24:21+0800:[Bearer] Discovering services...
2022-03-09T17:24:21+0800:[LowerTransport] The bearer is closed.
2022-03-09T17:24:21+0800:[LowerTransport] Sending Segmented Access Message (akf: 0, szmic: 0, seqZero: 7174, segO: 1, segN: 1, data: 0x64D22661BDF34A4EC1F0C0B3)
2022-03-09T17:24:21+0800:[Bearer] Service found
2022-03-09T17:24:21+0800:[Bearer] Discovering characteristics...
2022-03-09T17:24:21+0800:[Network] Sending Network PDU (ivi: 0, nid: 0x34, ctl: 0, ttl: 5, seq: 7176, src: 0001, dst: 0003, transportPdu: 0xB0648234E7034A6BFA1DDF7DA83392FD, netMic: 0xD4F11565) encrypted using Primary Network Key (index: 0)
[... cut ...]
2022-03-09T17:24:26+0800:[Bearer] -> 0x4034F8838929A6F210CEEE61240BBF3977CC5E8C
2022-03-09T17:24:26+0800:[Bearer] -> 0xC09D39C8CB95551D088124
2022-03-09T17:24:26+0800:[Network] Sending Network PDU (ivi: 0, nid: 0x34, ctl: 0, ttl: 5, seq: 7184, src: 0001, dst: 0003, transportPdu: 0x22862892209B97E0B0216EF324B43A24, netMic: 0xACC0A478) encrypted using Primary Network Key (index: 0)
2022-03-09T17:24:26+0800:[Bearer] -> 0x4034AAFC1A33936A63CD22862892209B97E0B021
2022-03-09T17:24:26+0800:[Bearer] -> 0xC06EF324B43A24ACC0A478

songhongkang, 2022-03-09:
@philips77  help me ！thank you  my  brother ！

philips77, 2022-04-26:
Hello, any update on this?

philips77, 2023-10-17:
Closing due to long time of inactivity.

closed 2023-10-17 by philips77