nordicsemi/IOS-nRF-Mesh-Library #398: Can we make the provisioned device a proxy from the provision delegate method?

bhautikVolansys, 2022-03-14:
ISSUE : After provision the device we are starting for proxy scanning and it will taking too much time. 
QUERY : Can we make the provisioned device a proxy from the provision delegate method?

Description
Suppose we can get provisioned device from provision delegate method we can make proxy instant and it will save time of proxy scanning for further process. 
GOAL : skip steps 4 & 5 to save time.

We follow the below steps into our project to make a proxy :
 1 Scanning for UnProvisioned Device
 2 Selecting device from that list
 3 Provision the selected device
 4 Start scanning for the provisioned device to make proxy
 5 Compared those devices with network id 
 6 We will make it proxy and open proxy-bearer 
 7 Configure that device (bind app key, composition data, subscription)

bhautikVolansys, 2022-03-17:
@philips77  Any Update on this?

bhautikVolansys, 2022-03-23:
@philips77
Hi, Any Update on this?

philips77, 2022-03-23:
I'm still busy with other projects, but how about an option to get `GattBearer` from the `PBGattBearer` instance. The `PBGattBearer` is an object you get when you scanned your Unprovisioned Device
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1755555f76fb6f393bfdad37a23566ddd581536/Example/nRFMeshProvision/View%20Controllers/Network/Provisioning/ScannerTableViewController.swift#L111
while `GattBearer` when a Network ID beacon or Node Identity beacon were scanned:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/c1755555f76fb6f393bfdad37a23566ddd581536/Example/nRFMeshProvision/Mesh%20Network/NetworkConnection.swift#L203

If really needed, the one could be converted to the second one so you could quickly reconnect to the same device (assuming it has the same MAC address).

philips77, 2023-10-17:
Hi,
Have a look at version 4.0. nRF Mesh app now asks should it switch GATT Proxy connection to the newly provisioned device if PB GATT was used for provisioning.

closed 2023-10-17 by philips77