nordicsemi/IOS-nRF-Mesh-Library #399: Can you let me know, how we can set proxy to selected device ?

bhautikVolansys, 2022-03-17:
Hello Team,  

Greetings!

Can you let me know, how we can set proxy to the selected device?

I have selected the device after that provision that device and started proxy scanning. I have a Unique ID of that selected device. Can we set proxy to the selected device with a unique ID?

Thanks

philips77, 2022-03-17:
After successful provisioning a mesh node starts advertising with Node Identity beacon, which includes Node's Unicast Address. A scanner can search for such packet and connect to the device. Have a look at:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/3fb1d7d1c12d9b5c9648ae1d6d3f5a3e7f532a88/Example/nRFMeshProvision/Mesh%20Network/NetworkConnection.swift#L183-L214

Using the obtained `hash` and `random` you get the matching Node as follows:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/3fb1d7d1c12d9b5c9648ae1d6d3f5a3e7f532a88/nRFMeshProvision/Classes/Mesh%20API/MeshNetwork%2BNodes.swift#L103-L126

Actually, it would be nice to get the matching Node, not only true/false.

You may also make any Node start advertising Node Identity beacon by sending `ConfigNodeIdentitySet` message (for some reason there's no such message in the current version, I'll add it soon, this looks like a mistake).

philips77, 2022-03-17:
Task: #400

philips77, 2022-03-17:
Return Node matching Node Identity beacon: #401

philips77, 2022-04-26:
#400 and #401 are fixed.

closed 2022-04-26 by philips77