nordicsemi/IOS-nRF-Mesh-Library #402: Can application get the chance to handle SecureNetworkBeacon?

Wtoto, 2022-03-18:
[image]
The Library handle the SecureNetworkBeacon. But we want control the ivindex update by ourself.  So a delegate maybe helpfule.

philips77, 2022-03-18:
Btw, you may point to a code using a permalink, like that:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/bedfaeaf29e47c866b300910effce81e19ef8bc6/nRFMeshProvision/Classes/Layers/Network%20Layer/NetworkLayer.swift#L113-L118

Wtoto, 2022-03-21:
We found in some case, the devices send mesh beacon to update the ivindex unexpectedly. So we just want to update ivindex by the provisoner. The ivindex is not updated as soon as the message is received.

philips77, 2022-04-26:
Hi,
I need more information to understand your case.
Are you interested in the current IvIndex? Or when it changes? Or when the Secure Network beacon is received (it may be discarded)? 

> So we just want to update ivindex by the provisoner. The ivindex is not updated as soon as the message is received.

Of which provisioner? The local one (the phone?)? The IvIndex is updated automatically when Secure Network beacon is received and the IvIndex can be increased (there are some rules when it can happen, see [here](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1043b541d00c35de5e4fab4e7598282ad3fb4986/nRFMeshProvision/Classes/Layers/Network%20Layer/SecureNetworkBeacon.swift#L128)).

eanderlind, 2023-03-19:
Forwarding to App easy to do - anyone can extend/modify NetworkConnection.bearer(_ bearer: Bearer, didDeliverData data: Data, ofType type: PduType)

Is e.g. useful for App to monitor which subnets are avaiable