nordicsemi/IOS-nRF-Mesh-Library #423: AccessLayer always times out for AcknowledgedMeshMessages sent to all nodes

bspinner, 2022-05-17:
**Describe the bug**

Disclaimer: One could argue the behaviour is as intended, but let's see... :)

We've implemented a TimeSet message which adopts `AcknowledgedMeshMessage`,
so we can send it to a specific node and get an acknowledgement.

Another use case is time synchronisation of the entire mesh network.
To do so, we just send a TimeSet message to all nodes (destination `0xFFFF`)

`AccessLayer` always waits for an acknowledgement. So it does in this case.
But it doesn't interpret all the status messages from all over the mesh network as acknowledgements.
Instead it sends the same message over and over again until it gives up and times out.

**To Reproduce**
Steps to reproduce the behavior:

1. Create message class adopting `AcknowledgedMeshMessage`

2. Send message to address `0xFFFF`

**Expected behavior**

Either:
a. Library does not wait for acks when message is sent to all nodes
b. Library somehow better explains what's happening (log a warning?)

**Platform details:**
not relevant

**Logs / Screenshots**

```
14:51:34.903 x INFO MeshLogger.log():12 - MeshLog: [model] Sending TimeSet(taiTime: Domain.TaiTime(taiSeconds: 706107121, subsecond: 3, uncertainty: 0, isTimeAuthority: false, taiUtcDelta: 37, timeZoneOffset: 8)) from: Primary Element (0x0001), to: FFFF
14:51:34.905 x INFO MeshLogger.log():12 - MeshLog: [access] Sending Access PDU (opcode: 0x5C, parameters: 0xF156162A000300480248)
14:51:34.991 x INFO MeshLogger.log():12 - MeshLog: [access] Access PDU (opcode: 0x5C, parameters: 0xF156162A000300480248) received (decrypted using key: xxx)
14:51:35.088 x INFO MeshLogger.log():12 - MeshLog: [access] Access PDU (opcode: 0x5D, parameters: 0xF156162A000300480248) received (decrypted using key: xxx)
14:51:35.093 x INFO MeshLogger.log():12 - MeshLog: [model] TimeStatus(taiTime: Domain.TaiTime(taiSeconds: 706107121, subsecond: 3, uncertainty: 0, isTimeAuthority: false, taiUtcDelta: 37, timeZoneOffset: 8)) received from: 0014, to: 0001
14:51:35.095 x VERBOSE MeshNetworkDelegateWrapper.meshNetworkManager():38 - didReceiveMessage: TimeStatus(taiTime: Domain.TaiTime(taiSeconds: 706107121, subsecond: 3, uncertainty: 0, isTimeAuthority: false, taiUtcDelta: 37, timeZoneOffset: 8)), source: 20, destination: 1
14:51:35.102 x INFO xxx():56 - incoming mesh message: Domain.TimeStatusMessage from source 20 to 1
.
.
.
.
.
14:51:37.219 x DEBUG MeshLogger.log():11 - MeshLog: [access] Resending Access PDU (opcode: 0x5C, parameters: 
14:52:05.006 x WARNING MeshLogger.log():13 - MeshLog: [access] Response to Access PDU (opcode: 0x5C, parameters: 0xF156162A000300480248) not received (timeout)
14:52:05.012 x WARNING MeshLogger.log():13 - MeshLog: [model] TimeSet(taiTime: Domain.TaiTime(taiSeconds: 706107121, subsecond: 3, uncertainty: 0, isTimeAuthority: false, taiUtcDelta: 37, timeZoneOffset: 8)) sent from: 0001, to: FFFF timed out
14:52:05.013 x INFO MeshLogger.log():12 - MeshLog: [access] Cancelling messages with op code: 92, sent from: 0001 to: FFFF
14:52:05.020 x VERBOSE MeshNetworkDelegateWrapper.meshNetworkManager():55 - failedToSendMessage: TimeSet(taiTime: Domain.TaiTime(taiSeconds: 706107121, subsecond: 3, uncertainty: 0, isTimeAuthority: false, taiUtcDelta: 37, timeZoneOffset: 8)), localElement: Primary Element (0x0001), to: 65535, error: timeout
```

philips77, 2022-05-18:
Thank you for creating the issue. Let me discuss with the team what should be the desired behavior. I thought it will wait for a first response form any Node, but indeed, there don't even has to be any other Node.

I'll come back to you soon.

bspinner, 2022-05-18:
For now we work around the problem by having two implementations of `TimeSet`. One adopting `GenericMessage`, the other one adopting `AcknowledgedGenericMessage`.

philips77, 2022-05-20:
Hello again,

Here's the reply I got from one of our colleagues:

> It is not necessary to send Time Set as broadcast to the entire network to set the time on all nodes. Time model has a way to propagate this information fairly quickly across the network. To do that, the network administrator have to configure some nodes as "Time Relays". These time relays should be such that all nodes in the network are near to at least one time relay. All time relays also should be configured for periodic publication of say 60 seconds with publish address set to 0xFFFF and publish TTL set to 0. Then, the node configured as a Time Authority, can send a Time Set message with TTL 0 to only first nearby time relay. The relay will then process this message and send a new Time Status message with TTL 0. This allow next time relay to pickup this message and adjust its own time and publish Time Status again. This way time is propagated across whole network in few seconds without generating too much traffic.

This is implicitly described in : "5.3.1.2.2 Sending a Time Status message" and "5.3.1.2.3 Receiving a Time Status message".

philips77, 2022-05-20:
Anyway, despite the mentioned use case, the library should not wait for Status messages sent to Group addresses. This seems to be a bug and I'll look into it.

philips77, 2022-05-20:
Fixed. Could you check on the main branch before I publish a new version some time next week?

closed 2022-05-20 by philips77

bspinner, 2022-05-23:
Seems to work as intended :)

PR #428 (2022-05-20): Bug fix: Acknowledged messages sent to group addr should not await Statuses
This PR fixes #423.

In the previous implementation a reliable context was sent for all Acknowledge messages. A timer in that context was awaiting a Status message sent from the address used as the destination of the acknowledged message. However, such Status was not possible, as every message is sent with source address set to a Unicast Address of the sending Element, therefore such messages were always timing out.

After the change, the reliable context is only set when the destination address is of Unicast type.