nordicsemi/IOS-nRF-Mesh-Library #425: Which DevicePropertyCharacteristic we can use for sensor Property CO2?

bhautikVolansys, 2022-05-19:
Hello Team, 
Greeting!

Can you help me out to suggest DevicePropertyCharacteristic for Sensor property CO2?
As a we are using Nordic Kit's Temperature, Humidity, Par & CO2 property for setSensorCadence in our project.
Which DevicePropertyCharacteristic we can use for sensor Property CO2? 
Notes:- In our exist code we have specific DevicePropertyCharacteristic as like given below list:
   - For Temperature we have temperature(Decimal?)
   - For Humidity we have humidity(Decimal?)
   - For Par we have illuminance(Decimal?)
   - For CO2 we are findings...

bhautikVolansys, 2022-05-19:
@philips77 , Can you please take a look into this?

philips77, 2022-05-19:
Here is the spec: https://www.bluetooth.com/specifications/specs/mesh-device-properties-2/

philips77, 2022-05-19:
Have a look at:

> 3.5.12 Present Ambient Carbon Dioxide Concentration device property
This property represents present ambient carbon dioxide concentration measured by a CO2 sensor.
Characteristic: CO2 Concentration

philips77, 2022-05-19:
Characteristics are specified here: https://www.bluetooth.com/specifications/specs/gatt-specification-supplement-6/

philips77, 2022-05-19:
Chapter 3.45 is what you're looking for.

philips77, 2022-05-19:
You're looking for this property:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/00ef0cc1ebaf054348e8a0ab2fa351dd4b5fcef1/nRFMeshProvision/Classes/Mesh%20Messages/DeviceProperty.swift#L159

philips77, 2022-05-19:
Indeed, it's missing here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/00ef0cc1ebaf054348e8a0ab2fa351dd4b5fcef1/nRFMeshProvision/Classes/Mesh%20Messages/DeviceProperty.swift#L933-L942
and somewhere here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/00ef0cc1ebaf054348e8a0ab2fa351dd4b5fcef1/nRFMeshProvision/Classes/Mesh%20Messages/DeviceProperty.swift#L1090-L1092

philips77, 2022-05-19:
I can add it tomorrow, i think.

bhautikVolansys, 2022-05-20:
> You're looking for this property:
> 
> https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/00ef0cc1ebaf054348e8a0ab2fa351dd4b5fcef1/nRFMeshProvision/Classes/Mesh%20Messages/DeviceProperty.swift#L159

Yes @philips77, but in DevicePropertyCharacteristic

bhautikVolansys, 2022-05-20:
> I can add it tomorrow, I think.

Thank you @philips77, this will help me to implement the sensorCadence feature for CO2.

philips77, 2022-05-20:
Merged. The next version is coming in few days.

philips77, 2022-05-20:
I forgot about few things...

PR #426 (2022-05-20): CO2 and VOD concentration added
This PR fixes #425 by adding `.co2Concentration` and `.vodConcentration` characteristics.
The value for both cases is `UInt16?` with 0xFFFE meaning "value is 65543 ppm or more".
closed 2022-05-20 by philips77