nordicsemi/IOS-nRF-Mesh-Library #429: overide ConfigClientHandler

eanderlind, 2022-05-20:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/Layers/Foundation%20Layer/ConfigurationClientHandler.swift#L33

I am looking to chain together a sequence of Config messages. To do so I need to know when a response has been received. Is there a way to hook into the message receipt flow (to manage message dependencies and tx flow control)? Periodically polling a node instance or a queue isn't attractive.

If not currently supported,simplest solution would be to either change the access class from `internal` to `open access`, or equivalently move the file from inside the Framework to the same folder as all the other Model delegates, so that can add functionality. Having to clone the whole Framwork to make the change isn't appealing. 

It is possible same argument can be made for the SceneClientHandler - I am not sure yet if will be needed.

philips77, 2022-05-21:
You'll get all Status messages in `MeshNetworkDelegate`, from where you may send next ones. But a queue would beat nice feature, I believe.

eanderlind, 2022-05-21:
Can you explain some more... 
I see in AccessLayer.handle() that it is possible to add "duplicate model listeners" for anything received that used an AppKey, however same functionality doesn't extend to Foundation model messages encrypted with deviceKey, and only the first handler in the models array receives the message. 
MeshNetwork.localElements() explicitly removes any Foundation models and then reinserts its own at the head of the array (and the access class for this code doesn't permit overriding the behaviour).
How can developer change ConfigurationClientHandler behavior when it is an inaccessible part of the framework?

eanderlind, 2022-05-21:
..and as to matching requests to responses, only the ModelDelegate appears to have  all the context to do so. For messages such as ConfigModelAppBind one needs to decode and look at the msg parameters to do a reliable match. If somehow take a copy of (sunny-day-only) status msg would have to repeat all the Msg switch work done in ModelDelegate.

philips77, 2022-05-23:
Sure.

In the `MeshNetworkManager` there's a `delegate` property of type [`MeshNetworkDelegate`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/MeshNetworkDelegate.swift#L35), which notifies about all received messages and the status of sent ones.
Indeed, the Configuration Server and Client model handlers are internal to the library, so the `ModelDelegate`s are not available, but you still may get the message using the above-mentioned delegate.
It is called after the messages were processed by the handlers, here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/Layers/Access%20Layer/AccessLayer.swift#L510-L511
which calls 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/Layers/NetworkManager.swift#L243-L255

So you may ignore the `ConfigServerHandle` and just work with the mesh network delegate to send following messages.

Have a look at the example app nRF Mesh. Each of the screens assigns a delegate to itself:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/ConfigurationViewController.swift#L69-L72
and the delegate is implemented here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/Example/nRFMeshProvision/View%20Controllers/Network/Configuration/ConfigurationViewController.swift#L474-L513

On this screen, when there is no Composition Data yet, it will first read the composition data, and the TTL afterwards. Anyway, the Status messages received are handled. You, however, don't get the request that triggered this Status, but you will get it earlier in 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/MeshNetworkDelegate.swift#L66-L68

closed 2022-05-24 by eanderlind